ImpulseAdventure/GUIslice #353: not more than 18 items possible when graph is created

mcarosh, 2021-01-22:
### Describe the bug
I am not able to create more than 18 elements if graph is drawn

### Device hardware
- *MCU*: Arduino Due
- *Display with link*: Generic 8bit ILI9486 3.5" resistive 
- *GUIslice config selected*: ard-shld-generic1_35_touch

### Expected behavior
more items could be created 

### Additional info
When graph is enabled I can create more than 18 items without any problems but they are not shown on the display
If I disable the graph then they are displayed ???
In this case there are a lot buttons involved to increase/decrease different values.
I guess it´s a buffer problem and I think/hope the Arduino Due should not have a problem with flash size.
My sketch uses 28% of storage space until now.

I would be happy for any help

mcarosh

Pconti31, 2021-01-23:
@mcarosh In your config file set:
```
  #define DEBUG_ERR               2   // 1,2 to enable, 0 to disable
```
and uncomment at the bottom of your config file:
```
  #define DBG_LOG           // Enable debugging log output
```
Then check your serial output.
Paul--

mcarosh, 2021-01-23:
Hi Paul
As always thanks for your help
I did enable the debug function and there is this one error message:

ERROR: GetElemFromRefD(Line 3314) pElemRef is NULL

I´ve checked all my pointers but couldn´t find any errors?
And what I really do not understand is (Line3314) within the error message because my sketch did not exceed 1200 Lines???

mcarosh

Pconti31, 2021-01-23:
@mcarosh 3314 is the line number inside your GUIslice.c file.  Now the current version doesn't make sense so you are running an earlier version of GUIslice so you need to look at your copy of GUIslice.c to determine the routine causing the problem.  Are you sure nothing else is coming up?  I would guess you added a Element by hand and didn't increase the pages memory in your _GSLC.h file, example the:

#define MAX_ELEM_PAGE_MAIN **2** // # Elems total on page
 
or whatever page you might have added.
Paul--

mcarosh, 2021-01-23:
Hi Paul
Really really thank you for your help and I´m so sorry It was absolutely my fault.  Added page manually and during copy and paste I totally forget to change 
#define MAX_ELEM_PG4_RAM   MAX_ELEM_PG3 - MAX_ELEM_PG4_PROG
to 
#define MAX_ELEM_PG4_RAM   MAX_ELEM_PG4 - MAX_ELEM_PG4_PROG

It was just one number and I ve always did not recognize this error :/ shame on me
and again really sorry for wasting your time although it was your hint which let me again checked my settings
i guess I watched it 100 times and only you made me look more precisely  
Now everything is working again 
big thanks and sorry for this stupid mistake
mcarosh

Pconti31, 2021-01-23:
@mcarosh Np problem.  Happy to help.  
By the way you can copy all the elements on a page to another project using the builder.  Simply use the Page layout tab, Rectangle Selection tool, select all elements on the Page, click the Copy icon, open your desired project, create the new page and click the paste icon.
Paul--

mcarosh, 2021-01-23:
Okay
i will try
Thanks
mcarosh

closed 2021-01-23 by mcarosh