ImpulseAdventure/GUIslice #356: wait for page build completely?

kellers76, 2021-01-24:
hi,
is there any way to wait for a page to build completely?

My case is that I run a startup after a reset in which I read several statuses from the EEprom and output them as text on the page.
I also output the text serially.
I see the serial text although the page is still under construction.
This has the consequence that the text is not displayed on the page.
Can I control the output of the text on the page and wait until the page has built up?

Translated with www.DeepL.com/Translator (free version)

Pconti31, 2021-01-25:
@kellers76 I'm afraid this is one of those cases where a small sample application would go a long way to helping you out.  
If you can post a zip file of a small program maybe using a const array instead of eeprom and a small number of text fields with serial output to the serial monitor that reproduces your problem I might be able to give you a solution.
I don't have the time to look over your full application so do not post that version.
Paul--

kellers76, 2021-01-26:
hello Paul,
I get it reproduced in a trimmed down original code of mine, however it's not as small as you might need.
I don't want to take up too much of your time.

What I am noticing, I have a total of 3 EEprom queries after the gslc_SetPageCur where I am reading the last statuses. If I have 2 queries active then the text comes, if I activate the 3rd query then the text doesn't come anymore....
[ske.zip](https://github.com/ImpulseAdventure/GUIslice/files/5876022/ske.zip)

Pconti31, 2021-01-26:
@kellers76 No problem.   I'll look into this tomorrow morning.
How do i run your program to reproduce the error?
Paul--

Pconti31, 2021-01-27:
@kellers76 Ok, this turned out to be easy fix.
You were overwriting the same text field.   On your third eprom read StatusBack==0 and you are replacing your previous message with spaces. So you don't see any messages.
I modified your sample to use three different text fields that you set up.
first eprom read uses m_pElemOutTxt122
second eprom read uses m_pElemOutTxt141
third eprom read uses m_pElemOutTxt141_198
I also modified button callback E_ELEM_BTN19 (BACK) to Main Page - Hauptseite
to set a few Boolean to show all three messages your demo uses.
Anyway, this should give you an idea of what was wrong.
Attached is my modified version of your sample.
Paul--
[ske.zip](https://github.com/ImpulseAdventure/GUIslice/files/5881478/ske.zip)

kellers76, 2021-01-28:
oh god, this is really embarrassing...
in german i would say, "i didn't see the forest because of the many trees".

Sure, I overwrite the text with the next query if it has the value 0...

Sorry, I must look better next time

closed 2021-01-28 by kellers76