ImpulseAdventure/GUIslice #357: compile error

coordautomation, 2021-01-30:
i am trying to compile the code in arduino IDE but i am getting following error
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Jagroop singh\Documents\Arduino\libraries\GUIslice\src/GUIslice.h:55:0,

                 from sketch\slicetrial_GSLC.h:18,

                 from C:\Users\Jagroop singh\Desktop\slicetrial\slicetrial.ino:15:

C:\Users\Jagroop singh\Documents\Arduino\libraries\GUIslice\src/GUIslice_config.h:223:6: error: #error No config selected in GUIslice_config.h. Please uncomment/select a config.

     #error No config selected in GUIslice_config.h. Please uncomment/select a config.

      ^

C:\Users\Jagroop singh\Documents\Arduino\libraries\GUIslice\src/GUIslice_config.h:224:6: error: #error For details: https:

     #error For details: https://github.com/ImpulseAdventure/GUIslice/wiki/Select-a-Config

      ^

In file included from sketch\slicetrial_GSLC.h:19:0,

                 from C:\Users\Jagroop singh\Desktop\slicetrial\slicetrial.ino:15:

C:\Users\Jagroop singh\Documents\Arduino\libraries\GUIslice\src/GUIslice_drv.h:57:4: error: #error No driver specified (DRV_DISP_*). Ensure a config is selected in GUIslice_config.h

   #error No driver specified (DRV_DISP_*). Ensure a config is selected in GUIslice_config.h

    ^

In file included from C:\Users\Jagroop singh\Desktop\slicetrial\slicetrial.ino:15:0:

sketch\slicetrial_GSLC.h:38:36: fatal error: Fonts/dosis_bold12pt7b.h: No such file or directory

 #include "Fonts/dosis_bold12pt7b.h"

                                    ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Pconti31, 2021-01-30:
@coordautomation Couple of problems here starting with either not reading the wiki or not understanding it. 
Check out the section on how to configure GUIslice API.
Assuming you are using a Arduino See:
[Configure GUIslice for Arduino](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-GUIslice-for-Arduino)
Then you pick a file inside GUIslice/config and edit it for your setup.  Say you have Arduino Mega with TFT display that uses ILI9341 with the STMPE610 touch driver.  You would choose the config/ard-adagfx-ili9341-stmpe610.h file as yours.   Then edit GUIslice/src/GUIslice_config.h and uncomment the line:
```
  //#include "../configs/ard-adagfx-ili9341-stmpe610.h"
```
You know, just like the error is telling you to do.  Then you need to calibrate your touch interface using:
```
GUIslice/examples/arduino/diag_ard_touch_calib
```
Copy and paste the output to your new config file, in my example case: ard-adagfx-ili9341-stmpe610.h

While the
```
sketch\slicetrial_GSLC.h:38:36: fatal error: Fonts/dosis_bold12pt7b.h: No such file or directory
```
Maybe due to not reading the Builder's User Guide?   You simply need to copy the dosis fonts from
GUIsliceBuilder/fonts/gfx/Dosis/BOLD to your Arduino/libraries/Adafruit-GFX/Fonts folder since Adafruit doesn't ship them as part of its package.

Good Luck!
Paul--

coordautomation, 2021-02-01:
Dear Paul,

Thanks for the nice explanation.
The compilation is done successfully.
I want to know that if we want to use Arduino with VGA Monitor and also
want to Generate the code from GUISlice for VGA Monitor then is it possible.
Do you have some example for this.

Regards

Jagroop Singh

Pconti31, 2021-02-01:
@coordautomation Sorry, I support the Builder not the API so its really an issue for @ImpulseAdventure Calvin to answer.   
I don't believe there is current support for VGA but it looks possible to add given a VGA library that uses Adafruit-GFX library interface.  
See;
[Interfacing Arduino with VGA Monitor](https://simple-circuit.com/interfacing-arduino-vga-example/)
 This link points to a library for UNO.  No idea if it supports Mega or if there are others.  I guess that is what Google is for.   If you want GUIslice API to support VGA you should open a new Enhancement request and be very specific on your hardware and software.   
Paul--

ImpulseAdventure, 2021-02-03:
@coordautomation -- although GUIslice doesn't currently integrate support for a library that drives VGA, it should be theoretically possible for one to add support for VGA libraries like the one that Paul referenced above. I won't have time at the moment, but if you were to fork GUIslice and try adding calls to VGA.h / VGA.cpp from GUIslice_drv_adagfx, I may be able to offer ideas in getting it going.

But the first step would be to confirm that you can use the VGA library directly with your Uno (if that's what you are using). It is also worth keeping in mind that the Uno has very little memory to work with, so any GUI would be very limited.

ImpulseAdventure, 2021-02-21:
Closing as the original compilation issue was addressed.

thanks

closed 2021-02-21 by ImpulseAdventure