ImpulseAdventure/GUIslice #360: ESP32 with Adafruit_FT6206  Library

Greginkansas, 2021-02-19:
Describe the bug

Can not get touch to swap X Y or flip

Device hardware

    ESP32 on the WT32-SC01 board
	https://www.cnx-software.com/2020/11/13/wt32-sc01-thin-3-5-inch-esp32-color-touchscreen-display/
    Display set in tftespi
	
#define ST7796_DRIVER 1  // for new 3.5 lcd 
#define TFT_WIDTH  480
#define TFT_HEIGHT 320
#define USE_HSPI_PORT 1
#define PIN_SDA 18  // 30
#define PIN_SCL 19  // 31
#define TFT_MISO 12
#define TFT_MOSI 13
#define TFT_SCLK 14
#define TFT_CS   15
#define TFT_DC   21
#define TFT_RST  22
#define TFT_BL   23

    GUIslice config selected:  esp-tftespi-default-ft6206.h

Have tryed  
#define GSLC_ROTATE   1
#define   TOUCH_ROTATION_FLIPX 1
gslc_TOUCH_ROTATION_FLIPYX(&m_gui, 1);
Both in Setup and in esp-tftespi-default-ft6206.h
I have never seen any remaping happening

Please confirm whether:

Display works correctly with TFT_eSPI graphicstest
Touch works with Adafruit_FT6206_Library
GUIbulider works and only needed to add a backlight on command to get test screens working

Expected behavior

Touch Y X axes to flip using   #define GSLC_ROTATE  1 in file-esp-tftespi-default-ft6206.h

Initialization messages
I did need Wire.begin(18, 19); for this board

GUIslice version [0.16.0]:
- Init display handler [TFT_eSPI] OK
- Init touch handler [FT6206(I2C)] OK

Touchscreen started.

DBG: PreRotate: x=0 y=472
DBG: Touch Press=255 Raw[0,472] Out[472,319]

DBG: PreRotate: x=43 y=78
DBG: Touch Press=255 Raw[43,78] Out[78,276]

DBG: PreRotate: x=319 y=479
DBG: Touch Press=0 Raw[319,479] Out[479,0]

Greginkansas, 2021-02-20:
Re-reading the issues thread I found a clue where to look
https://github.com/ImpulseAdventure/GUIslice/issues/48
This lead me to these lines. This hack fixes the touch for this board.

  if (m_touch.touched()) {
    TS_Point ptTouch = m_touch.getPoint();
    // FT6206 coordinates appear to have flipped both axes vs other controllers
    // - Confirmed by comments in Adafruit_FT6206 library example code:
    //   "rotate coordinate system. flip it around to match the screen."
//    m_nLastRawX = nDispOutMaxX-ptTouch.x;  //ORG line
//    m_nLastRawY = nDispOutMaxY-ptTouch.y;  //ORG line
    m_nLastRawX = ptTouch.x;  // new line for the new WT32-SC01 board 
    m_nLastRawY = ptTouch.y;  // new line for the new WT32-SC01 board 
    m_nLastRawPress = 255;  // Select arbitrary non-zero value
    m_bLastTouched = true;
    bValid = true;

  } else {
    if (!m_bLastTouched) {
      // Wasn't touched before; do nothing
    } else {
      // Touch release
      // Indicate old coordinate but with pressure=0
      m_nLastRawPress = 0;
      m_bLastTouched = false;
      bValid = true;
    }
  }

ImpulseAdventure, 2021-02-21:
Hi @Greginkansas —

Usually capacitive screens are oriented correctly versus the display, but occasionally we find some that need a different orientation.

I had created a branch that enables capacitive displays to use the GUIslice calibration sketch which should address both orientation and scaling issues.

If you would like to try it out (as opposed to applying your code workaround), please download the branch identified below and update your config file to include the following lines:

```c++
  #define DRV_TOUCH_CALIB
  // Dummy values that should be changed after first calibration:
  #define ADATOUCH_X_MIN    905
  #define ADATOUCH_X_MAX    187
  #define ADATOUCH_Y_MIN    950
  #define ADATOUCH_Y_MAX    202
  #define ADATOUCH_REMAP_YX 0
```

The branch with this added capability can be found here:

https://github.com/ImpulseAdventure/GUIslice/tree/WIP-CapCalib

thanks

Greginkansas, 2021-02-21:
Thanks for your work on this library.  I had not found that branch, how would that work on a upgrade in the future?

ImpulseAdventure, 2021-02-21:
Yes, I still need to make an adjustment to this branch, but the intention is to merge this so that the functionality will be present in all new versions going forward.

As for the outstanding change: I plan to modify it so that users don't have to enter in the "dummy" values the first time running the calibration sketch.

Greginkansas, 2021-02-22:
Thanks, good work.  The  FT6206 is a very responsive chip/screen.

closed 2021-02-22 by Greginkansas

ImpulseAdventure, 2021-02-22:
Thanks! Can you confirm whether you tried out this branch and whether it worked for your particular screen rotation?

Greginkansas, 2021-02-22:
no it did not work
GUIslice version [0.16.0.13]:
- Init display handler [TFT_eSPI] OK
- Init touch handler [FT6206(I2C)] OK

=== Touch Calibration ===

CALIB: Averaging mode: BASIC

Capture TL: X=311 Y=466 Cnt=915, Z=(255..255)
Capture TR: X=0 Y=452 Cnt=753, Z=(255..255)
Capture BL: X=308 Y=4 Cnt=632, Z=(255..255)
Capture BR: X=0 Y=2 Cnt=688, Z=(255..255)
SegDetect: Normal display (FlipX,FlipY)
CALIB: Result(adj) XMin=319 XMax=-10 YMin=468 YMax=-6

CALIB: Rotate
Capture TL (Rotated): X=0 Y=457 Cnt=726, Z=(255..255)
CALIB: Detected normal rotation

Recommended calibration settings are shown below.
- Copy and paste these lines into your GUIslice config file
  over top of the existing ADATOUCH_X/Y_MIN/MAX settings.

---
  #define ADATOUCH_X_MIN    319
  #define ADATOUCH_X_MAX    -10
  #define ADATOUCH_Y_MIN    468
  #define ADATOUCH_Y_MAX    -6
  #define ADATOUCH_REMAP_YX 0
---
put this in the config file

Also tryed 0,1,2,3 in ADATOUCH_REMAP_YX just for fun

using in portrait y and x are fliped
 
debug in --  #define DBG_TOUCH         // Enable debugging of touch-presses
  
DBG: PreRotate: x=23 y=444
DBG: Touch Press=255 Raw[23,444] Out[444,296]
DBG: PreRotate: x=23 y=444
DBG: Touch Press=255 Raw[23,444] Out[444,296]
DBG: PreRotate: x=23 y=444
DBG: Touch Press=0 Raw[23,444] Out[444,296]
Trk: (444,296) P=0 : TouchUp

DonpK, 2022-01-22:
@ImpulseAdventure @Greginkansas 

**Describe the bug:**  Touch screen points flipped horizontally and vertically relative to related elements on display. A touch button in the lower-left of screen is triggered by touching the upper-right of the screen.

**Device hardware:** 
      Moteino M0,  SAMD21G18A Cortex M0+ 32bit ARM processor,  https://lowpowerlab.com/shop/product/184
      EastRising 3.5" display,  Display: ILI9488,   Touch:   FT6236 (Capacitive),  ` ard-shld-eastrising_35_ili9488_cap.h`

**GUIslice Ver.** ` GUISLICE_VER "0.17.0"`

I ran `diag_ard_touch_calib.ino`. It produced the following settings:

 ```
  #define ADATOUCH_X_MIN    291
  #define ADATOUCH_X_MAX    -9
  #define ADATOUCH_Y_MIN    488
  #define ADATOUCH_Y_MAX    19
  #define ADATOUCH_REMAP_YX 0
```
I used these settings to replace the dummy settings in the above config. To test the results, I used a modified version of `ex04_ard_ctrls.ino` (the GUIslice (not Builder version); scaled up to fit the 3.5" screen).

The results look good. Touch points now line up nicely with corresponding elements on screen. I will try other sketches to see how they perform. I'm having some other issues using this processor/board with the GUIslice library, but I don't think their related to this config problem.

Thanks, as always, for your help Cal.