ImpulseAdventure/GUIslice #367: Teensy 4.1 Support

MkMunich, 2021-04-13:
As far I can see teensy 3.x is already supported. I got the teensy 4.1 and would like to use GUIslice. with ILI9341_t3 and its XPT2046_Touchscreen.

In the comments in section "; MCU: Teensy" it is written, that teensy 4.1 could be added by changing the line
board = teensy41
Unfortunately this leads to an error:
.platformio\packages\framework-arduinoteensy\libraries\UTFT\UTFT.cpp:81:4: error: #error "Unsupported ARM MCU!"

Best Regards

ImpulseAdventure, 2021-04-18:
Hi @MkMunich — I assume this has been superseded by #369 ? If so, perhaps we will close this ticket and we’ll continue in the discussion thread. Thanks!

MkMunich, 2021-04-18:
Hi Calvin, this is okay with me. Best Regards

closed 2021-04-18 by MkMunich