ImpulseAdventure/GUIslice #370: Using TFT_eSPI both TFT and touch driver over STM32 doesn't compile

Akrai, 2021-04-25:
So I have an STM32F411 with a tested and running TFT_eSPI config, and over the GUIslice I have set it to use both TFT and touch the TFT_eSPI driver. But compiling any example gives this output:

/home/akrai/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: libraries/GUIslice/GUIslice.c.o: in function `gslc_InitTouch':
GUIslice.c:(.text.gslc_InitTouch+0x4): undefined reference to `gslc_TDrvInitTouch'
/home/akrai/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/9.3.1-1.3/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: libraries/GUIslice/GUIslice.c.o: in function `gslc_GetTouch':
GUIslice.c:(.text.gslc_GetTouch+0x12): undefined reference to `gslc_TDrvGetTouch'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for Generic STM32F4 series.

The GUIslice config file would have:

#ifndef _GUISLICE_CONFIG_ARD_H_
#define _GUISLICE_CONFIG_ARD_H_

#ifdef __cplusplus
extern "C" {
	#endif // __cplusplus

#define DRV_DISP_TFT_ESPI
#define DRV_TOUCH_TFT_ESPI
#define GSLC_ROTATE     1
#define XPT2046_DEFINE_DPICLASS SPIClass XPT2046_spi(1);
#define TFT_ESPI_TOUCH_CALIB { 152,3738,223,3638,7 }
#define DEBUG_ERR               0
#define INIT_MSG_DISABLE
#define GSLC_FEATURE_COMPOUND       1
#define GSLC_FEATURE_XTEXTBOX_EMBED 1
#define GSLC_FEATURE_INPUT          1
#define GSLC_SD_EN    0
#define GSLC_SPIFFS_EN               0
#define GSLC_TOUCH_MAX_EVT    1
#define GSLC_SD_BUFFPIXEL   50
#define GSLC_CLIP_EN 1
#define GSLC_BMP_TRANS_EN     1
#define GSLC_BMP_TRANS_RGB    0xFF,0x00,0xFF
#define GSLC_USE_FLOAT        0
#define GSLC_DEV_TOUCH ""
#define GSLC_USE_PROGMEM      0
#define GSLC_LOCAL_STR        0
#define GSLC_LOCAL_STR_LEN    30

	#ifdef __cplusplus
}
#endif // __cplusplus
#endif // _GUISLICE_CONFIG_ARD_H_

Pconti31, 2021-04-25:
@Akrai You might get more and faster help if you mention what GUIslice config file you are using, and what IDE (Arduino, or PlatformIO) and its revision.  In your case I would think you should be using GUIslice/config/esp-tftespi-default-xpt2046_int.h to enable the internal to tft_espi touch support.  
Paul--

Akrai, 2021-04-25:
Oh right, i'm using latest Arduino IDE (1.8.13) with STM32duino 2.0, the GUIslice config is already posted, on the OP (the contents are all there)

Right, reading that config file again, i decided to copy the two touch drivers and now it works. The thing is that I have seen that before but as I saw two lines for defining touch drivers i thought the config file was "wrong" or that i had to choose one of them, so I only picked one and not the other. Now that I add the other line, everything works. Thanks!

closed 2021-04-25 by Akrai