ImpulseAdventure/GUIslice #371: M5Stack Issue

JLightMedia, 2021-04-25:
### Describe the bug
Doesn't work with M5stack board. I've uncommented the necessary line in GUIslice_config file. It successfully uploads but nothing appears on the screen. I try with the builder file and also with the demo project but the result was the same. 

### Device hardware
- *MCU*: M5stack
- *Display with link*: Build in
- *GUIslice config selected* #include "../configs/esp-shld-m5stack.h"

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* Yess
- *Touch works correctly with standalone touch library examples without GUIslice:* ???

### Expected behavior
UI elements appears on the screen 

### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.*

### Additional info
*Add any other context about the problem here.*

MickTheMechanic, 2021-04-25:
Hi, some things you can check out in the meantime before Paul or Cal come to the rescue:

- Please confirm if the M5stack library examples work. This is an important first step, as guislice is dependent on external drivers.

- Restart your device with the serial monitor open, there is some basic debugging built in to the GUISlice library and you might get a message that will help with troubleshooting.

- You can enable advance debugging in guislice->configs->esp-shld-m5stack.h. In section 5 change   ``#define DEBUG_ERR``   from ``0`` to ``2```. Right down the bottom of the file there are more debugging option that can be uncommented to further aid in troubleshooting.

- If you are using the builder to build a base sketch, make sure you have your project settings set to m5stack (E_PROJECT_OPTIONS->Target Platform->m5stack)

Hope this helps :)

ImpulseAdventure, 2021-04-25:
Hello Roman --

Which version of GUIslice are you running? (It is usually reported in the serial monitor or can be found in GUIslice_version.h)

M5stack appears to be working in my testing, so let's see if we can determine anything that might be different about your setup.

For example:
- GUIslice library v0.16.0 (from Library Manager)
- M5stack library v0.3.1 (from Library Manager)
- GUIslice_config.h modified to load "../configs/esp-shld-m5stack.h"
- Run examples/arduino/ex23_m5_input_btn

If you modify esp-shld-m5stack.h to change DEBUG_ERR to 2, open the serial monitor and upload the ex23_m5_input_btn sketch, what do you see reported to the serial monitor?

thanks!

JLightMedia, 2021-04-26:
> Hi, some things you can check out in the meantime before Paul or Cal come to the rescue:
>
> * Please confirm if the M5stack library examples work. This is an important first step, as guislice is [...]

Hello! 

1. Yes M5stack library examples and methods work well. 
2.It's serial monitor dialog
 `ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
GU	                                                                                                                                                                                                                                                                                                                                                                                                                                                      `

3. Not tried yet.
4. I use Arduino IDE

JLightMedia, 2021-04-26:
> Hello Roman --
> 
> Which version of GUIslice are you running? (It is usually reported in the serial monitor or can be found in GUIslice_version.h)
> 
> M5stack appears to be working in my testing, so let's see if we can determine anything that might be different about your setup.
> 
> For example:
> 
> * GUIslice library v0.16.0 (from Library Manager)
> * M5stack library v0.3.1 (from Library Manager)
> * GUIslice_config.h modified to load "../configs/esp-shld-m5stack.h"
> * Run examples/arduino/ex23_m5_input_btn
> 
> If you modify esp-shld-m5stack.h to change DEBUG_ERR to 2, open the serial monitor and upload the ex23_m5_input_btn sketch, what do you see reported to the serial monitor?
> 
> Mick -- thanks for the assist!! Looks like we posted at the same time :)
> 
> thanks!

3. All debugs options was unccoments:
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
GUI⸮                                                                                                                                                                                                                                                                                                                                                                                                                 
```

JLightMedia, 2021-04-26:
> Hello Roman --
>
> Which version of GUIslice are you running? (It is usually reported in the serial monitor or can be found in GUIslice_version.h)
>
> M5stack appears to be working in my testing, so let's [...]

1. GUISLICE_VER "0.16.0.21"
2. M5stack library v0.3.1
3. ../configs/esp-shld-m5stack.h was uncomment in config file.
4. Triуd but nothing. It's weird.

JLightMedia, 2021-04-26:
And again, all M5Stack lib display methods work fine. For more information, I use M5stack,  not M5stack core 2

MickTheMechanic, 2021-04-26:
try Upgrading to the latest 1.0.3 version of arduino-esp32

JLightMedia, 2021-04-26:
> try Upgrading to the latest version of arduino-esp32

I use 1.0.6 arduino-esp32 board lib. 2.0.0 it's super unstable and my sketches don't work with that alpha-1 version.

Pconti31, 2021-04-26:
@JLightMedia It's odd that you are not getting out the full GUIslice start message, `GUIslice version [0.16.0.21]:` only `GUI` Most likely it crashes with the error data still in the message buffer.  Try replacing the `DebugOut` function in your test *.ino file with this routine to flush the serial buffer on each write and maybe it will give more details.
```
// Define debug message function with flush
static int16_t DebugOut(char ch) { 
 if (ch == (char)'\n') 
  Serial.println(""); 
 else Serial.write(ch);
 Serial.flush();
 return 0; 
}
```
Paul--

Pconti31, 2021-04-26:
@JLightMedia Also uncomment the `//delay(1000);` line after the `Serial.begin(9600);`
Paul--

ImpulseAdventure, 2021-04-28:
Hi Roman --

I had been doing most of my recent M5stack testing with ESP32 board manager version 1.0.4.
I just tried upgrading to 1.0.6 and am seeing a similar early termination during startup (particularly after adding in Paul's Serial flush command above; baud rate 115200).
This suggests that some incompatibility may have crept into the latest ESP core releases.

Could you try rolling back to ESP v1.0.4 in the board manager and run "ex23_m5_input_btn"?
In parallel I will start to investigate what mods might be required to support the latest ESP32 core (if that is indeed the issue).

Thanks

JLightMedia, 2021-04-28:
> Hi Roman --
>
> I had been doing most of my recent M5stack testing with ESP32 board manager version 1.0.4.
> I just tried upgrading to 1.0.6 and am seeing a similar early termination during startup [...]

It's not compiling with 1.0.4 board version. Terminal:

```
Traceback (most recent call last):
  File "esptool.py", line 57, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports.py", line 29, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_posix.py", line 31, in <module>
  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "serial/tools/list_ports_osx.py", line 32, in <module>
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
Failed to execute script esptool
Multiple libraries were found for "SD.h"
 Used: /Users/romanvakulyuk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
 Not used: /Users/romanvakulyuk/Documents/Arduino/libraries/SD
Multiple libraries were found for "WiFiClient.h"
 Used: /Users/romanvakulyuk/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
 Not used: /Users/romanvakulyuk/Documents/Arduino/libraries/WiFi
exit status 255
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board M5Stack-Core-ESP32.
```

ImpulseAdventure, 2021-04-28:
Are you running mac OS? From some digging, I understand that ESP v1.0.4 was broken as a result of some changes in the OS (you can see workarounds in the https://github.com/espressif/esptool repo under issues 540).

That said, I can get ESP32 v1.0.6 working on my setup if I comment out the "Serial.begin(9600)" line in the ex23_m5_input_btn sketch. It appeared that the additional Serial.begin() in M5stack's begin() function was causing a lock to occur. Removing the Serial initialization in the sketch appeared to resolve the issue for me.

Can you give this a try with ESP v1.0.6 and see if it makes any further progress?
thx

JLightMedia, 2021-04-28:
> Are you running mac OS? From some digging, I understand that ESP v1.0.4 was broken as a result of some changes in the OS (you can see workarounds in the https://github.com/espressif/esptool repo under issues 540).

Yes, I use macOS Arduino IDE, that's why I use 1.0.6v

> That said, I can get ESP32 v1.0.6 working on my setup if I comment out the "Serial.begin(9600)" line in the ex23_m5_input_btn sketch. It appeared that the additional Serial.begin() in M5stack's [...]

Actually, you are absolutely right, If you use M5stack init methods, it's not necessary to init Serial yourself. Indeed, when I deleted and extra Serial init method, the example was running well on board. It's very interesting, what is the extra Serial init method in example code for? Ok, I think that information enough for developers to fix that in the future lib realize. Thank you guys for help!

ImpulseAdventure, 2021-04-30:
There are reports of a bug in **Arduino-ESP32** relating to the calling of Serial.begin() twice that weren't a problem in the past (see https://github.com/espressif/arduino-esp32/ issues/5112).

By default, GUIslice normally initializes the serial interface in the sketch so that we can provide debug logging in the user sketch. This usually works fine, but M5stack itself also tries to initialize the serial interface. In the past, this was fine, but it is now triggering the new issue in Arduino-ESP32 that we are observing here. The workaround to remove the default Serial.begin() in the GUIslice sketch is probably adequate for now until the Arduino-ESP32 issue has been resolved.

thanks

JLightMedia, 2021-04-30:
OK! Thank you for your help!

closed 2021-04-30 by JLightMedia