ImpulseAdventure/GUIslice #380: Assigning from int16_t to int8_t

en-ot, 2021-05-15:
Hello!

### Describe the bug
When trying to use big Listbox (>256 lines), lines stopped to display.
Instead of 274 (0x112) lines I see only first 18(0x12) 

There are assignments from 16-bit variables to 8-bit variables with loss of data.

file XListbox.c
lines 624...626

  int8_t nItemTop = pListbox->nItemTop;
  int8_t nItemCnt = pListbox->nItemCnt;
  int8_t nItemInd;

XListbox.h:
  int16_t         nItemCnt;       ///< Number of items in the list
  int16_t         nItemTop;         ///< Item to show at top of list after scrolling (0 is default)

I suspect there are more of similar bugs in other places.

### Device hardware
- *MCU*: ESP32
- *Display with link*: ST7789 320x240 (TTGO TM Music Albums)
- *GUIslice config selected*: esp-tftespi-default-notouch.h

ImpulseAdventure, 2021-05-18:
Thanks very much for catching this @en-ot .

I’ll look at adding a fix to support larger listbox buffers.

en-ot, 2021-05-18:
I recommend to add something like this:

XListbox.h
...
/// Callback function for Listbox feedback
typedef bool (*GSLC_CB_XLISTBOX_SEL)(void* pvGui,void* pvElem,int16_t nSel);
typedef bool (*GSLC_CB_XLISTBOX_GETITEM)(void* pvGui, void* pvElem, int16_t nItem, char* pStrItem, uint8_t nStrItemLen); 
...
  // Callbacks
  GSLC_CB_XLISTBOX_SEL pfuncXSel; ///< Callback func ptr for selection update
  GSLC_CB_XLISTBOX_GETITEM pfuncXGet;
...

XListbox.c:
...
  gslc_tsXListbox* pListbox = (gslc_tsXListbox*)gslc_GetXDataFromRef(pGui, pElemRef, GSLC_TYPEX_LISTBOX, __LINE__);
  if (!pListbox) return false;

  if (pListbox->pfuncXGet) {
    bool res = (pListbox->pfuncXGet)((void*)(pGui), (void*)(pElemRef), nItemCurSel, pStrItem, nStrItemLen);
    return res;
  }

  // Ensure user provided valid string
  if ((pStrItem == NULL) || (nStrItemLen == 0)) {
    // ERROR
    return false;
  }
...

I did it for myself, now I can use listbox for listing full microSD filesystem, passing filename strings from callback while rendering listbox.
First I change listbox size nItemCnt to files count, then set the callback function, and then Listbox receives items text from main app to gslc_ElemXListboxDraw. Also in callback function I can change other properties of each line on the fly: color f.e.

ImpulseAdventure, 2021-05-20:
Thanks — your item callback sounds like a great idea!

If you’d like to submit a PR, that would be great. Otherwise I’ll see what I can do this weekend. Thx

en-ot, 2021-05-20:
Sorry, I am new to git / github, I cannot into PR yet. And I am not sure about the quality of my code. Maybe later I'll join, but for now I can only tell the idea.

ImpulseAdventure, 2021-05-22:
Thanks @en-ot -- no worries! I'll create the fix for you and check it in.

As for the enhancement for the callback, I'll address that in a second update.
thx

ImpulseAdventure, 2021-05-23:
Fixes to support very large row counts have now been added into the main GUIslice branch.
Let me know if it works better for you.

en-ot, 2021-06-24:
I updated my project to new version of GuiSlice, and now it works as it should
Thank You

closed 2021-06-24 by en-ot