ImpulseAdventure/GUIslice #382: reaction from Toutch Event is slow because many buttons and inputs

kellers76, 2021-05-21:
Hi,
In my project, I distributed a total of almost 140 buttons as well as number inputs and text inputs on 15 pages and noticed that the reaction to a Toutch event takes on average more than 1 second and therefore too long.

I am using an ATMega2560 with a 3.95 "TFT.
The configuration that fits me is ard-shld-generic1_35_touch.h.
My memory usage is currently around 75%.

If I use an example sketch that has only a few buttons, the reaction is lightning fast.

Is it possible to speed up the process with so many switch / case options?

I thought that the CallBack method does not have to go through all the options for a toch event, but only the options that are displayed on the current page. Which page is currently displayed is not unknown.

But I cannot assess whether this would be feasible, I personally have too little experience for it.

I have already considered switching to an STM32, but I cannot estimate how much of my code is still possible there and whether my TFT will work there with one of your configurations.

It would be great if you could help me here.

best regards

stefan

Pconti31, 2021-05-22:
@kellers76 It is hard to believe that touch is the problem.  When a touch event occurs it is only checked against the currently active page, unless you also have a base page which, if so, is also checked.   So no, 150 buttons are never checked unless all are defined on one page.  As for the callback, assuming its generated by the Builder its a case statement and gcc optimizes its code.  Besides even if it didn't scanning a 150 values doesn't take a full second.   

I suspect something else is going on in your code.  

For example, are you doing any serial output, say for debugging?  Too much of that can slow down response time.  even worst if you are using 9600 baud instead of 115200.

Otherwise, I would suggest doing some profiling of your code.  Maybe using millis() function, example:
```
/*
  millis() demonstration
*/

unsigned long start, finished, elapsed;

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  Serial.println("Start...");
  start=millis();
  // some event to time inserted here....
  finished=millis();
  Serial.println("Finished");
  elapsed=finished-start;
  Serial.print(elapsed);
  Serial.println(" milliseconds elapsed");
}
```
Paul--

ImpulseAdventure, 2021-05-22:
Hi @kellers76 --

As Paul correctly mentioned, upon a touch press, the GUI only checks for contact with the buttons associated with the current page. If a popup/overlay or base page is also active, they may be checked as well.

Wrapping Paul's `millis()` function around the `gslc_Update()` call in your main loop should help determine if it is indeed the part responsible for the 1 second delay. Could you let us know how long it shows?

Thanks!

ImpulseAdventure, 2021-05-23:
Thanks for the extra details.

Given that the full update loop is 30ms, my guess is that the touch threshold are causing many of your touch presses to be filtered out (by the Adafruit touch driver), making it seem like the touch responses are sometimes sluggish.

1) If you created a simple sketch with just a single button, do you get the same touch response behavior?
2) If you try touching firmly on the display (careful not to crack it!), do the touches respond more quickly?

In your configuration, there are two values in SECTION 4D that control the touch driver sensitivity:
- `ADATOUCH_PRESS_MIN`
- `ADATOUCH_PRESS_MAX`
- You could try experimenting by adjusting these values (eg. increasing PRESS_MIN to 100, decreasing PRESS_MAX, etc.)
- Ultimately, I should probably create a new calibration sketch that calculates these values for you.

It is also possible to enable `DBG_TOUCH` in the config. This will cause your touch presses to be displayed in the serial monitor window. From there, you'll be able to see if it is actually detecting the presses right away, and what pressure readings are detected.

kellers76, 2021-05-25:
hello Calvin,

As a test, I played the example Ex02 Btn txt, the touch reacts very quickly and always cleanly.

A higher pressure does not lead to a better reaction from the touch.

I have set the minimum pressure to 100 in the config, but without a positive change.

I have also activated DBG_TOUCH and get the following message as an example:
DBG: remapX: (165,58,938,0,319)
DBG: remapY: (769,926,101,0,479)
DBG: PreRotate: x = 38 y = 91
DBG: RotateCfg: remap = 1 nSwapXY = 1 nFlipX = 0 nFlipY = 1
DBG: Touch Press = 363 Raw [769.165] Out [91.281]
Trk: (91.281) P = 363: TouchDown

DBG: Touch Cont = 363 Raw [769.165]
DBG: remapX: (165,58,938,0,319)
DBG: remapY: (769,926,101,0,479)
DBG: PreRotate: x = 38 y = 91
DBG: RotateCfg: remap = 1 nSwapXY = 1 nFlipX = 0 nFlipY = 1
DBG: Touch Press = 363 Raw [769.165] Out [91.281]
DBG: Touch End = 0 Raw [769.165] *****
DBG: remapX: (165,58,938,0,319)
DBG: remapY: (769,926,101,0,479)
DBG: PreRotate: x = 38 y = 91
DBG: RotateCfg: remap = 1 nSwapXY = 1 nFlipX = 0 nFlipY = 1
DBG: Touch Press = 0 Raw [769.165] Out [91.281]
Trk: (91,281) P = 0: TouchUp

I have now observed one effect more and more consciously over time:
my touch reacts much better when the serial cable is connected. With the cable connected, the response time is less than a second and the interface is ok to use.
when i unplug the serial cable then suddenly i have twice the reaction time and again the effect that i push and push because there is no reaction.

i suspected the power supply and instead of my pc connected a usb charger, which had no similar effect.
i also cannot measure any electrical difference with my measuring devices.

does it make a difference in the software whether a serial cable is connected or not?

stefan

Pconti31, 2021-05-25:
@kellers76 
@ImpulseAdventure Try commenting out `Serial.begin(115200);` and see if that makes a difference.
Paul--

kellers76, 2021-05-26:
hello paul and calvin,

I checked the power supply again this evening.

I have a gnd problem! if I connect externally from the usb or from an osciloscope, the touch reacts as fast as an arrow ...

it was and is not due to your software or the drivers.

please excuse me for stealing your time.

best regards

stefan

closed 2021-05-26 by kellers76

Pconti31, 2021-05-26:
@kellers76 No problem.  I'm just happy you found your problem.  
Paul--