ImpulseAdventure/GUIslice #388: Touch and Serial Monitor

boromyr, 2021-06-05:
### Describe the problem
The touchscreen and the diplay works perfectly if the serial monitor is open. When the serial monitor is closed, the touch does not work or only works on a part of the screen. I tried several examples,diag_ard_touch_test, diag_ard_touch_calib, ex26_ard_calc...

### Device hardware
- *MCU*: ESP32
- *Display*:ILI9341 2.4" resistive 
- *GUIslice config*: `esp-tftespi-default-simple`

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples without GUIslice:* Yes, it works with TFT_eSPI 
- *Touch works correctly with standalone touch library examples without GUIslice:* Yes, it works with Adafruit Touchscreen

### Additional info
The Adafruit Touchscreen library has been changed to read the values ​​from a scale to 10bit (ADC Arduino) to 12 (ADC ESP32), and the touch was calibrated with diag_ard_touch_calib.

ImpulseAdventure, 2021-06-05:
Hi @boromyr —

In the default configuration, GUIslice enables debugging messages (to report errors) and also indicate the startup initialization.

Please give the following a try:
1) Disable startup messaging by adding the following line to your config file:
- `#define INIT_MSG_DISABLE`

2) If that doesn’t avoid the issue, please try disabling all serial debug messaging by changing the existing DEBUG_ERR setting in Section 5 of your config to 0:
- `#define DEBUG_ERR 0`

Does either change the behavior for you?

boromyr, 2021-06-06:
I tried with your proposed solutions but no results, I also tried to comment on serial.begin (n) without any results. I did some research but it seems to be the first to have this problem, it's probably not a bug, but a PIN problem, since the ones I use also have a serial function.
[image]