ImpulseAdventure/GUIslice #393: Debug messages with address of invalid length

en-ot, 2021-06-20:
### Describe the bug
Hello!
In debug messages like this (in 4 different _drv files):

  char addr[6];
  GSLC_DEBUG_PRINT("DBG: DrvDrawImage() with ImgBuf address=","");
  sprintf(addr,"%04X",(unsigned int)sImgRef.pImgBuf);
  GSLC_DEBUG_PRINT("%s\n",addr);

the addr buffer is 6(why not 5?) bytes long, and %04X is trying to cut it to the length of 4 (with no success).
While on 32-bit platforms (ESP32 f.e.) the buffer must be (8+1) chars minimum, and format string is %08X
As a result I get this in terminal:

DBG: DrvDrawImage() with ImgBuf address=3F4020␟
DBG: DrvDrawImage() with ImgBuf address=3F401A=
DBG: DrvDrawImage() with ImgBuf address=3F4014[
DBG: DrvDrawImage() with ImgBuf address=3F400Ey
DBG: DrvDrawImage() with ImgBuf address=3F4008�
DBG: DrvDrawImage() with ImgBuf address=3F4002�

### Device hardware
- *MCU*: ESP32

I am not sure if it leads to bugs already, but it is potentially dangerous, because it overruns the buffer allocated in stack (local variable)

ImpulseAdventure, 2021-06-25:
Hi @en-ot --

Thanks for catching this! Yes, with 32-bit MCUs, we should be using `%08X` and expanding the storage (I'll do this for all drivers).
I am currently away at the moment but should be able to fix (and respond to your other issues) early next week.

thx

ImpulseAdventure, 2021-07-04:
Fixed in latest version in repo (0.16.1.1)

closed 2021-07-04 by ImpulseAdventure