ImpulseAdventure/GUIslice #394: ard-shld-generic1_35_touch.h for STM32?

kellers76, 2021-06-21:
### Feature Request or Enhancement
*Please describe what you would like to see GUIslice do*: ???

Hallo Paul,
I am currently using your ingenious software on an Atmega2560 with a 3.95" TFT in parallel mode.
Unfortunately I'm running out of memory on the project, so I want to transfer the code to a STM32F407 soon.

Now my question without having tested anything yet:

I currently use the config "ard-shld-generic1_35_touch.h" which works perfectly.

could you imagine to transfer the config "ard-shld-generic1_35_touch.h" to the STM32?
or is there already a config for the STM32 which would be comparable?

Translated with www.DeepL.com/Translator (free version)

### Additional context
*Add any other context or screenshots about the feature request here.*

Pconti31, 2021-06-22:
@kellers76 
@ImpulseAdventure 
Some configs for STM32 are supported depending upon your chosen tft display.  
Checkout the wiki on how to configure and test
[Configure GUIslice for STM32](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-GUIslice-for-STM32)

While I occasionally answer questions here it should be noted that GUIslice has been written by Calvin Haas (ImpulseAdventure) and this repository is his.  I have written the Builder and do have access to its repository but not this one.  I can't really help you with new drivers that's all Calvin.  If you have any questions about config after reading the wiki I'm sure Calvin can help you out.
Paul--

kellers76, 2021-06-22:
hello Paul, hello Calvin,

sorry this is embarrassing, I got it mixed up.

i would describe the hardware in more detail:
- STM32F407vgt6 board
- 3,95" TFT with ST7796S and parallel port

In the docu I found, if I understood it right, only configurations with SPI interface.

As already written it would be great if I could use the hardware

best regards
stefan

kellers76, 2021-08-10:
Hello Calvin and paul,
thanks for your advice, i have looked through the introductions again and i am one step further.
i am closing this isssue....

thanks

stefan

closed 2021-08-10 by kellers76