ImpulseAdventure/GUIslice #396: New TFT 3.5" v2.0 comes with HR2046 instead of XPT2046 and doesn't work

Akrai, 2021-07-01:
So I've been using the classic 3.5" TFT display (v1.0 label printed on the PCB), but I broke it and bought another again. Now from this store as the previous one just closed: https://es.aliexpress.com/item/32985467436.html
The picture states v1.0, but I received a v2.0
Images:
https://imgur.com/a/xvoJspn

So without modifying anything from the code, it just doesn't work. I tought that it could be just the calibration, that the new coordinates are outside of the new screen, so I just compiled the diag_ard_touch_calib but still doesn't respond to touch
I haven't found anything about the HR2046 so I don't know what to do

STM32F411 v3.0 blackpill running it
[guislice config.txt](https://github.com/ImpulseAdventure/GUIslice/files/6747045/guislice.config.txt)
[tft espi config.txt](https://github.com/ImpulseAdventure/GUIslice/files/6747049/tft.espi.config.txt)

ImpulseAdventure, 2021-07-04:
Hi @Akrai -- for your particular configuration, GUIslice is relying on TFT_eSPI for handling the touch events.
- TFT_eSPI only supports the XPT2046 touch controller.
- I assume the HR2046 should be mostly compatible with the TFT_eSPI's XPT2046 handler code.

First step would be to run the TFT_eSPI touch examples (without GUIslice), to ensure that the TFT_eSPI library is working properly with your new display. You should rule out any damage to the touch overlay and also confirm whether any configuration needs to be updated within TFT_eSPI.

Examples to look at:
- Keypad_240x320
- Test_Touch_Controller

thx

Akrai, 2021-07-04:
Oh that's right. I'll try to make the TFT_eSPI to work with the example programs. If I make those run but GUISlice doesn't work, I'll tell you here, otherwise I'm gonna close this issue for now, thanks!

closed 2021-07-04 by Akrai