ImpulseAdventure/GUIslice #400: Can't compile for ard-adagfx-hx8357-simple.h (SAMD51/Feather_M4)

LowPowerLab, 2021-07-12:
Tried several SAMD51 based boards, they all seem to produce similar errors in Arduino IDE.
The display is a [3.5" 480x320 adafruit HX8357D](https://www.adafruit.com/product/2050).
Display driver used: **ard-adagfx-hx8357-simple.h** (the ard-adagfx-hx8357-notouch.h compiles and works but can't use touch).
GUIslice version=0.16.0, latest adafruit gfx and required libraries.

Here is an excerpt from the error stack:

```
GUIslice_drv_adagfx.cpp: In function 'int gslc_TDrvGetPinMode(uint8_t)':
C:\...Arduino15\packages\adafruit\hardware\samd\1.7.2\variants\feather_m4/variant.h:66:38: error: invalid conversion from 'PortGroup*' to 'uint8_t' {aka 'unsigned char'} [-fpermissive]
   66 | #define digitalPinToPort(P)        ( &(PORT->Group[g_APinDescription[P].ulPort]) )
      |                                    ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      |
      |                                      PortGroup*
C:\Users\flexi\Desktop\Dropbox\.EE\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:2097:31: note: in expansion of macro 'digitalPinToPort'
 2097 |     uint8_t nPort           = digitalPinToPort(nPin);
      |                               ^~~~~~~~~~~~~~~~
C:\...Arduino15\packages\adafruit\hardware\samd\1.7.2\variants\feather_m4/variant.h:71:44: error: base operand of '->' is not a pointer
   71 | #define portModeRegister(port)     ( &(port->DIR.reg) )
      |                                            ^~
C:\Users\flexi\Desktop\Dropbox\.EE\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:2100:31: note: in expansion of macro 'portModeRegister'
 2100 |     volatile uint8_t *nReg  = portModeRegister(nPort);
      |                               ^~~~~~~~~~~~~~~~
C:\...Arduino15\packages\adafruit\hardware\samd\1.7.2\variants\feather_m4/variant.h:69:44: error: base operand of '->' is not a pointer
   69 | #define portOutputRegister(port)   ( &(port->OUT.reg) )
      |                                            ^~
C:\Users\flexi\Desktop\Dropbox\.EE\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:2106:31: note: in expansion of macro 'portOutputRegister'
 2106 |     volatile  uint8_t *nOut = portOutputRegister(nPort);
      |                               ^~~~~~~~~~~~~~~~~~
```

ImpulseAdventure, 2021-07-12:
Hi - this is due to a feature that may not be supported on your board. We can add detection for SAMD51 boards like we do for the Due & Nano.

In the meantime, can you try commenting out this line in GUIslice_drv_adagfx.cpp?
`#define FIX_4WIRE_PIN_STATE`

LowPowerLab, 2021-07-13:
Your suggestion makes compilation succeed. I can't get touch to work though, not sure if that has anything to do with this. I will have to check some more if all the touch pins (x+/-, y+/-) I used are defined correctly or try other pins.