ImpulseAdventure/GUIslice #401: Crashed when using TFT_ESPI Smooth Font with XKeyPadAlpha

Lucy2003, 2021-07-14:
### Describe the bug
When I click a InputText element which pre-set a string contained Chinese characters,XKeyPadAlpha is going to popup (grid has been drawn),then CPU 1 panic.

### Device hardware
- *MCU*: `ESP32`
- *Display with link*: `general 2.8'' ILI9341 SPI TFT with XPT2046 Touch (No link available)`
- *GUIslice config selected*: `esp-tftespi-default-xpt2046_int`

### Checklist to try first
- *Display works correctly with standalone display driver library examples:* YES
- *Touch works correctly with standalone touch library examples without GUIslice:* YES

### Expected behavior
Well,no crash at that time.Characters that doesn't support in XKeyPadAlpha could be removed.

### Initialization messages
GUIslice version [0.16.1.1]:
- Init display handler [TFT_eSPI] OK
- Init touch handler [TFT_eSPI(XPT2046)] OK

Lucy2003, 2021-07-15:
After remove `unloadfont` as [here](https://github.com/ImpulseAdventure/GUIslice-Builder/issues/174#issuecomment-880102497),the XKeyPadAlpha popuped normally and refresh speed returned normal.

@ImpulseAdventure Maybe you should fix this in API?

ImpulseAdventure, 2021-07-17:
Thanks @Lucy2003 for pointing this out. I’ll definitely look into integrating the `unloadfont` fix that you and Paul have been trying out. As I won’t have access to the computer for testing until later next week, I will create a branch for you to confirm that it tests ok on your setup.

thx

Lucy2003, 2021-07-17:
Hi,

I'm sorry that I may have some delay since next week.I will do the test when I'm available.

ImpulseAdventure, 2021-07-23:
Hi @Pconti31 and @Lucy2003 --

I have made some changes to the TFT_eSPI driver to address the loadFont / unloadFont optimization.
Please try out the code in https://github.com/ImpulseAdventure/GUIslice/compare/WIP401-unloadFont

Note that I decided to save the font reference pointer rather than explicitly comparing the filename.

Lucy2003, 2021-08-01:
I am sorry to tell you that I can't compile these code. The compiler output:
```
{DIR}\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp: In function 'bool gslc_DrvDrawTxtAlign(gslc_tsGui*, int16_t, int16_t, int16_t, int16_t, int8_t, gslc_tsFont*, const char*, gslc_teTxtFlags, gslc_tsColor, gslc_tsColor)':

{DIR}\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:429:40: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]

           pDriver->pvFontLast = pFont->pvFont;

                                        ^

{DIR}\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp: In function 'bool gslc_DrvDrawTxt(gslc_tsGui*, int16_t, int16_t, gslc_tsFont*, const char*, gslc_teTxtFlags, gslc_tsColor, gslc_tsColor)':

{DIR}\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:489:40: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]

           pDriver->pvFontLast = pFont->pvFont;

                                        ^
```

That's a small problem.By modifing like the line below,I can compile it.
```C++
pDriver->pvFontLast = (void*)pFont->pvFont;
```

In the end,it works well.

ImpulseAdventure, 2021-08-01:
Thanks @Lucy2003 -- that's good news...

Can you retry the latest version in the pull request here:
- https://github.com/ImpulseAdventure/GUIslice/pull/407

Lucy2003, 2021-08-05:
Ha,the updated code compiled withot any warning or error.And the code works well.

The issue had been solved.Thanks,@ImpulseAdventure .

ImpulseAdventure, 2021-08-06:
Good to hear it!
The changes have been merged now.

closed 2021-08-06 by ImpulseAdventure