ImpulseAdventure/GUIslice #406: ER-TFTM070-6 EastRising display with RA8876 controller and FT5316

joshelgin, 2021-07-23:
I need to use the below display/capactive touch screen with Arduino Mega

Link to the display and shield: https://www.buydisplay.com/7-inch-1024x600-tft-touch-shield-for-arduino-capacitive-touch-screen.

It doesn't say on that page which touch controller it is using, maybe it is FT5206 or possibly FT5316. 

I can get the display to work with configuration: /configs/ard-adagfx-ra8876-notouch.h
but when I try /configs/ard-adagfx-ra8876-ft5206.h, I get the following error in the Arduiono Compiler:
```
/libraries/FT5206-master/FT5206.h:87:70: error: braces around scalar initializer for type 'const uint8_t {aka const unsigned char}'
  const uint8_t coordRegStart[5] = {{0x03},{0x09},{0x0F},{0x15},{0x1B}};
                                                                      ^
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
```

I have tried a few different FT5316 standalone libraries, but haven't had any luck getting the capactive touch screen to work. 

https://github.com/sparkfun/SparkFun_TouchInput_Driver_FT5xx6
https://github.com/hellange/arduino/tree/master/capacitive_7in_panel

joshelgin, 2021-07-23:
I also tried the following configs:
- ard-shld-eastrising_70_ra8876_gv.h config
- ard-shld-gevino_tft.h

Received the same compile error shown above.

DonpK, 2021-07-23:
@joshelgin  You might be interested in looking at [this issue](https://github.com/ImpulseAdventure/GUIslice/issues/405) I posted a few days ago. I've also been trying to get an East Rising capacitive touch display working with GUIslice. 

GUIslice/Builder is a great project It would be nice to be able to use it with capacitive touch, which has a number of advantages over resistive touch displays.

Don

Pconti31, 2021-07-24:
@joshelgin @DonpK For the complier error you should remove the braces: { and } around single values.  Try editing FT5206.h line 87 and change to:
```
	const uint8_t coordRegStart[5] = {0x03,0x09,0x0F,0x15,0x1B};
```
Also, its useful when posting complier errors to mention the version of Arduino IDE you are using like 1.8.13 or whatever.
Paul--

Pconti31, 2021-07-24:
@JohannesWilde @DonpK Also, you can download from your link above a zip file `Arduino_Libraries-Examples_ER-TFTM070-6.zip` of samples for this display and try out the stand alone touch test file ER-TFTM070-6_Capacitive_Touch_Test.ino
Paul--

joshelgin, 2021-07-24:
I was able to get the ER-TFTM070-6_Capacitive_Touch_Test.ino to run after commenting out all the "Ra8876_Lite" display driver references. It shows the number of touch points in the serial monitor. The code that ran is below:
```
#include <stdint.h>
#include <Wire.h>
#include "SPI.h"
#include "Arduino.h"
#include "Print.h"
//#include "Ra8876_Lite.h"

uint8_t addr  = 0x38;  //CTP IIC ADDRESS

#define FT5316_WAKE 6
#define FT5316_INT   7  

const int RA8876_XNSCS = 10;
const int RA8876_XNRESET = 9;

 char stringEnd ='\0';
 char string1[] = {0xa6,0xb0,0xa4,0xe9,0xaa,0x46,0xa4,0xe8,stringEnd};  //BIG5
 char string2[] = {0xbb,0xb6,0xd3,0xad,0xca,0xb9,0xd3,0xc3,stringEnd};  //BG2312

enum {
  eNORMAL = 0,
  eTEST   = 0x04,
  eSYSTEM = 0x01
};

struct TouchLocation
{
  uint16_t x;
  uint16_t y;
};

TouchLocation touchLocations[5];

uint8_t readFT5316TouchRegister( uint8_t reg );
uint8_t readFT5316TouchLocation( TouchLocation * pLoc, uint8_t num );
uint8_t readFT5316TouchAddr( uint8_t regAddr, uint8_t * pBuf, uint8_t len );

uint32_t dist(const TouchLocation & loc);
uint32_t dist(const TouchLocation & loc1, const TouchLocation & loc2);

bool sameLoc( const TouchLocation & loc, const TouchLocation & loc2 );

uint8_t buf[30];

uint8_t readFT5316TouchRegister( uint8_t reg )
{
  Wire.beginTransmission(addr);
  Wire.write( reg );  // register 0
  uint8_t retVal = Wire.endTransmission();
  
  uint8_t returned = Wire.requestFrom(addr, uint8_t(1) );    // request 6 uint8_ts from slave device #2
  
  if (Wire.available())
  {
    retVal = Wire.read();
  }
  
  return retVal;
}

uint8_t readFT5316TouchAddr( uint8_t regAddr, uint8_t * pBuf, uint8_t len )
{
  Wire.beginTransmission(addr);
  Wire.write( regAddr );  // register 0
  uint8_t retVal = Wire.endTransmission();
  
  uint8_t returned = Wire.requestFrom(addr, len);    // request 1 bytes from slave device #2
  
[... 382 lines cut ...]
    
    ra8876lite.putHex(10,10,1,4,"n");
    ra8876lite.putHex(10,44,255,6,"n");
    ra8876lite.putHex(10,78,0xa7c8,6,"n");
    ra8876lite.putHex(10,112,0xdd11ff55,10,"n");
    
    ra8876lite.putHex(10,146,1,4,"0");
    ra8876lite.putHex(10,180,255,6,"0");
    ra8876lite.putHex(10,214,0xa7c8,6,"0");
    ra8876lite.putHex(10,248,0xdd11ff55,10,"0");
    
    ra8876lite.putHex(10,282,1,4,"#");
    ra8876lite.putHex(10,316,255,6,"#");
    ra8876lite.putHex(10,350,0xa7c8,6,"#");
    ra8876lite.putHex(10,384,0xdd11ff55,10,"#");
    
    ra8876lite.putHex(10,418,1,4,"x");
    ra8876lite.putHex(10,452,255,6,"x");
    ra8876lite.putHex(10,486,0xa7c8,6,"x");
    ra8876lite.putHex(10,520,0xdd11ff55,10,"x");
    delay(3000);
*/
  
}
```

Pconti31, 2021-07-25:
@joshelgin Did FT5206-master, which I assume is the version: [sumotoy/FT5206](https://github.com/sumotoy/FT5206), work after doing the edit to FT5206.h I suggested?

If not have you tried this touch library?  [arduino/capacitive_7in_panel/](https://github.com/hellange/arduino/tree/master/capacitive_7in_panel)
Paul--

ImpulseAdventure, 2021-07-25:
@joshelgin -- Thanks for creating a stripped down version of the EastRising capacitive touch test code.

Note that I had encountered the same compilation issue with the sumotoy/FT5206 code (not too surprising as the code has been static for several years). Here is a pull request I filed in January 2020 for the issue:
- https://github.com/sumotoy/FT5206/pull/3
- At the time, I had created a branch with a proposed fix here. Perhaps you can try loading this version.
- https://github.com/ImpulseAdventure/FT5206/tree/fix_compile
- I expect that this would be identical to the fix Paul referred to above

Alternatively, the link Paul provided to "capacitive_7in_panel" seems quite similar in operation to your stripped-down EastRising example code and is closer to something that can be linked into GUIslice (though unfortunately the touch portion isn't currently written as a standalone library).

As Paul suggested, it would be good to hear if the sumotoy/FT5206 library works (see my branch or fix above), and if not, try the "capacitive_7in_panel". If only the latter works, perhaps we can see if Helge might be interested in copying the code into a separate repository, or enabling someone else to do so, so that we can add direct support from GUIslice.

@DonpK -- I can't recall off-hand whether the above may also apply to one of the displays you were evaluating (I think your 3.5" EastRising may have used a FT6236).

joshelgin, 2021-07-27:
> * [Fix compilation of coordRegStart initializer sumotoy/FT5206#3](https://github.com/sumotoy/FT5206/pull/3)
> * At the time, I had created a branch with a proposed fix here. Perhaps you can try loading this version.

I was able to compile with the following, but got no response from the touch panel:
```
#include <SPI.h>
//#include <RA8875.h>
#include <Wire.h>
#include <FT5206.h>

//#define RA8875_CS         10
//#define RA8875_RESET      9
#define CTP_INT           7    // touch data ready for read from FT5206 touch controller

uint8_t registers[FT5206_REGISTERS];
uint16_t new_coordinates[5][2];
uint16_t old_coordinates[5][2];
uint8_t current_touches = 0;
uint8_t old_touches = 0;

//RA8875 tft = RA8875(RA8875_CS, RA8875_RESET);
FT5206 cts = FT5206(CTP_INT);

void setup()
{
  Serial.begin(9600);
  long unsigned debug_start = millis ();
  while (!Serial && ((millis () - debug_start) <= 5000)) ;
  //tft.begin(RA8875_800x480);
  cts.begin();
  //tft.setTextColor(0xFFFF,0x0000);
  Serial.println("inited...");
}

void loop()
{
  //cts.setTouchLimit(1);//from 1 to 5
  if (cts.touched()) {
    Serial.println("Touch detected");
    uint8_t i;
    uint16_t x, y;
    cts.getTSregisters(registers);
    current_touches = cts.getTScoordinates(new_coordinates, registers);
    if (current_touches < 1) return;

    /*
      uint8_t temp1 = cts.getGesture(registers);
      if (temp1 > 0){
      tft.setCursor(0,0);
      tft.print("                  ");
      tft.setCursor(0,0);
      tft.print(temp1);
      }
    */
    /*
      uint8_t temp2 = cts.getTSflag(registers);
      if (temp2 > 0){
      tft.setCursor(0,20);
      tft.print("                  ");
      tft.setCursor(0,20);
      tft.print(temp2);
      }
    */

    for (i = 1; i <= old_touches; i++) { // remove previous touches on screen
      //tft.fillCircle(old_coordinates[i-1][0], old_coordinates[i-1][1], 70, RA8875_BLACK);
    }
    for (i = 1; i <= current_touches; i++) { // mark touches on screen
      x = new_coordinates[i - 1][0];
      y = new_coordinates[i - 1][1];
      // Mark touches on screen
      //tft.fillCircle(x, y, 70, RA8875_BLUE);
      //tft.fillCircle(x, y, 50, RA8875_WHITE);
      //tft.fillCircle(x, y, 30, RA8875_WHITE);
      old_coordinates[i - 1][0] = x;
      old_coordinates[i - 1][1] = y;
    }
    old_touches = current_touches;
  }
}
```

joshelgin, 2021-07-27:
> Alternatively, the link Paul provided to "capacitive_7in_panel" seems quite similar in operation to your stripped-down EastRising example code and is closer to something that can be linked into [...]

The following code "capacitive_7in_panel"  worked! However, I had to connect pin 7 from the shield to pin 2 on the Arduino Mega, because pin 7 is not usable as an interrupt...

```
#include <SPI.h>
#include "Wire.h"
#include "FT5xx6.h"

#define CTP_INT           2    // touch data ready for read from touch panel
FT5xx6 cmt = FT5xx6(CTP_INT);

void serialDebugOutput(int nr_of_touches, word *coordinates) {
  for (byte i = 0; i < nr_of_touches; i++) {

    word x = coordinates[i * 2];
    word y = coordinates[i * 2 + 1];

    Serial.print("x");
    Serial.print(i);
    Serial.print("=");
    Serial.print(x);
    Serial.print(",");
    Serial.print("y");
    Serial.print(i);
    Serial.print("=");
    Serial.print(y);
    Serial.print("  ");
  }
}

void printRawRegisterValuesToSerial(byte *registers) {
  // print raw register values
  for (int i = 0; i < FT5xx6_NUMBER_OF_REGISTERS ; i++) {
    Serial.print(registers[i], HEX);
    Serial.print(",");
  }
  Serial.println("");
}

void setup()
{

  pinMode(8, OUTPUT); digitalWrite(8, HIGH);
  Serial.begin(9600);
  cmt.init(true);
}

word prev_coordinates[10]; // 5 pairs of x and y
byte nr_of_touches = 0;

void loop()
{
  byte registers[FT5xx6_NUMBER_OF_REGISTERS];
  word coordinates[10];

  if (cmt.touched()) {
    cmt.getRegisterInfo(registers);
    nr_of_touches = cmt.getTouchPositions(coordinates, registers);

    serialDebugOutput(nr_of_touches, coordinates);
    printRawRegisterValuesToSerial(registers);

    delay(10);
  }
}
```

Pconti31, 2021-07-28:
@joshelgin @ImpulseAdventure 
_The following code "capacitive_7in_panel" worked! (I did have to connect pin 7 from the shield to pin 2 on the Arduino Mega, because pin 7 is not usable as an interrupt.)_
Great news.  Now we need to use this code to get a library for you to use with GUIslice.
I started with code from [sumotoy/FT5206](https://github.com/sumotoy/FT5206) hacked the hell out of using capacitive_7in_panel and came up with a heavily modified version with a new example program test_no_display.ino.   Now without a display I can't test it but at least it compiles.  Let us know if it works for you.
[FT5206.zip](https://github.com/ImpulseAdventure/GUIslice/files/6896162/FT5206.zip)
Paul--

joshelgin, 2021-07-28:
> Let us know if it works for you.
> [FT5206.zip](https://github.com/ImpulseAdventure/GUIslice/files/6896162/FT5206.zip)
> Paul--

**Yes, this works!**

One caveat: I have to have the backlight of the display turned on. Otherwise, it works for a few seconds, then stops responding. However, that is an easy work around.

joshelgin, 2021-08-10:
> **Yes, this works!**
>
> One caveat: I have to have the backlight of the display turned on. Otherwise, it works for a few seconds, then stops responding. However, that is an easy work around.

Bumping this up, whenever you get a chance.

ImpulseAdventure, 2021-08-30:
Hi Josh --

Were you waiting to see if a library update could be created (to use Paul's modified FT5206), or to see if your backlight issue could be debugged?

If you're debugging the backlight, I'm afraid I might not be able to help much as it sounds like it could be specific to your hardware and less likely to be due to GUIslice.

Couple quick questions, though: how are you powering the display? (Are you using an external DC adapter? As you're likely aware, these displays generally consume more current than can be supplied via USB). Also, can you share the commands you are using to enable the backlight?

thx

joshelgin, 2021-08-31:
> Hi Josh --
> 
> Were you waiting to see if a library update could be created (to use Paul's modified FT5206), or to see if your backlight issue could be debugged?

**I'm hoping for a library update to use the display and touchscreen with GUIslice.**

I will mention again that the Arduino shield provided with the display attempts to use pin 7 as an interrupt. To make Paul's modified FT5206 work, I had to wire that connection to pin 2.

[image]

Full datasheets:
- https://www.buydisplay.com/arduino/ER-AS-RA8875_Datasheet.pdf
- https://www.buydisplay.com/download/manual/ER-TFTM070-6_Datasheet.pdf

>
> If you're debugging the backlight, I'm afraid I might not be able to help much as it sounds like it could be specific to your hardware and less likely to be due to GUIslice.
> 
> Couple quick questions, though: how are you powering the display? (Are you using an external DC adapter? As you're likely aware, these displays generally consume more current than can be supplied via USB. Although this is unlikely to be the issue as one would expect an increase in backlight draw to make things less stable). Also, can you share the commands you are using to enable the backlight?
> 
I'm powering the display with an external power supply.

Regarding the backlight, like I said, the backlight issue is an easy work around, it just has to be enabled by digital pin 8 on the Arduino: `digitalWrite(8,HIGH);`

Thanks for the help!

joshelgin, 2021-10-21:
Need any more information from me?

Thanks for all the help!