ImpulseAdventure/GUIslice #410: Corner of XPT2046 touchscreen inactive only with esp-tftespi-default-xpt2046

hyperlite2, 2021-08-03:
### Describe the bug
One of the corners of my xpt2046 touch screens is not responsive during the calibration procedure or in use. This only occurs with esp-tftespi-default-xpt2046 setup and does not occur with other xpt2046 standalone drivers (ex tft_espi or xpt2046_touchscreen)

### Device hardware
- ESP32
- https://www.buydisplay.com/qvga-2-4-inch-tft-lcd-touch-shield-320x240-serial-module-display-ili9341 
- esp-tftespi-default-xpt2046

ImpulseAdventure, 2021-08-03:
Hi @hyperlite2 —

Some touch overlays are less sensitive in the corners (and may get filtered). We can move the calibration points away from the edge by adjusting TARGET_MARGIN from 10 to 20 or so. You will find this in the following line:

https://github.com/ImpulseAdventure/GUIslice/blob/42422bd33836735f97a6d84944bc6f92da099a3c/examples/arduino/diag_ard_touch_calib/diag_ard_touch_calib.ino#L168

Does that make a difference?

hyperlite2, 2021-08-03:
Hi @ImpulseAdventure 

First, thank you so much for your library.

I can try this tomorrow, but it is only in one particular corner and the corner does not seem less sensitive when using tft_espi driver or xpt2046_touchscreen (outside of GuiSlice).

The corner seems completely dead. Its a dead spot about .25x.25in in the corner, so it isn't just on the edge. Otherwise everything is working great.

Pconti31, 2021-08-03:
@hyperlite2 I would contact buydispay and ask for a replacement.
Paul--

hyperlite2, 2021-08-03:
@Pconti31 

Maybe I am not being clear. The screen works just fine with other drivers. All corners work perfectly with other xpt2046 standalone drivers (ex tft_espi or xpt2046_touchscreen). 

I also have several of these same screens. Each of them acts the same. I only get the issue when using the esp-tftespi-default-xpt2046 setup.

Pconti31, 2021-08-03:
@hyperlite2 Ah!  That's not uncommon which is why many use this [PaulStoffregen/XPT2046_Touchscreen](https://github.com/PaulStoffregen/XPT2046_Touchscreen) instead of the TFT_eSPI internal xpt2046 driver.

If you want to try this library first turn off the TFT_eSPI internal driver inside your User_Setup.h file by commenting out TOUCH_CS.  Also, if you use the interanl driver the correct GUIslice config file is esp-tftespi-default-xpt2046_int.h but if you switch to Paul Stoffregen's you need to use esp-tftespi-default-xpt2046.h instead.

Paul--

hyperlite2, 2021-08-04:
@Pconti31 

I actually am using Paul Stoffregen's driver using esp-tftespi-default-xpt2046.h. I do have the TOUCH_CS commented out in my user_setup.h file for tft_espi as well. I only get the screen corner error when using this setup.

If I use esp-tftespi-default-xpt2046_int.h with TOUCH_CS NOT commented out, the corners all work fine. 

@ImpulseAdventure 
I get the same issue with the target margin raised. It is a strange issue. I combed through a ton of stuff trying to find the source of the issue, but haven't had any progress.

hyperlite2, 2021-08-04:
@ImpulseAdventure 
There is something I forgot to mention that may the problem clearer. If I rotate the screen orientation, the order will move. So if I move from screen orientation 1 to 3 the dead corner will flip 180 degrees. So it isn't a hardware thing.

ImpulseAdventure, 2021-08-07:
@hyperlite2 -- yes, your last observation is very important indeed!

Just to confirm:
- When you say that the corner is not responsive during calibration, I presume you mean that the calibration test doesn't get past the point where it asks you to touch in the "non-responsive" corner, since it doesn't register you pressing on the crosshair?

Can you uncomment the `DEBUG_CALIB` line from your **diag_ard_touch_calib.ino** sketch and capture a log of when you do the following:
- Ensure serial monitor log is active
- Press and hold in a "responsive" corner for 0.5 seconds
- Press and hold in a "non-responsive" corner for 0.5 seconds

Please copy the results of your test to this issue. In particular, I will be looking to see if you are getting any `AddDatapoint()` log entries when you are pressing in the non-responsive corner.

thanks

hyperlite2, 2021-08-08:
@ImpulseAdventure 

Correct, the calibration test doesn't get past the point where it asks you to touch in the "non-responsive" corner. It doesn't register when I press on the crosshair.

I ran the diag_ard_touch_calib.ino sketch with the DEBUG_CALIB line uncommented. The first two corners show values for AddDatapoint(), but the 3rd corner is the unresponsive corner and did not return aly results when pressed.

[Touch test.txt](https://github.com/ImpulseAdventure/GUIslice/files/6951346/Touch.test.txt)

ImpulseAdventure, 2021-08-11:
Thanks @hyperlite2 for the log.

I understand that the built-in TFT_eSPI touch driver appeared to be working in all corners. Were you trying out the PS driver to get better touch performance?

Two suggestions:
- Please adjust the touch pressure thresholds (`ADATOUCH_PRESS_MIN` and `_MAX`). See https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-Pressure
- In your config, try uncommenting `#define DBG_TOUCH`. Then, try the calibration again (attempting to touch the same corners as your previous test) and attach the log file.

thanks!