ImpulseAdventure/GUIslice #411: Nucleo 64 STM32F411RE with mcufriend and parallel tft st7796 problem

kellers76, 2021-08-10:
Hello Calvin,
I have a problem with my current hardware combination.

My hardware:
- STM32f4, Nucleo 64, stm32F411RE
- 3,95" TFT with ST7796, parallel Interface like http://www.lcdwiki.com/3.95inch_Arduino_Display-UNO
- stm32-adagfx-mcufriend-notouch.h and stm32-adagfx-mcufriend-simple.h

When I use the config stm32-adagfx-mcufriend-simple.h then I get no indication in the display, it stays completely white but the touch seems to work. the touch debug messages I see in the serial monitor.
If I use the config stm32-adagfx-mcufriend-notouch.h the display works fine, but of course the touch does not.
The display is plugged directly onto the board, similar to the arduino uno board, i have tested it primarily with the example ex02_ard_btn_txt.

this message with stm32-adagfx-mcufriend-simple.h:
-------------------------------------------
GUIslice version [0.15.1.9]:
- Init display handler [ADA_MCUFRIEND] OK
- Init touch handler [SIMPLE(Analog)] OK
INFO: Add elem to PageID=0: ElemID=0 Rect=(X=10,Y=50,W=300,H=150)
INFO:   Add elem to collection: ElemRef=1/2, ElemRAM=1/2, ElemId=0 (RAM)
INFO: Add elem to PageID=0: ElemID=1 Rect=(X=120,Y=100,W=80,H=40)
INFO:   Add elem to collection: ElemRef=2/2, ElemRAM=2/2, ElemId=1 (RAM)
DBG: remapX: (526,53,927,0,319)
DBG: remapY: (482,98,935,0,479)
DBG: PreRotate: x=172 y=219
DBG: RotateCfg: remap=1 nSwapXY=1 nFlipX=0 nFlipY=1
DBG: Touch Press=584 Raw[526,482] Out[219,147]
Trk: (219,147) P=584 : TouchDown

DBG: Touch End  =0 Raw[526,482] *****
DBG: remapX: (526,53,927,0,319)
DBG: remapY: (482,98,935,0,479)
DBG: PreRotate: x=172 y=219
DBG: RotateCfg: remap=1 nSwapXY=1 nFlipX=0 nFlipY=1
DBG: Touch Press=0 Raw[526,482] Out[219,147]
Trk: (219,147) P=0 : TouchUp
---------------------------------------------------

My combination works with the mcufriend stand allone driver completely, display and touch go without problems.

do you have a tip what i could try or change?

thanks
stefan

Pconti31, 2021-08-10:
@kellers76 @ImpulseAdventure Since you have your hardware working with mcufriend standalone its most likely just a minor config issue with stm32-adagfx-mcufriend-simple.h
Did you start by running GUIslice\examples\arduino\diag_ard_touch_detect? then editing or checking that inside stm32-adagfx-mcufriend-simple.h the pinouts match what this program reports.
```
  // - Definition of the pinout configuration options:
  //     ADATOUCH_PIN_YP      // "Y+": Must be an analog pin
  //     ADATOUCH_PIN_XM      // "X-": Must be an analog pin
  //     ADATOUCH_PIN_YM      // "Y-": Can be a digital pin
  //     ADATOUCH_PIN_XP      // "X+": Can be a digital pin

  // Pin connections from diag_ard_touch_detect:
  #define ADATOUCH_PIN_YP     PA6
  #define ADATOUCH_PIN_XM     PA7
  #define ADATOUCH_PIN_YM     PB7
  #define ADATOUCH_PIN_XP     PB6
```
Then did you run diag_ard_touch_calib.ino? and copy its value to stm32-adagfx-mcufriend-simple.h?
```
  // Calibration settings from diag_ard_touch_calib:
  //
  // - A number of example calibration settings for common MCUFRIEND
  //   shields have been provided in SECTION 4C, each marked with their
  //   corresponding MCUFRIEND ID. However, note that these example
  //   calibration values may not provide accurate touch tracking, therefore
  //   using the diag_ard_touch_calib utility is strongly recommended.
  #define ADATOUCH_X_MIN    150
  #define ADATOUCH_X_MAX    910
  #define ADATOUCH_Y_MIN    108
  #define ADATOUCH_Y_MAX    934
  #define ADATOUCH_REMAP_YX 0    // Some touch controllers may swap X & Y coords
```
Paul--

Pconti31, 2021-08-10:
@kellers76 @ImpulseAdventure zip your config files and I'll desk check them
Paul--

kellers76, 2021-08-10:
hallo paul,

in the zip is the GUIslice_config.h and the 2 configs stm32-adagfx-mcufriend-notouch.h and stm32-adagfx-mcufriend-simple.h.

i hope you find my error.

thank you already

thanks

stefan

[my_config.zip](https://github.com/ImpulseAdventure/GUIslice/files/6963534/my_config.zip)

Pconti31, 2021-08-10:
@kellers76 Also, please upgrade to GUIslice 16.  I can't debug 15 at this point.  What version of Adafruit_TouchScreen are you using?  The latest is 1.1.2

kellers76, 2021-08-11:
hello paul,

I have updated guislice to 0.16.0.
unfortunately there was no improvement...

the version of adafruit_touchscreen was already the 1.1.2.

in the attachment you can find my Buton_simple.ino from mcufriend.

thank you

stefan
[button_simple.zip](https://github.com/ImpulseAdventure/GUIslice/files/6966841/button_simple.zip)

Pconti31, 2021-08-11:
@kellers76 @ImpulseAdventure The code for GUIslice_drv_adagfx.cpp has gotten to the point with its many #if #elif and #endifs to be almost impossible to follow.  Nevertheless, I may have found your problem.   Try this version by simply replacing the one in GUIslice/src and let us know if it now works.
Remember this is the 0.16.0 version and no other.
Paul--

[GUIslice_drv_adagfx.zip](https://github.com/ImpulseAdventure/GUIslice/files/6968354/GUIslice_drv_adagfx.zip)

Pconti31, 2021-08-11:
@kellers76 @ImpulseAdventure Sorry, I screwed up the fix.  Should never touch code before my morning coffee.  Here is the real fix test routine.  Please disregard the previous version.
Paul--

[GUIslice_drv_adagfx.zip](https://github.com/ImpulseAdventure/GUIslice/files/6968804/GUIslice_drv_adagfx.zip)

kellers76, 2021-08-11:
hi paul,

you are my hero!

it works!
I can't believe it, your morning coffee was more than worth it.

thank you 

stefan

kellers76, 2021-08-11:
hi paul,

unfortunately i rejoiced too soon...

in further tests, with under the attempt to calibrate the touch I have found that when pressing the touch the screen suddenly turns white and partly also then no longer responds.

Pconti31, 2021-08-11:
@kellers76 @ImpulseAdventure I'm not surprised.  The fix I made to GUIslice_drv_adagfx.cpp was to reset the XM and YP pins to OUTPUT and set HIGH after doing a getPoint() call.   I assume something like that is also needed for the calibration code.  Sorry,  the calibration code is not something I'm willing to dive into.  

Also, we will still need Calvin to look at my fix and determine how to repair GUIslice long-term.  My version is just a quick hack.

One suggestion is while you wait for Calvin is to try MCUFRIEND's calibration and copy the numbers to your _simple.h config file.  Maybe that will work for you.

In anycase, I have gone as far as I can without your hardware..
Paul--

kellers76, 2021-08-12:
hi paul,

yes i am fine at the moment, my main concern is to get the display running with the mcufriend driver on a STM32F4 processor.
This is thanks to your quick help succeeded!

If Calvin then still looks at the thing to get it stable in the long term then I can go this way with the processor further.

Thanks for your quick help.

thanks 

stefan

ImpulseAdventure, 2021-08-13:
Thanks Paul for taking a stab at this!

@kellers76 -- as you may be aware, the Adafruit TouchScreen library doesn't play nicely with certain displays (particularly those that share pins with the touch overlay). Unfortunately the issue is outside of GUIslice, generally leading to a white screen when touch activity is enabled concurrently with the display. Nonetheless, GUIslice has attempted to provide a workaround for the incompatibility (effectively "cleaning up" pin state), but this has only been enabled for a certain set of MCUs.

By default, the workaround is disabled for STM32, but we can activate a partial workaround that forces the pins (similar to the intent of Paul's code above) instead of using pin save/restore mechanisms (used in the full workaround).

I have updated the ADAGFX driver file to add the partial workaround for STM32 variants. Please give it a try and let me know if you observe anything different. Note that I am not able to test this with the same hardware as you, so the best I can do is confirm that it appears to compile properly.

Please try using the attached `GUIslice_drv_adagfx.cpp` file by overwriting the same file in your Arduino/libraries/GUIslice/src folder, assuming you are using the [most recent version](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source) of GUIslice from GitHub (should be about `0.16.1.3`).

As mentioned, there is no guarantee this will work, but it may be worth a try. Some 4-wire touch-display combinations have been difficult to get going due to the specific pins in use.

thx
[GUIslice_drv_adagfx.zip](https://github.com/ImpulseAdventure/GUIslice/files/6980340/GUIslice_drv_adagfx.zip)

kellers76, 2021-08-16:
Hello calvin,

you are ingenious!

it works without any problems!

I have tried several examples and my application and can operate the touch cleanly without the screen turning white or bringing other errors.

Many many thanks to you!

have a nice evening
stefan

ImpulseAdventure, 2021-08-28:
stefan -- So glad to hear it worked well!
I have now merged these changes into the main GUIslice branch.

closed 2021-08-28 by ImpulseAdventure