ImpulseAdventure/GUIslice #414: Text Element not updating

deadlocklogic, 2021-09-04:
When I try to update the Text, it is not updating:
```c
  char acTxt[5];
  snprintf(acTxt, 5, "%s", "1234");
  gslc_ElemSetTxtStr(&m_gui, m_pElemText1, acTxt);
```

Even when I try it in the `setup`.
I am calling in the loop `gslc_Update(&m_gui);` just saying.

### Device hardware
- *MCU*: Uno
- *Display with link*: Custom (a mix/ a cheap chinese device)
- *GUIslice config selected*: custom

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* yes
- *Touch works correctly with standalone touch library examples without GUIslice:* yes

### Expected behavior
The textelement not updating its text

Pconti31, 2021-09-05:
@deadlocklogic The FAQ covers this.

You first need to first assign storage to the text element inside the property view otherwise the text area is write protected. Without extra storage the text string is a constant in FLASH not a variable in SRAM.
 
Optionally you should assign a meaningful names to the ENUM and ElementRef not keep the defaults that the Builder generates so you and others can better understand your App.
As an example say I have a global counter m_nCount.  I rename the Text UI Element E_TXT_COUNT and the ElementRef as m_pElemCnt. 

| Key                   | Text$2               |
|-----------------------|----------------------|
| **ENUM**                  | **E_TXT_COUNT**     |
| X                     | 70                   |
| Y                     | 60                   |
| Width                 | 44                   |
| Height                | 12                   |
| **ElementRef**            | <span style="color: red;">**m_pElemCnt**</span>           |
| Font                  | BuiltIn(1x)->5x8pt7b |
| Text                  |                      |
| UTF-8?                | false                |
| **External Storage Size** | <span style="color: red;">**7**</span>                    |
| Text Alignment        | GSLC_ALIGN_MID_LEFT  |

Paul--

deadlocklogic, 2021-09-05:
## **First thanks for your quick reply!**
I much appreciate open source projects and the effort put behind them.
Yeah you are right about naming, but this is a very simple evaluation example (I am a disciplined developer and really appreciate good naming/convention/style...).
So this is what I was missing `External Storage Size`.
I assume the escape character `'/0'` is not counted.

## By the way, few questions/suggestions:
1- Why not allow a custom sub-configuration tested at the top if available:
For instance I am using a cheap Chinese LCD which has few driver defects (in return for the cost), it has `readID()` always return 0 so I have to manually replace it with `0x9341` when calling `begin`, (it is based on `DRV_DISP_ADAGFX_ILI9341_8BIT`), so instead of this:
https://github.com/ImpulseAdventure/GUIslice/blob/5d4d79de382d845c6c4ba8eb4e03cbb474763d3a/src/GUIslice_drv_adagfx.cpp#L653
I have to call:
```c
uint16_t identifier = 0x9341; 
```
So I suggest inserting this option at the top in  [GUIslice_drv_adagfx.c](https://github.com/ImpulseAdventure/GUIslice/blob/master/src/GUIslice_drv_adagfx.cpp):
```c
#ifndef  DRV_DISP_ADAGFX_FIXED_ID
    #define DRV_DISP_ADAGFX_FIXED_ID(_DISP) _DISP.readID()
#endif
```
then 
```c
uint16_t identifier = DRV_DISP_ADAGFX_FIXED_ID(m_disp); 
```
<br>
Also the touches are flipped so instead of calling:

https://github.com/ImpulseAdventure/GUIslice/blob/5d4d79de382d845c6c4ba8eb4e03cbb474763d3a/src/GUIslice_drv_adagfx.cpp#L3011-L3012

I have to call:
```c
nOutputX = map(nInputX, pGui->nTouchCalXMin, pGui->nTouchCalXMax, nDispOutMaxX, 0);
nOutputY = map(nInputY, pGui->nTouchCalYMin, pGui->nTouchCalYMax, nDispOutMaxY, 0);
```
So I suggest this:
```c
#ifdef  DRV_DISP_ADAGFX_TOUCH_FLIP_X
    nOutputX = map(nInputX, pGui->nTouchCalXMin, pGui->nTouchCalXMax, nDispOutMaxX, 0);
#else
    nOutputX = map(nInputX, pGui->nTouchCalXMin, pGui->nTouchCalXMax, 0, nDispOutMaxX);
#endif
#ifdef  DRV_DISP_ADAGFX_TOUCH_FLIP_Y
    nOutputY = map(nInputY, pGui->nTouchCalYMin, pGui->nTouchCalYMax, nDispOutMaxY, 0);
#else
    nOutputY = map(nInputY, pGui->nTouchCalYMin, pGui->nTouchCalYMax, 0, nDispOutMaxY);
#endif
```
<br>
Also in my case, the `width/height` can be queried directly using `width()/height()`:
I suggest to test when `ILI9341_TFTWIDTH` or its equivalent if defined, and when not replace with `width()/height()`.

_In my case:_
My config file will add: 
```c
#define DRV_DISP_ADAGFX_FIXED_ID(_DISP) 0x9341
#define DRV_DISP_ADAGFX_TOUCH_FLIP_X
#define DRV_DISP_ADAGFX_TOUCH_FLIP_Y
// I won't define ILI9341_TFTWIDTH/ILI9341_TFTHEIGHT
```
2- Did you consider using Qt for the builder application? I think it would fit better for such application. Noting the builder app in it current form is not mature enough:
  - Undo/Redo are glitchy.
  - Dialogs on actions, can be replaced with a console or a statusbar.
  - Add a reset default action for each property
  - Auto suggest names for added elements
  - And many more improvements can be added, like a a preview button (open a live test windows) etc...

Pconti31, 2021-09-05:
@deadlocklogic First, I feel the need to point out I have nothing to do with the API so Calvin (@ImpulseAdventure)  will need to answer for any improvements.

You ask why not use Qt for the UI.  The answer requires a little history. A few years back I have a few household projects that required using an Arduino. I found various API's for TFT displays and found them lacking.  I liked GUIslice but quickly decided that it required more effort to use than I liked.  Calvin will get mad at me if I go any deeper.  Being a typical software engineer I would rather write code than spend time doing anything I view as boring.  So I slapped together the Builder in a few days using Java.   After a few projects it evolved into something I thought others might want to use so I threw it over the wall to Calvin who agreed to post it.   

The first versions looked quite different and was harder to use than the current version.  I have added new features over the years but at its heart its still a quick and dirty program that has certain design limitations.   

Frankly, this is free software and it is still useful if limited.  So I'm not currently interested in doing any major work on it.  It's open source, so if anyone wants a web based implementation, conversions to C++, or  complex new features I say have add it!  We will be more than willing to accept pull requests.

As to your first question, maybe you noticed in your config file the line:
```
  //#define DRV_DISP_ADAGFX_MCUFRIEND_FORCE  0x9481 // MCUFRIEND write-only displays may need to force an ID value
```
Paul--

deadlocklogic, 2021-09-05:
I see, I thought you are the main maintainer of the project (never paid enough attention).
Yes me too as a software engineer, I always prefer to write code than reviews/modify others.
A web version would be perfect, never though of this actually but of course as any open source free project, it requires free time. I may possibly have my take on something like that with Qt.
Still good project overall, even thought room for improvement exist.

By the way, as a quick review I saw that strings are store as ref counted or copied depending on a flag, didn't dig any deeper for the time being in hope I get some insights. Is the overall memory allocation model good enough to avoid fragmentation?

Pconti31, 2021-09-05:
@deadlocklogic The strength and weakness of GUIslice API is that memory fragmentation is impossible unless you do memory allocs in your code.  That is because all UI storage must be pre-allocated at compile time.  Thus the reason for arrays like m_asFont, m_asPage, m_asPage1Elem, m_asPage1ElemRef, etc...  Setting the memory up is one of the book keeping challenges to hand coding GUIslice API.  The Builder takes care of these detail for you. 

The string flags are generally to deal with write protected FLASH based strings vs RAM based text.  There are a few places in the API code where the strings may need to be copied under the covers so to speak.

Paul--

deadlocklogic, 2021-09-05:
Thank you so much man, I appreciate your help. If I face any new issue I will open a new thread. For now I will close this as the issue is resolved.

closed 2021-09-05 by deadlocklogic