ImpulseAdventure/GUIslice #415: pgmspace.h error?

4contact, 2021-09-19:
Hello, I'm a newbie here and also on arduino so please take it easy if I'm asking a dumb question.
First off, I really really love this project. It helps so much for a non-coder like me.
I appreciate it, thank you!

I've been testing a TFT touchscreen with Arduino-Due(32bit arm) mainly because of the huge lack of srams of other series.
But I keep getting this error message below while compiling it.
Btw I'm using codes generated by GUIslice Builder, a single button on a single page, set "Use Flash API" false for testing.
I think I've installed all the required libraries, honestly I have no idea what else I'm missing to edit.
I've tried pretty much the same codes with arduino-mega with different GUIslice_config settings before, and it worked just fine.
Can anybody help? Thanks in advance.

TFT info: 3.2inch TFT Touch screen(ili9341, XPT2046, SPI connection)
Uncommented driver-set: due-adagfx-ili9341-urtouch.h

/////////////////////////////////////////////////////////////////////////////////////////////////////////////
Error message:
Arduino: 1.8.11 (Windows 7), Board: "Arduino Due (Programming Port)"

C:\Users\owner\Documents\Arduino\libraries\GUIslice-master\src\GUIslice.c:62:26: fatal error: pgmspace.h: No such file or directory

     #include <pgmspace.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board Arduino Due (Programming Port).
/////////////////////////////////////////////////////////////////////////////////////////////////////////////

Pconti31, 2021-09-19:
@4contact Well, I don't know if its the 'correct' config file or not.  The GUIslice developer @ImpulseAdventure Calvin will need to help you with that.  I might have gone with ard-adagfx-ili9341-xpt2046.h and used [PaulStoffregen/XPT2046_Touchscreen](https://github.com/PaulStoffregen/XPT2046_Touchscreen) library but if you have the urtouch library working for you that's fine by me.   

Remember that in all cases before using GUIslice to get the example standalone programs provided  with your chosen drivers Adafruit_GFX and in your case URTOUCH working with your hardware first then and only then try GUIslice.

In any case I think you can turn off this error by going into your config file, due-adagfx-ili9341-urtouch.h in your case and simply editing this line:
```
#define GSLC_USE_PROGMEM      0  // was 1 
```
Paul--

4contact, 2021-09-19:
**@Paul**
Thank you for your fast reply!!

I just edited "GUIslice_config.h" only like below. I touched nothing else.
  .
  .
  .
  //#include "../configs/ard-adagfx-ra8876-ft5206.h"
  //#include "../configs/ard-adagfx-ssd1306-notouch.h"
  //#include "../configs/ard-adagfx-st7735-notouch.h"
  //#include "../configs/due-adagfx-ili9225-notouch.h"
  //#include "../configs/due-adagfx-ili9341-ft6206.h"
  #include "../configs/due-adagfx-ili9341-urtouch.h"   // <------for due
  //#include "../configs/due-adagfx-ra8875-urtouch.h"
  .
  .
  .

oh...is it possible to use ard-adagfx-ili9341-xpt2046.h for Arduino-Due as well? when I tried with Mega, I used ard-adagfx-ili9341-xpt2046.h. and yes it worked for Mega. But I thought I couldn't use ard-xxx for Due or thought it might possibly cause some internal issues or i don't even know what I'm talking about. well, I will test it this way, too.
Actually ili9341 example programs worked good but urtouch samples have never worked before...

Also I changed the parameter 1 to 0 as you indicated, then I get a different error. maybe I just should go with ard-adagfx-ili9341-xpt2046.h...

Error message:
Arduino: 1.8.11 (Windows 7), Board: "Arduino Due (Programming Port)"

C:\Users\owner\Documents\Arduino\libraries\GUIslice-master\src\GUIslice_drv_adagfx.cpp:85:29: fatal error: ILI9341_due.h: No such file or directory

     #include <ILI9341_due.h>

                             ^

compilation terminated.

exit status 1
Error compiling for board Arduino Due (Programming Port).

4contact, 2021-09-21:
**@Paul**
Finally it worked out with due-adagfx-ili9341-urtouch.h using your method to set 0 for GSLC_USE_PROGMEM.
It seems like I never had ILI9341_due library installed. so I installed it then succeeded compiling it. 
I thought I had it but it was ILI9341_Fast library. that was so dumb of me...
Thank you so much for your kind advice!!

closed 2021-10-03 by ImpulseAdventure