ImpulseAdventure/GUIslice #416: Raspberry pi

nawazawanswl, 2021-09-19:
hi,
with Puttty
i compiled test sdl with sdl1, with sucess and it was running fine with fb0 on HDMi 7 inch display on Raspberry Pi zero w
on debian buster official pi os.
after that i compiled the linux examples of guislice and , compilation was done without any errors.
But...............the the images shown on the 7 inch display are very smal just like 128x64 pix.
could you help me in this regard Plz
the library is awsome , but i am doing something wrong 
Plz help , i am stuck.

nawazawanswl, 2021-12-30:
long ago i posted this issue but unfortunatly .....................NO RESPONSEEEEEEEEEEEEEEEEEEEEEE.

ImpulseAdventure, 2022-01-16:
Hi @nawazawanswl -- my sincere apologies for the lengthy delay. My computer and electronics gear were all packed away for what was supposed to be a short move / reno, but it took two months longer than expected :( I still don't have access yet to my gear but would like to help get it sorted out for you.

I don't think I have an equivalent HDMI display to test with on hand, but perhaps we could rule out a couple options first, in case you're still looking for a solution:
- Can you clarify for me if it appears that the full GUI content is squished into the corner of the screen or just cut off in the corner?
- Through raspi-config, does changing the resolution make a difference in what you see?
- Please have a look at https://stackoverflow.com/questions/22891235/how-to-change-screen-resolution-of-raspberry-pi in case some of the suggestions there enable you to configure a suitable resolution.

thx

nawazawanswl, 2022-01-16:
Hi,
hope you will get full free soon.
it starts from top left corner and image is full but the size is around 128x64 and touch is working at 128 x 2 and 64 x 2, away from actual image.
the config file has no effect over.
i have also checked by changing the display RES in pi env.
i tried everything according to my knowledge and what was available on the net and in the last i could not get it so i posted you.
it is not displayed in cut form
it is displaying everything.
if required i will again flash and send you images.
hope you can help, in the start i was happy that i am already using GUISLICE so it will be nice to implement it on Pi
but a struuggle of one week did not yeild any thing.
i hope you can help... waiting

nawazawanswl, 2022-02-02:
hi,
any help regarding my question.....?