ImpulseAdventure/GUIslice #425: How to explicitly set I2C pin numbers for touch controller and flip touch X, Y directions?

Kabron287, 2021-12-15:
### Device hardware
- *MCU*:ESP32.
- *Display with link*: ST7796 3.5" capacitive FT6336
- *GUIslice config selected*: ??? `esp-tftespi-default-ft6206.h`
- *Board*: WT32-SC01
### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with*: diag_ard_touch_test.ino
- *Touch works correctly with standalone touch library examples without GUIslice:* YES

### Observing behavior
*Touch works **partially correctly** with GUIslice*
I.e. X and Y direction are opposite

### Additional info
*Touch controller FAILS to initialize out of the box*

I have to add:
#define SDA 18
#define SCL 19
#include <Wire.h>

at the beginning of example and:

Wire.begin(SDA, SCL)
in setup to make it work. 

Question:
Where could I setup I2C SCL/SDA pins?

ImpulseAdventure, 2022-01-16:
Hi @Kabron287 -- in your setup, GUIslice is using Adafruit_FT6206 which uses the default `Wire.begin()`. Perhaps we can add an override option for you.

Could you try inserting your `Wire.begin()` call into `GUIslice_drv_tft_espi.cpp` and let me know if that helps you initialize your touch controller? Perhaps near line 131, for example: (sorry can't test at the moment)
```c++
#elif defined(DRV_TOUCH_ADA_FT6206)
  const char* m_acDrvTouch = "FT6206(I2C)";
  // Always use I2C
  Wire.begin(SDA, SCL);
  Adafruit_FT6206 m_touch = Adafruit_FT6206();
  #define DRV_TOUCH_INSTANCE
```

As for the flipping of touch directions, please try running example **diag_ard_touch_calib** and let me know if the calibration / flipping looks correct at the end of the test?

thanks

derlaft, 2022-01-18:
Hi @ImpulseAdventure. For some reason it does not compile if it is put over there:

```
Compiling .pio/build/wt32-sc01/lib604/GUIslice/GUIslice_drv_tft_espi.cpp.o
.pio/libdeps/wt32-sc01/GUIslice/src/GUIslice_drv_tft_espi.cpp:131:3: error: 'Wire' does not name a type
   Wire.begin(SDA, SCL);
   ^
*** [.pio/build/wt32-sc01/lib604/GUIslice/GUIslice_drv_tft_espi.cpp.o] Error 1
```

Why - not clear to me (too deep into macro?).

However, touch controller initializes if I put the same line into `void setup()`. 

Cannot check `diag_ard_touch_calib` as my dislay is currently not working properly with GUISlice :) Will test when/if I can make it work.

derlaft, 2022-01-19:
Ok, got screen working.

* `diag_ard_touch_calib` works
* However, red line appears not where it is supposed to be
    - It is flipped both horizontally and vertically
    - So If I touch bottom left corner, line appears in top right one.

ImpulseAdventure, 2022-01-21:
Thank you for sharing your PlatformIO config!

Regarding the touch calibration, when you run the diag_ard_touch_calib sketch:
- The first stage asks you to touch the crosshairs in each corner, then attempts to recalibrate your display and finally begin a test stage based on your calibration.
- It sounds like the test stage does not appear to be showing a successful calibration.

Please try the following:
- Uncomment the line DEBUG_CALIB in the sketch (near line 51)
- Start the serial monitor
- Run the calibration sketch, and hold down for at least a half second in each corner on the crosshairs. Then in the test phase press on the top left then bottom right corners of your display
- Attach the output of your calibration (serial output window log) to this issue

thanks

derlaft, 2022-01-22:
> It sounds like the test stage does not appear to be showing a successful calibration.

It does show "Rotation detected OK":

[image]

> Please try the following:

[Calibration output.](https://gist.github.com/derlaft/b6cf9f1e7297ad494b857e9ad5662253)

[Video of result](https://user-images.githubusercontent.com/852606/150650325-81f07fde-ece7-4765-bb79-75fa59f0085c.mp4).

> Then in the test phase press on the top left then bottom right corners of your display

Also shows red dot in the wrong corner.

GeorgeIoak, 2022-03-10:
@derlaft I'm running a similar setup as you. You shared your platformio.ini so thank you but I wanted to confirm that in the end you did just add the SDA and SCL defines and the "custom" `Wire.begin(SDA, SCL)` in setup.h?

How did you handle screen rotation and touch calibration, are those solved?

derlaft, 2022-03-10:
>  but I wanted to confirm that in the end you did just add the SDA and SCL defines and the "custom" Wire.begin(SDA, SCL) in setup.h?

I just added that to `void setup()`

> How did you handle screen rotation and touch calibration, are those solved?

I think the problem is that one of the axis is flipped, while the other is not. Here how it looks in lgvl code:

https://github.com/derlaft/WT32-SC01-lgvl-platformio/blob/103851c24808ed87c27915a87c252715794d9e13/src/main.cpp#L34-L35

In current GUIslice bindings both axis are flipped. Therefore GUIslice cannot figure out rotation.

I will try to make a patch a bit later to confirm this theory.

GeorgeIoak, 2022-03-10:
For another project I tested using https://github.com/DustinWatts/FT6236 with TFT_eSPI and that seemed to work. I also had to create a custom I2C but I did it a little differently:
```
#define I2C_Freq 100000
#define I2C_SDA 33
#define I2C_SCL 27

TwoWire I2C_0 = TwoWire(0);
```
and then in setup()
```
I2C_0.begin(I2C_SDA, I2C_SCL, I2C_Freq);

  if (!ts.begin(TOUCH_THRESHOLD, I2C_SDA, I2C_SCL))
  {
    Serial.println("[INFO]: Capacitive touch FAILED to start");
  }
  else
    Serial.println("[INFO]: Capacitive touch started!");
```
Not sure if that will solve anything but maybe Dustin's driver works "better"

GeorgeIoak, 2022-03-10:
I can confirm that using that method works but the touch not aligned correctly. I've got other issues in that the screen is not drawn correctly so I'll have to open another thread to discuss that.

derlaft, 2022-03-11:
Strange, drawing works fine for me. Do you have some different hardware?

GeorgeIoak, 2022-03-11:
Yes, the ILI9488 based screen using ER-TFT035-6 screen w/ CTP and my own custom board which worked fine with just TFT_eSPI

GeorgeIoak, 2022-03-12:
OK, strange, after trying a million different combinations I finally got the screen to draw correctly and touch kind of works. It might be a flipped axis as you suspect but I'm not sure. Do you know how we can enable sending the touch point to the serial log? That should be enough to confirm what it is sensing and what needs to change.

ASMotionLab, 2022-03-12:
Hi guys,
I'm too in the same boat, with the WT32-SC01.
It kept crashing on boot until I added the custom I2C initialisation.
My touch is also flipped.
Added the calibration parameters to my ``esp-tftespi-default-ft6206.h``, as shown in the calibration sketch, but they seem to be ignored - touch still flipped.

GeorgeIoak, 2022-03-14:
@ImpulseAdventure I don't have time to check other rotations but I found that the following changes correct the touch location for the default rotation

https://github.com/ImpulseAdventure/GUIslice/blob/ca3a62cac9395d2cb91a81c7607afa9d064a3aa7/src/GUIslice_drv_tft_espi.cpp#L1748
```
//m_nLastRawX = nDispOutMaxX-ptTouch.x;
//m_nLastRawY = nDispOutMaxY-ptTouch.y;
 m_nLastRawX = ptTouch.x; 
 m_nLastRawY = ptTouch.y; 
```

I didn't dig deeper but it appears that somewhere else in the code there's some flipping going on because when you turn on `DBG_TOUCH` and `DBG_LOG` you'll see these statements (which are after my changes)
```
DBG: PreRotate: x=144 y=249
DBG: Touch Press=0 Raw[144,249] Out[249,175]
Trk: (249,175) P=0 : TouchUp
```
Hopefully tomorrow I'll have some time to test in different screen rotations and see if this change works for all orientations.

GeorgeIoak, 2022-03-14:
Decided to come back and check the results with different rotations and everything works as expected with the above changes. I touched near the upper left corner and lower right corner of the screen and recorded the reported touch locations.

[image]

The screen I'm using has a 320x480 resolution and it's the raw panel. The orientation in the table is with the FFC on the right side

ImpulseAdventure, 2022-03-14:
Hi @ASMotionLab and @GeorgeIoak !

If you are using a capacitive touch display and observing that the horizontal or vertical axes are flipped, then you should be able to adjust the orientation with the calibration sketch. Many capacitive displays do not require any orientation adjustments, but we do come across some that do appear to be flipped.

When running the calibration sketch, the new calibration settings will be reported at the end, and these settings can be pasted into your updated configuration file. First step is to confirm that the touch detection / orientation looks right at the end of the calibration sketch (during the "test" phase after touching each of the crosshairs).

**IMPORTANT**: when forcing calibration on these capacitive displays, we need to set `DRV_TOUCH_CALIB` in the config file so that the override settings will take effect. By default, this will generally not appear in the default configs.

As this "capacitive calibration" feature was only seen on a smaller number of displays, it was not as well documented and the setting easily overlooked.

To avoid this, I have posted a new update to the calibration sketch that now recommends adding `DRV_TOUCH_CALIB` to the config for these displays.

Does that help the axis orientation?

Please note that I will be away on vacation for the next two weeks, so please anticipate some delays in my responses.

thanks,
Cal

ASMotionLab, 2022-03-14:
@ImpulseAdventure Thank you for walking us through it: now that I did a search for the term, I can see DRV_TOUCH_CALIB in a lot of the drivers.
After adding this line to my driver `esp-tftespi-default-ft6206.h` and the calibration values (just rounded them up to the screen dimensions) it's now working perfectly for me 🙌
```
  // -----------------------------------------------------------------------------
  // Touch Handling
  // -----------------------------------------------------------------------------

  #define DRV_TOUCH_CALIB 1
  #define ADATOUCH_X_MIN    320
  #define ADATOUCH_X_MAX    0
  #define ADATOUCH_Y_MIN    480
  #define ADATOUCH_Y_MAX    0
  #define ADATOUCH_REMAP_YX 0
```
[image]

ImpulseAdventure, 2022-03-15:
Excellent — thank you for the confirmation, @ASMotionLab 

@GeorgeIoak — I will close the issue now assuming we have addressed the request for orientation adjustment on the capacitive display. If you still see an issue with this, let me know and I will reopen it.

thanks!

closed 2022-03-15 by ImpulseAdventure

blackcloud2025, 2024-05-04:
use this in your main.cpp it works fluent.

#define SCR 8
class LGFX : public lgfx::LGFX_Device
{

  lgfx::Panel_ST7796 _panel_instance;

  lgfx::Bus_SPI _bus_instance;

  lgfx::Light_PWM _light_instance;

  lgfx::Touch_FT5x06 _touch_instance;

public:
  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config(); // Get the structure for bus configuration.

      // SPI bus settings
      cfg.spi_host = VSPI_HOST; // Select the SPI to use ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
      // * With the ESP-IDF version upgrade, VSPI_HOST and HSPI_HOST descriptions are deprecated, so if an error occurs, use SPI2_HOST and SPI3_HOST instead.
      cfg.spi_mode = 3;                  // Set SPI communication mode (0 ~ 3)
      cfg.freq_write = 27000000;         // SPI clock when sending (up to 80MHz, rounded to 80MHz divided by an integer)
      cfg.freq_read =   6000000;           // SPI clock when receiving
      cfg.spi_3wire = false;             // set to true if receiving on MOSI pin
      cfg.use_lock = true;               // set to true to use transaction lock
      cfg.dma_channel = SPI_DMA_CH_AUTO; // Set the DMA channel to use (0=not use DMA / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=auto setting)
      // * With the ESP-IDF version upgrade, SPI_DMA_CH_AUTO (automatic setting) is recommended for the DMA channel. Specifying 1ch and 2ch is deprecated.
      cfg.pin_sclk = 14; // set SPI SCLK pin number
      cfg.pin_mosi = 13; // Set MOSI pin number for SPI
      cfg.pin_miso = -1; // set SPI MISO pin number (-1 = disable)
      cfg.pin_dc = 21;   // Set SPI D/C pin number (-1 = disable)

      _bus_instance.config(cfg);              // Apply the settings to the bus.
      _panel_instance.setBus(&_bus_instance); // Sets the bus to the panel.
    }

    {                                      // Set display panel control.
      auto cfg = _panel_instance.config(); // Get the structure for display panel settings.

      cfg.pin_cs = 15;   // Pin number to which CS is connected (-1 = disable)
      cfg.pin_rst = 22;  // pin number where RST is connected (-1 = disable)
      cfg.pin_busy = -1; // pin number to which BUSY is connected (-1 = disable)

      // * The following setting values ​​are set to general default values ​​for each panel, and the pin number (-1 = disable) to which BUSY is connected, so please try commenting out any unknown items.

      cfg.memory_width = 320;  // Maximum width supported by driver IC
      cfg.memory_height = 480; // Maximum height supported by driver IC
      cfg.panel_width = 320;   // actual displayable width
      cfg.panel_height = 480;  // actual displayable height
      cfg.offset_x = 0;        // Panel offset in X direction
      cfg.offset_y = 0;        // Panel offset in Y direction
      cfg.offset_rotation = 0;
      cfg.dummy_read_pixel = 8;
      cfg.dummy_read_bits = 1;
      cfg.readable = false;
      cfg.invert = false;
      cfg.rgb_order = false;
      cfg.dlen_16bit = false;
      cfg.bus_shared = false;

      _panel_instance.config(cfg);
    }

    {                                      // Set backlight control. (delete if not necessary)
      auto cfg = _light_instance.config(); // Get the structure for backlight configuration.

      cfg.pin_bl = 23;     // pin number to which the backlight is connected
[... 128 lines cut ...]
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);

    /* Initialize the input device driver */
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register(&indev_drv);

    ui_init();

    Serial.println("Setup done");
  }

}

void loop()
{
  lv_timer_handler(); /* let the GUI do its work */
  delay(5);

}

blackcloud2025, 2024-05-04:
use this code... 

#define SCR 8
class LGFX : public lgfx::LGFX_Device
{

  lgfx::Panel_ST7796 _panel_instance;

  lgfx::Bus_SPI _bus_instance;

  lgfx::Light_PWM _light_instance;

  lgfx::Touch_FT5x06 _touch_instance;

public:
  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config(); // Get the structure for bus configuration.

      // SPI bus settings
      cfg.spi_host = VSPI_HOST; // Select the SPI to use ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
      // * With the ESP-IDF version upgrade, VSPI_HOST and HSPI_HOST descriptions are deprecated, so if an error occurs, use SPI2_HOST and SPI3_HOST instead.
      cfg.spi_mode = 3;                  // Set SPI communication mode (0 ~ 3)
      cfg.freq_write = 27000000;         // SPI clock when sending (up to 80MHz, rounded to 80MHz divided by an integer)
      cfg.freq_read =   6000000;           // SPI clock when receiving
      cfg.spi_3wire = false;             // set to true if receiving on MOSI pin
      cfg.use_lock = true;               // set to true to use transaction lock
      cfg.dma_channel = SPI_DMA_CH_AUTO; // Set the DMA channel to use (0=not use DMA / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=auto setting)
      // * With the ESP-IDF version upgrade, SPI_DMA_CH_AUTO (automatic setting) is recommended for the DMA channel. Specifying 1ch and 2ch is deprecated.
      cfg.pin_sclk = 14; // set SPI SCLK pin number
      cfg.pin_mosi = 13; // Set MOSI pin number for SPI
      cfg.pin_miso = -1; // set SPI MISO pin number (-1 = disable)
      cfg.pin_dc = 21;   // Set SPI D/C pin number (-1 = disable)

      _bus_instance.config(cfg);              // Apply the settings to the bus.
      _panel_instance.setBus(&_bus_instance); // Sets the bus to the panel.
    }

    {                                      // Set display panel control.
      auto cfg = _panel_instance.config(); // Get the structure for display panel settings.

      cfg.pin_cs = 15;   // Pin number to which CS is connected (-1 = disable)
      cfg.pin_rst = 22;  // pin number where RST is connected (-1 = disable)
      cfg.pin_busy = -1; // pin number to which BUSY is connected (-1 = disable)

      // * The following setting values ​​are set to general default values ​​for each panel, and the pin number (-1 = disable) to which BUSY is connected, so please try commenting out any unknown items.

      cfg.memory_width = 320;  // Maximum width supported by driver IC
      cfg.memory_height = 480; // Maximum height supported by driver IC
      cfg.panel_width = 320;   // actual displayable width
      cfg.panel_height = 480;  // actual displayable height
      cfg.offset_x = 0;        // Panel offset in X direction
      cfg.offset_y = 0;        // Panel offset in Y direction
      cfg.offset_rotation = 0;
      cfg.dummy_read_pixel = 8;
      cfg.dummy_read_bits = 1;
      cfg.readable = false;
      cfg.invert = false;
      cfg.rgb_order = false;
      cfg.dlen_16bit = false;
      cfg.bus_shared = false;

      _panel_instance.config(cfg);
    }

    {                                      // Set backlight control. (delete if not necessary)
      auto cfg = _light_instance.config(); // Get the structure for backlight configuration.

      cfg.pin_bl = 23;     // pin number to which the backlight is connected
[... 128 lines cut ...]
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);

    /* Initialize the input device driver */
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register(&indev_drv);

    ui_init();

    Serial.println("Setup done");
  }

}

void loop()
{
  lv_timer_handler(); /* let the GUI do its work */
  delay(5);

}