ImpulseAdventure/GUIslice #429: Compile Error with SAMD21 Cortex M0 processor

DonpK, 2022-01-18:
I'm getting this error in this section GUIslice.c when compiling:

```
#if (GSLC_USE_PROGMEM)
  #if defined(__AVR__)
    #include <avr/pgmspace.h>
  #else
    #include <pgmspace.h>
  #endif
#endif
```

This error is: 
      "pgmspace.h: No such file or directory"

Processor:
 SAMD21 Cortex M0+ based processor  (ATSAMD21G18A microcontroller)    ( https://lowpowerlab.com/guide/moteino/moteinom0/ )
Eastrising 3.5" capacitive touch display. /configs/ard-shld-eastrising_35_ili9488_cap.h config file.

Display works okay on Arduino AVR processors.

Suggestions?
  Thanks

Pconti31, 2022-01-19:
@DonpK I don't believe SAM MCUs support flash with special api's.  They simply use flash as part of the main memory space so try setting inside your guislice/config/xxxx.h file:
```
#define GSLC_USE_PROGMEM 0
```
Paul--

Pconti31, 2022-01-20:
@DonpK the only thing that can cause the prmspace.h file not found is having GSLC_USE_PROGMEM 1 so if it's not your config must be another. Double check entry inside guislice/stc/gui_slice_config.h and also delete all files inside guislice/config but yours and re compile. Still problem copy all output from Arduino ide and zip up and post.

DonpK, 2022-01-20:
@Pconti31     I found GSLC_USE_PROGMEM 1 in the Misc section of the display config. Changed it to 0 and the compile error disappeared.
Thanks,
    Don

ImpulseAdventure, 2022-01-21:
Thank you @Pconti31 for helping identify the config fix!

@DonpK — I’m thinking there might be value in creating a wiki page for device-specific config details. There are a few devices which require overriding parameters like PROGMEM, so it may be helpful to have a place where we can start capturing it. In the case of shields, we would simply add a new config file, but for custom-wired breakout boards, a general guidance page to collect this info could be the way to go.

DonpK, 2022-01-21:
@ImpulseAdventure @Pconti31 Yes, information for custom-wired boards could be useful. Given the the potentially large combination of custom boards/displays, organizing it could be a challenge and in such a way so that the user easily find it.  

Although I did an "Issues" for my progmem problem, I came up with nothing. Later, after Paul had answered it, I did another search with other terms and found that Paul and others had already answered it.

closed 2022-01-25 by DonpK