ImpulseAdventure/GUIslice #443: Support M5Stack Core 2

lewcianci, 2022-02-11:
### Feature Request or Enhancement
Please add support for M5Stack Core 2 (https://www.cnx-software.com/2020/09/01/m5stack-launches-core2-esp32-iot-development-kit-with-2-inch-touch-display/#:~:text=M5Stack%20has%20just%20launched%20an,motion%20sensor%20and%20a%20microphone.)

### Additional context
The display is ILI9342C, and the touchscreen driver is FT6336U.

Considering I have the board, I'm happy to perhaps work it out myself and submit a PR for the configs directory. Is there a guide on how to make your own config file, what to look for, etc?

ImpulseAdventure, 2022-03-07:
Hi @lewcianci --

Apologies for the delay as I was tied up with a significant work deadline over the past month or so.
Yes, it would be great to add support for M5stack core 2. I only have the original M5stack hardware, so wouldn't be able to test directly, but would be happy to assist you in whatever is needed to get v2 working.

My guess is that the modifications to GUIslice might be quite minor, and these might amount to additional `IFDEF` checks within GUIslice_drv_m5stack (if needed) along with a new config file.

If you could post an update of what you find when compiling GUIslice example ex01 with the `esp-shld-m5stack.h` configuration selected, that would be great.

thanks

ymnb, 2022-03-18:
Hi @lewcianci and @ImpulseAdventure,

I am also interested in this issue and I am working on a personal project using M5Tough.  How about switching the library to M5Unified instead of using M5Stack for supporting multiple multiple M5 series?

If it would be possible, I guess we can support M5Stack/M5Core2 and M5Tough with small changes like  : https://github.com/ymnb/GUIslice/tree/feature/enable_touch_screen_with_M5Unified. I have tested it on M5Tough, I don't have other hardware..

Thanks,
(I haven't work on OSS project before, so I wasn't sure if I can randomly drop the comment to issue.. If this message is confusing, please feel free to hide/delete it)