ImpulseAdventure/GUIslice #444: Request: Add Circle Element

nadjet09, 2022-02-15:
Describe the solution you'd like
The draw circle is in the GUI library already exist, may be need to be linked to an element in the GuiSlice.

Additional context
additionally if the circle element can be added and still be able to set as oval shape or circle shape.

ImpulseAdventure, 2022-03-07:
This seems like a reasonable request to add. We could probably use additional primitive drawing elements.

As there is a question about future Builder support for such elements, I'd like to ask @Pconti31 for this thoughts -- particularly if he envisioned that this is something that might be practical to support in the Builder later as well.

thanks

Pconti31, 2022-03-07:
@nadjet09 @ImpulseAdventure Yes, no problem supporting in the builder.
However, it might be useful to know how it will be used? 
Just draw a circle or does it need callbacks like Box?
Paul--

nadjet09, 2022-03-07:
Not sure about the callback function necessity, I guess it would be better to be able to control it's colors, fillable or not, and visibility, as well as the frame thickness.
Additionally, if fill could be set false, the frame thickness could be used to draw circle instead.

Pconti31, 2022-03-08:
@nadjet09 @ImpulseAdventure 
I would assume a circle frame would just be a circle with no line thickness feature,  same way Box works, filled rectangle or just outline.   I am open to adding line thickness, if needed.

I'm fine with no callback support but Cavin has been adding it to more and more Elements that traditionally didn't support it so I thought he might want to support it at the start.  The Builder can support either, easily.
Paul--