ImpulseAdventure/GUIslice #450: Adding gslc_ElemXGraphReset() to XGraph broke the code base.

Pconti31, 2022-03-08:
The new  gslc_ElemXGraphReset() has syntax errors, also new errors inside gslc_ElemXGraphCreate().
Attached is a repaired version of XGraph.<h c>
```
C:\Users\Paul\Documents\Arduino\libraries\GUIslice\src\elem\XGraph.c:136:52: error: expected ')' before ';' token

     gslc_ElemXGraphReset(pGui, &(pGui->sElemRefTmp);

C:\Users\Paul\Documents\Arduino\libraries\GUIslice\src\elem\XGraph.c:139:3: error: expected ';' before '}' token

   }

   ^

C:\Users\Paul\Documents\Arduino\libraries\GUIslice\src\elem\XGraph.c: In function 'gslc_ElemXGraphReset':

C:\Users\Paul\Documents\Arduino\libraries\GUIslice\src\elem\XGraph.c:233:3: error: 'pXData' undeclared (first use in this function)

   pXData->nBufCnt         = 0;

   ^

C:\Users\Paul\Documents\Arduino\libraries\GUIslice\src\elem\XGraph.c:233:3: note: each undeclared identifier is reported only once for each function it appears in

C:\Users\Paul\Documents\Arduino\libraries\GUIslice\src\elem\XGraph.c:243:10: error: 'pBuf' undeclared (first use in this function)

   memset(pBuf,0,nBufMax*sizeof(int16_t));

          ^

C:\Users\Paul\Documents\Arduino\libraries\GUIslice\src\elem\XGraph.c:243:17: error: 'nBufMax' undeclared (first use in this function)

   memset(pBuf,0,nBufMax*sizeof(int16_t));

                 ^
```

I restored gslc_ElemXGraphCreate() to it's original version and changed gslc_ElemXGraphReset() from:
```
void gslc_ElemXGraphReset(gslc_tsGui* pGui,gslc_tsElemRef* pElemRef) {

  pXData->nBufCnt         = 0;
  pXData->nPlotIndStart   = 0;

  // Default scale is
  // - Each data point (buffer row) gets 1 pixel in X direction
  // - Data value is directly mapped to height in Y direction
  pXData->nPlotValMin   = 0;
  pXData->nPlotValMax   = pXData->nWndHeight;
  pXData->nPlotIndMax   = pXData->nWndWidth;

  memset(pBuf,0,nBufMax*sizeof(int16_t));
}
```
To:
```
void gslc_ElemXGraphReset(gslc_tsGui* pGui,gslc_tsElemRef* pElemRef) {

  gslc_tsXGraph*  pBox;
  gslc_tsElem*    pElem = gslc_GetElemFromRef(pGui,pElemRef);
  pBox = (gslc_tsXGraph*)(pElem->pXData);

  pBox->nBufCnt  = 0;
  pBox->nPlotIndStart   = 0;

  // Default scale is
  // - Each data point (buffer row) gets 1 pixel in X direction
  // - Data value is directly mapped to height in Y direction
  pBox->nPlotValMin   = 0;
  pBox->nPlotValMax   = pBox->nWndHeight;
  pBox->nPlotIndMax   = pBox->nWndWidth;

  memset(pBox->pBuf,0,pBox->nBufMax*sizeof(int16_t));
}
```
While I was at it I also addressed issue`#439` div-by-zer on XGraph with (still) 0 points by changing line 320 from:
```
  pBox->nPlotIndStart  = pBox->nPlotIndStart % pBox->nBufMax;
```
To:
```
  pBox->nPlotIndStart = pBox->nBufMax ? (pBox->nPlotIndStart % pBox->nBufMax) : 0;
```
[XGraph.zip](https://github.com/ImpulseAdventure/GUIslice/files/8207136/XGraph.zip)

Paul--

Pconti31, 2022-03-08:
Crap, Once again i shouldn't answer these before my morning coffee.
void gslc_ElemXGraphReset() needs this added to the end:
```
  // Set the redraw flag
  // - As we are clearing the buffer, force a full redraw
  gslc_ElemSetRedraw(pGui,pElemRef,GSLC_REDRAW_FULL);
```
Attached is another version with this fix.
[XGraph.zip](https://github.com/ImpulseAdventure/GUIslice/files/8207843/XGraph.zip)
Paul--

ImpulseAdventure, 2022-03-10:
Hey @Pconti31 -- thanks very much for the alert regarding PR #442 . As I didn't have my regression environment running quite yet (just now unpacking from our reno), I had made an incorrect assumption about the submission's testing status.

Thanks for the revisions to XGraph that you attached to this issue! I intend to go through your proposed changes and merge those into the main as soon as I can rerun my regression.