ImpulseAdventure/GUIslice #454: CST816 touchscreen  +ST7735 display

Ash8356, 2022-03-12:
### Feature Request or Enhancement
*Please describe what you would like to see GUIslice do*: ???
we are using CST816 as touchscreen driver , its a capacitive touch screen , i am using arduino board , which example can i follow to Test the same . display used is ST7735 
### Additional context
*Add any other context or screenshots about the feature request here.*
Display is tested with ST7735  driver .

Pconti31, 2022-03-12:
@Ash8356 @ImpulseAdventure GUIslice doesn't supply drivers.  It's a feature rich set of additional API's built on top of other drivers with a Low Code Builder that allows you to create User Interfaces with a drag and drop interface.

I don't believe GUIslice currently supports the CTS816S chip so this needs to be an enhancement request first.

If you already have your hardware working with say Adafruit libraries you will need to wait for Calvin's response about software support.  You can test your display without touch using GUIslice config file `ard-adagfx-st7735-notouch.h` if you are using Adafruit GFX library.

The Adafruit drivers I found are:
[An Arduino library for the CST816S capacitive touch screen IC](https://github.com/fbiego/CST816S)
[Adafruit-ST7735-Library](https://github.com/adafruit/Adafruit-ST7735-Library)

If you have examples with these or other drivers working then it would help Calvin (@ImpulseAdventure)  out to know what drivers you are using, version numbers and the exact Arduino board you are using, Arduino Mega? Other?

Paul--

Pconti31, 2022-03-12:
@Ash8356 @ImpulseAdventure I should also mention that you can write your own interface to GUIslice for the CTS816S chip by following example `ex16_ard_touch_hnd`.  I have done that myself and its not that hard.
Paul--

florinbaciuu, 2023-11-14:
Hello . I write a driver for that touch module. Look at [PR](https://github.com/ImpulseAdventure/GUIslice/pull/521). Tested . Worked perfect.