ImpulseAdventure/GUIslice #471: How to remove a corner around a picture in the "Image Button" control?

tankist-git-2, 2022-08-31:
I wanna use picture as a button (typical scenario - change the page by clicking a picture). This functional is completely done by using "Image Button" control, but, unfortunatelly, there are a corner around my picture in this mode (see the attached picture). I can't remove that by GUIslice builder settings. If I try to use "Image" component instead there is no built-in functional as in the "Image Button" control.

[image]

So I wonder if it possible to remove that corner now? If not, could you add a setting to remove it in the future?

Pconti31, 2022-08-31:
@tankist-git-2 If you post a zip file of a small sample program that just displays your icon with the problem corner I will see if I can debug the issue.
Paul--

tankist-git-2, 2022-08-31:
Here is a project
[3.zip](https://github.com/ImpulseAdventure/GUIslice/files/9462015/3.zip)

Pconti31, 2022-08-31:
@tankist-git-2 Turns out to be an easy fix. Simply set "Transparent?=true" inside the Builder on each of your images.
Paul--

tankist-git-2, 2022-09-01:
This works. Thank you for the quick response!

closed 2025-02-17 by ImpulseAdventure