ImpulseAdventure/GUIslice #472: There is an error "characters outside the range" when I add my own font

tankist-git-2, 2022-09-01:
I try to add my own VLW font into a GUIslice Builder project but is was unsuccessful.

Device hardware:
ESP32 + tft_espi

I follow the instruction on the user_guide to add my own font. There are steps that I've done:
1) Run Processing -> Create font utility

[image]

2) Choose font (Aharoni-Bold) and size

[image]

As you can see, I select ALL fonts inside the file including non-Latin ones.
3) I put ahronbd.ttf (original) file into c:\Program Files (x86)\GUIsliceBuilder\fonts\vlw\ folder
4) I put AharoniBold18.vlw (result) file into /data folder of my GUIslice Builder project
5) I put TEXT control to the page in "Simulated TFT panel" area of GUIslice Builder
6) There are settings of this TEXT control:

[image]

7) I try to see a character map of my new font but it looks ridiculous: it shows 220 chars only

[image]

8) If I try to set non-latin string into "Text" property of my TEXT control I've gpt an error:

[image]

What is wrong?

Pconti31, 2022-09-01:
@tankist-git-2 You need to help me out to help you.  Post a zip file containing ahronbd.ttf, and your builder_fonts.json.
By the way, in the future please post Builder issues inside my Repository since this one is Calvin's GUIslice API.  It would just make tracking Builder bug fixes easier for me.
Paul--

Pconti31, 2022-09-01:
@tankist-git-2 I found the problem.  The character was stopping scanning of your font at SHORT.MAX_VALUE which is 32767.  Your font ends at 65535.  I have fixed this in release 17.b12.  You will need to use this builder_font.json file which has you font added when using this new release.
[builder_fonts.zip](https://github.com/ImpulseAdventure/GUIslice/files/9472631/builder_fonts.zip)
Paul--

tankist-git-2, 2022-09-02:
I installed a new version of GUIslice Builder and repeat steps to install new font:
1) Replaced builder_fonts.json by yours
2) Put ahronbd.ttf into GUIsliceBuilder\fonts\vlw\ directory
3) Put AharoniBold18.vlw into GUIsliceBuilder\fonts\vlw\data\ directory

Unfortunatelly, new font does'n appear in a font selection combobox.
Also I see in new version you added GUIsliceBuilder\fonts\vlw\flash\ directory which contains header files. I have no  header file for my new font, could it be a problem?

Pconti31, 2022-09-02:
@tankist-git-2 No, it's not the data folder that's just so people don't have to go through and create the vlw files for the fonts I ship.

Of course, I tested this version with your font and json file before releasing so it works fine, at least for me :)

In looking over your your posted images the fonts being listed are for Adafruit_GFX library not TFT_eSPI so I would guess that's your problem.

As the README file says I did a code refactoring to better describe our options for IDE and graphics libraries.  Where before you picked either arduino, m5stack, tft_espi, or linux and I assumed IDE of Arduino IDE you now must pick your IDE then your Graphics lib.  so in your case your Project Options tab should look something like this:

[image]

Hopefully, this will address your issue.  If not, I'll build a special version that logs everything so we can see why its not working for you.  Just let me know your development platform, windows, etc...

Paul--

tankist-git-2, 2022-09-05:
O, sorry, that was really my fault.
But actually even when I choose the right platform the problem still remains for me...

[image]

I see 260 symbols only and still can't input non-latin letters

tankist-git-2, 2022-09-06:
I tried your file but result is the same. I'm a bit confused. Both files (I will refer to them as "old" - what included in release and "new" - what you sent to me in the previous post) shows cyrillic fonts if I use Windows standard utility. But I tried to use https://fontdrop.info/ website and I can't see any other font besides latin although they report about 36 languages support. Just in case I attach both files. Maybe you will find out what the problem is or even offer another font for your program with built-in cyrillic font.

[ahronbd_new.zip](https://github.com/ImpulseAdventure/GUIslice/files/9495052/ahronbd_new.zip)

[ahronbd_old.zip](https://github.com/ImpulseAdventure/GUIslice/files/9495056/ahronbd_old.zip)

[image]

Pconti31, 2022-09-06:
@tankist-git-2 Sorry, I have no idea what windows utility you re using but windows character map doesn't show any of these characters.   As for Cyrillic font support I already ship with Google's notosansbold support built-in for vlw.  Noto Sans is an unmodulated (“sans serif”) design for texts in the Latin, Cyrillic and Greek scripts.
Paul--

Pconti31, 2022-09-07:
@tankist-git-2 I posted a question about this on the Microsoft forum:
[Russian fonts display confusion](https://answers.microsoft.com/en-us/windows/forum/windows_10-wintop_language/russin-fonts-display-confusion/6d807143-0ac9-4fa0-b089-ad2de2035528)
Paul--

Pconti31, 2022-09-07:
@tankist-git-2 So I got the answer from the Microsoft forum as why you are seeing the Cyrillic characters when you font doesn't contain them.
[Answer](https://answers.microsoft.com/en-us/windows/forum/windows_10-wintop_language/russin-fonts-display-confusion/6d807143-0ac9-4fa0-b089-ad2de2035528?messageId=afe8bee9-0a22-48ab-a77d-51a9f0238bf3)
See the following in particular **Font Fallback**.
[Font technology](https://docs.microsoft.com/en-us/globalization/input/font-technology)

Again you can use the already installed NotoSanBold.vlw font shipped with the builder or maybe take a look at one of these:
[The 30+ Best Cyrillic Supported Fonts for Graphic Design & Branding](https://justcreative.com/cyrillic-supported-fonts/)

Paul--

tankist-git-2, 2022-09-08:
Thank you very much for the investigation. Well, I will use existing font and maybe find another one with cyrillic and try to convert it again.