ImpulseAdventure/GUIslice #477: Configuration Issue: ESP32+stmpe610       ERROR: "pin detection"

Bwanna, 2022-09-29:
### Discussed in https://github.com/ImpulseAdventure/GUIslice/discussions/476

<div type='discussions-op-text'>

<sup>Originally posted by **Bwanna** September 26, 2022</sup>
Hi,
I have an ESP32 with an ST7789+ STMPE610 Touch and am getting the following error msg when compiling the "diag_ard_touch_detect" sketch:
```
D:\Arduinofiles\libraries\GUIslice-master\examples\arduino\diag_ard_touch_detect\diag_ard_touch_detect.ino:31:4: error: #error "Only DRV_TOUCH_ADA_SIMPLE supports pin detection"
   #error "Only DRV_TOUCH_ADA_SIMPLE supports pin detection"
    ^~~~~
D:\Arduinofiles\libraries\GUIslice-master\examples\arduino\diag_ard_touch_detect\diag_ard_touch_detect.ino:35:2: error: #error "Pin detection not supported on ARM or ESP platforms"
 #error "Pin detection not supported on ARM or ESP platforms"
  ^~~~~

exit status 1

Compilation error: #error "Only DRV_TOUCH_ADA_SIMPLE supports pin detection"
```
Both the ST7789 (SPI) and the STMPE610 (I2C) are working using Adafruit sketches/drivers. However, this error is appearing here.

**In "GUISLICE_CONFIG.h"  the following:**
  #include "../configs/esp-tftespi-default-stmpe610.h"

**In esp-tftespi-default-stmpe610.h" the following:** 
  #define ADATOUCH_I2C_HW 1  // Touch controller via hardware I2C (uses ADATOUCH_I2C_ADDR)
  #define ADATOUCH_SPI_HW 0  // Touch controller via hardware SPI (uses ADATOUCH_PIN_CS)
  #define ADATOUCH_SPI_SW 0  // Touch controller via software SPI [not yet supported]

Any assistance would be greatly appreciated.

Can someone explain where the 'pin detection' is configured?  which file?  Have searched and unable to locate.

Thanks!</div>

Pconti31, 2022-09-29:
@Bwanna Sorry but there is nothing to help with.  This is an unimplemented feature for your  touch driver as the error clearly states.   You must already know your pins for touch or the Adafruit sketches would not work for you.  You simply need Calibration of your touch driver.  The x and y min and maxes that should be entered into your version of /esp-tftespi-default-stmpe610.h.  For that simply run the diag_ard_touch_calib.ino sketch and place the numbers it gives you into the config file.   I generally use the serial monitor an just cut and paste from that.
Paul--

Bwanna, 2022-09-30:
> @Bwanna Sorry but there is nothing to help with.  This is an unimplemented feature for your  touch driver as the error clearly states.   You must already know your pins for touch or the Adafruit [...]

First  thanks for the feedback, but there is possibly a misunderstanding here ... 

**This is a question about 'which' file would the "Pin Detection" be configured?**.  

I'm unable to locate this in any of the GUISlice files. 

Thank you.

Pconti31, 2022-09-30:
@Bwanna Yes, I can see there is a misunderstanding.   So I'll try again.  There is no **PIN Detection**.
The pin detection feature is only needed to debug  and report the set of pins currently wired to a 4-wire resistive touch display. It exists only inside the test program diag_ard_touch_detect.ino and nowhere else.

These 4-wire displays unlike the  touch driver for STMPE610 generally comes with almost no documentation so this program is needed for them.  

Since you have stated above that your Adafruit_STMPE610 test program works I assume its TouchTest.ino and you have correctly defined your touch interface at the top of this routine so you can simply edit your  "../configs/esp-tftespi-default-stmpe610.h" file with the correct information.

Now how to config GUIslice detailed in the wiki:
[Configure Touch Support](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-Touch-Support)
And for your specific touch driver in:
[Configuring Touch for STMPE610](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-STMPE610) 

As an example inside my configs/esp-tftespi-default-stmpe610.h in   **SECTION 4A**: Update your pin connections here
 I edit the file and set:
```
  // TFT_eSPI: Chip Select for Touch Device
  // - Since we are not using the TFT_eSPI's integrated XPT2046 touch driver,
  //   it is best to disable it. This can be done by ensuring that the TOUCH_CS line
  //   in TFT_eSPI's User_Setup.h is commented out.

  // Select touch device wiring method by setting one of the following to 1, others to 0
  #define ADATOUCH_I2C_HW 0  // Touch controller via hardware I2C (uses ADATOUCH_I2C_ADDR)
  #define ADATOUCH_SPI_HW 1  // Touch controller via hardware SPI (uses ADATOUCH_PIN_CS)
  #define ADATOUCH_SPI_SW 0  // Touch controller via software SPI [not yet supported]

  // Touch bus & pinout
  #define ADATOUCH_I2C_ADDR   0x41  // Touch device I2C address (for ADATOUCH_I2C_HW=1)
#ifdef ESP8266
  #define ADATOUCH_PIN_CS     D0    // Touch device chip select (for ADATOUCH_SPI_HW=1)
#elif ESP32
  #define ADATOUCH_PIN_CS     32    // Touch device chip select (for ADATOUCH_SPI_HW=1)
//  #define ADATOUCH_PIN_CS     13    // Touch device chip select (for ADATOUCH_SPI_HW=1)
#endif
```
I hope this clears things up for you.

Paul--

Bwanna, 2022-09-30:
Very much appreciate the reply!  I've been researching the Wiki and testing this for days with no luck.  

 The Adafruit_STMPE610 test program (TouchTest.ino)  only  requires a simple call, so nothing obvious to incorporate into the "../configs/esp-tftespi-default-stmpe610.h" file:  (unless I'm missing something?)
```
// Option #1 - uses I2C, connect to hardware I2C port only!
// SCL to I2C clock (#A5 on Uno) and SDA to I2C data (#A4 on Uno)
// tie MODE to GND and POWER CYCLE (there is no reset pin)
Adafruit_STMPE610 touch = Adafruit_STMPE610();
```

Here is what I have in "configs/esp-tftespi-default-stmpe610.h" in SECTION 4A: 
What information should I transfer from the TouchTest.ino to this file?
```
  // . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .
  // SECTION 4A: Update your pin connections here
  // . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .

  // TFT_eSPI: Chip Select for Touch Device
  // - Since we are not using the TFT_eSPI's integrated XPT2046 touch driver,
  //   it is best to disable it. This can be done by ensuring that the TOUCH_CS line
  //   in TFT_eSPI's User_Setup.h is commented out.

  // Select touch device wiring method by setting one of the following to 1, others to 0
  #define ADATOUCH_I2C_HW 1  // Touch controller via hardware I2C (uses ADATOUCH_I2C_ADDR)
  #define ADATOUCH_SPI_HW 0  // Touch controller via hardware SPI (uses ADATOUCH_PIN_CS)
  #define ADATOUCH_SPI_SW 0  // Touch controller via software SPI [not yet supported]

  // Touch bus & pinout
  #define ADATOUCH_I2C_ADDR   0x41  // Touch device I2C address (for ADATOUCH_I2C_HW=1)
#ifdef ESP8266
  #define ADATOUCH_PIN_CS     D0    // Touch device chip select (for ADATOUCH_SPI_HW=1)
#elif ESP32
  #define ADATOUCH_PIN_CS     13    // Touch device chip select (for ADATOUCH_SPI_HW=1)
#endif
```

Again, REALLY appreciate the assistance!!!  Been at this for many days and still unsuccessful.  :(
I'm really at a loss what to try next to get past this "pin detection" error.  It's preventing me from uploading the sketch.

Thank you!
Steve

Pconti31, 2022-09-30:
@Bwanna First please ignore the diag_ard_touch_detect.ino program it doesn't apply to your touch driver.
Use ex02_ard_btn_txt.ino as your test platform.  Note to circle back to diag_ard_touch_calib.ino to get an accurate set of X And Y min maxes for **SECTION 4B**  once you get ex02 to compile and run.

As for your "configs/esp-tftespi-default-stmpe610.h" in SECTION 4A the TouchTest if unmodified sets:
```
  // If using I2C you can select the I2C address (there are two options) by calling
  // touch.begin(0x41), the default, or touch.begin(0x44) if A0 is tied to 3.3V
  // If no address is passed, 0x41 is used
  if (! touch.begin()) {
```
So your defaults inside "configs/esp-tftespi-default-stmpe610.h" should be set to:
```
  // Select touch device wiring method by setting one of the following to 1, others to 0
  #define ADATOUCH_I2C_HW 1  // Touch controller via hardware I2C (uses ADATOUCH_I2C_ADDR)
  #define ADATOUCH_SPI_HW 0  // Touch controller via hardware SPI (uses ADATOUCH_PIN_CS)
  #define ADATOUCH_SPI_SW 0  // Touch controller via software SPI [not yet supported]

  // Touch bus & pinout
  #define ADATOUCH_I2C_ADDR   0x41  // Touch device I2C address (for ADATOUCH_I2C_HW=1)```
```
and believe it or not you are done.
Paul--

Bwanna, 2022-09-30:
@Pconti31  Perfect!  Thank you VERY much for the clarifications!  

Touchscreen is working with the other files.  Ran Calibration and inserted the numbers into the config file.  I'll stay away from the "Detect" sketch.   (I probably didn't need to get hung up on that sketch as much as I did. )

Cheers!
Steve