ImpulseAdventure/GUIslice #487: error E_PROJECT_OPTIONS tab->Graphics Library should be TFT_eSPI

umsmin, 2022-11-21:
### Describe the bug
I use the TFT_eSPI library from Bodmer. Wen I create the code, in the "_GSLC.h" file it shows this error

### Device hardware
- *MCU*: ESP32
- *Display with link*: ILI9341 2.8" resistive [https://www.amazon.de/gp/product/B08SBTBRZV/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1](url)
- *GUIslice config selected*:  `esp-tftespi-default-xpt2046_int.h`

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* I use it in a few own programs with TFT_eSPI
- *Touch works correctly with standalone touch library examples without GUIslice:* I use TFT_eSPI

### Expected behavior
I selected in the GUIslice Builder -> General -> Graphics Library ---> TFT_eSPI
Also I enabled the include in the GUIslice_config.h  ->  esp-tftespi-default-xpt2046_int.h

### Initialization messages
#error E_PROJECT_OPTIONS tab->Graphics Library should be TFT_eSPI

### Additional info
I manually slashed the error message in _GSLC.h file out.

`#error E_PROJECT_OPTIONS tab->Graphics Library should be TFT_eSPI`

Pconti31, 2022-11-22:
@umsmin This comes up often enough that I just added it to my faq.pdf with release 0.17.b16.  A future Release will issue  Warning to modify your Project using the Project Options tab after making changes to your User Preferences.

**All new projects** will use the values inside General preferences  but your existing projects will still have the old values and will need to be modified by going to the Project Options tab and changing the graphics library.  
Otherwise your project will continue to use the previous values. Anytime you make changes inside the General Tab you should always double check your values inside your open project.
Paul--

umsmin, 2022-11-22:
Thanks Paul for this information. Now it's working. But one question, how can I add the background image I've added in the GUIslice Builder. I use the ESP32 and display without SD-card. Must I create a data folder into the project and upload it with the SPIFFS uploader in the Arduino IDE?

Uli

Pconti31, 2022-11-23:
@umsmin You know I wasn't sure myself if spiffs would work or not.  So I made a test program using my builder and it does work but for one minor issue.  After you set **use background=true** and select your BMP file the builder assumes you are using the SD card for storage:

[image]

You need to change this to **gslc_GetImageFromFile**:

[image]

and of course upload your image from your project's sub-folder **data** before running your App.

Paul--

umsmin, 2022-11-24:
Sorry for my late replay. This things i know. I uploaded the BMP image with 24 Bit and resolution of 320x240. Added in the Builder config exactly on this rows you shown. But nothing, okay not nothing. Now all pages will be smashed together. When I push any button to get a page, the last screen will also been shown. Also the popups will stay there.

Even your examples with the Logo from Guislice Builder will not be shown. Only a white place on display.

Thanks Uli

Pconti31, 2022-11-24:
@umsmin Well, once again I did a test with a background BMP loaded from spiffs and it worked fine.  Since you also had a problem with the GUIslice logo I assume you mean you tried the example ex44_bld_spiffs and it failed for you.  I just now ran this example and as expected it also worked fine.  

Before you mentioned the example ex44_bld_spiffs I thought it might be your BMP file but the example failing rules that out.   That leaves us with very few answers.  What version of GUIslice API are you using?
Post the number from GUIslice/src/GUIslice_version.h my version is:
```
#define GUISLICE_VER "0.17.1.3"
```
while I used TFT_eSPI version 2.4.72 and there is only one release of TFT_eFEX 0.0.5

If the versions match then the only thing I can think of is that BMP are not being loaded into your data folder correctly.   GUIslice uses SPIFFS but LITTLEFS is another file system and its not same format.  

Paul--

umsmin, 2022-11-24:
The TFT_eSPI version is 2.4.76 but on the Arduino IDE it shows 2.4.79 but when I get your version, nothing changed.
The Guislice version is `#define GUISLICE_VER "0.17.0"`. On the Arduino IDE it shows only version 0.16.0 and 0.17.0 or older.
But also on Github it shows the same. How can I get your version, maybe that's the reason?

Another question what I have, is it possible to make a "Bootscreen"? I tried but it's too late when I get it.

Uli

Pconti31, 2022-11-25:
@umsmin No, I just downloaded GUIslice 0.17.0 and it also worked fine.  So that's not it. 
Just for information to get the latest GUIslice API go to the GUIslice repository [GUIslice](https://github.com/ImpulseAdventure/GUIslice) and select the **code** drop down and then **download zip file**.
I am at a lost to say what is your problem.  Seems only file corruption is left.  Attached is a stand alone app I just put together using just TFT_eSPI , TFT_eFEX and JPEGDecoder.  

It will scan spiffs file system and list all files it finds and if one of them is BMP  or JPG file it will display it on your TFT screen, wait 3 seconds and go find another.

You should open your Serial Monitor and set it to 9600 baud.
If you load images from my supplied **data** folder you should see this output on your Serial Monitor:
```
Listing SPIFFS files:
=================================================
  File name                              Size
=================================================
/back1_24.bmp                        230456 bytes
/back2_24.bmp                        230456 bytes
/exit_g24.jpg                         23364 bytes
/exit_n24.jpg                         22943 bytes
/logo.jpg                             31765 bytes
run time in ms: 22237
=================================================
```
attached is the zip file of this test.

[bmp_esp32.zip](https://github.com/ImpulseAdventure/GUIslice/files/10093816/bmp_esp32.zip)

Pconti31, 2022-11-25:
@umsmin If you want to create a splash screen or bootscreen at startup please post a new topic so this one doesn't get muddied up.
Paul--

umsmin, 2022-11-25:
I think that's the problem:

`/wasser.bmp                          153666 bytes`
`BMP format not recognised.
Now I used a sketch for SPIFFS handling. With this the upload works perfectly.
Thanks

Uli

Pconti31, 2022-11-25:
@umsmin Glad I was able to help.
Paul--

umsmin, 2022-11-25:
Sure!
You made this great stuff.
Thanks

Uli

I think we can close it

closed 2022-11-25 by umsmin