ImpulseAdventure/GUIslice #488: How making a Bootup splashscreen

umsmin, 2022-11-25:
Another question what I have, is it possible to make a "Bootscreen"?

Device hardware
MCU: ESP32
Display with link: ILI9341 2.8" resistive [https://www.amazon.de/gp/product/B08SBTBRZV/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1](https://github.com/ImpulseAdventure/GUIslice/issues/url)
GUIslice config selected: esp-tftespi-default-xpt2046_int.h

### Checklist to try first
I tried with all manuals from github.

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* It's not a bug
- *Touch works correctly with standalone touch library examples without GUIslice:* yes

### Expected behavior
I need for my Hardware a screen who show the progress. It works but was shown to late. The hardware was nearly initialized

Uli

Pconti31, 2022-11-25:
@umsmin Not really sure what you are looking for here.
I took my best guess and created modified a demo I have for doing a splash screen at startup that shows a logo(GUIslice).  The modification is removing a timer and replacing it with a progress bar.  Once the progress bar is full the popup is closed.  I simulate hardware loading using a simple counter.  

If this isn't what you are looking for please **post a small example** simulating your hardware with a counter and I'll take another look.  

[ex_boot_screen.zip](https://github.com/ImpulseAdventure/GUIslice/files/10095134/ex_boot_screen.zip)

Paul--

umsmin, 2022-11-25:
Perfectly it works. 
Thanks a lot

Uli

Pconti31, 2022-11-26:
@umsmin No Problem.   If this works for you I would appreciate a star in my repository
[GUIsliceBuilder](https://github.com/ImpulseAdventure/GUIslice-Builder).
Also, since **GUIslice** isn't my repository I have no access to close this issue you have to do it.   
Paul--

umsmin, 2022-11-26:
Now it works. The maximal count of 100 was never reached. So he can't close it.

The STAR will be there because you preserve it!

Thanks Uli

closed 2022-11-26 by umsmin