ImpulseAdventure/GUIslice #491: Updating a runtime modifiable text on PAGE: E_PG_BASE not updating

umsmin, 2022-11-30:
### Describe the bug
The display don't show and update the values

### Device hardware
- *MCU*: ESP32
- *Display with link*: https://www.amazon.de/gp/product/B08SBTBRZV/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1
- *GUIslice config selected*: TFT_eSPI

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* perfectly
- *Touch works correctly with standalone touch library examples without GUIslice:* same

### Expected behavior
I try to get the time and date on basepage. On serial output it shows exactly what expected.

### Initialization messages
No errors will be shown.

### Additional info

```
strftime(acTxt, MAX_STR, "%d.%m.%y", localtime(&now));
gslc_ElemSetTxtStr(&m_gui, m_pElemDatum1, acTxt);
Serial.println(acTxt);

gslc_Update(&m_gui);
```

Thanks
Uli

Pconti31, 2022-12-01:
@umsmin Sorry, but since I can see your whole program I have no idea what you are doing wrong.

So, I have attached a sample i just put together using NTP Time server to get time and I display it on a base page. 

It does show up on the two pages I created, E_PG_MAIN and E_PG2.

Make sure you are running the latest GUIslice API 0.17.3 and follow the directions I placed in comments at the top of this example on how to set it up.  Basically, add your wifi name and password, and set your time zone offset.

[ntp-test.zip](https://github.com/ImpulseAdventure/GUIslice/files/10128258/ntp-test.zip)

I think tomorrow I'll add this to my [GUIslice-Solutions](https://github.com/Pconti31/GUIslice-Solutions) Repository.
Paul--

Pconti31, 2022-12-01:
I cleaned up the sample code and added a README.md file.
[ntp-test.zip](https://github.com/ImpulseAdventure/GUIslice/files/10134810/ntp-test.zip)

umsmin, 2022-12-01:
@Pconti31 Thanks for your help. It's really funny because your example works perfectly. When I add your code to mine and separate it in two parts. One for time, the other for the date. Time works, date not. 
I changed now from two to one longer field and it works. Not nice but okay for now. Curious on this is that the code with time works. Maybe it will not be actualized without the seconds in time? 
From this time I add the `"%s"` into `"%d.%m.%y"` he show the date. Otherwise not. Your code works both perfectly. 

For me I use a small helper to see what's there:
```
//------------------------------------------------------------------------------------------
// Helper for debugging the acTxt parameter on screen 
//------------------------------------------------------------------------------------------
void debugacTxt() {
#if (DEBUG_ACT)
  Serial.print("D E B U G ---> acTxt :   ");
  Serial.println(acTxt);
  Serial.println("");
#endif
}
```

Every time looking about the trouble like this so I enable the `DEBUG_ACT `
And there I see the date was correctly build.

I use the `NTP_Time` and also the `RTC_Time` in my code without problems. But on the display not.

I tried a lot of things but the only one with a long string for time and date together works for now. Maybe later I've more time for this problem.

Uli

Pconti31, 2022-12-01:
@umsmin I suspect the problem may be with your definition of the m_pElemDatum1 text field.  Maybe the storage length isn't long enough? Anyway, seems like you have a work-around so good luck and if you get things working maybe post in the discussion section a description of your project.

umsmin, 2022-12-01:
The length is the same like the time. In my debugging on serial he show it.
Here my work-around:

```
void printLocalTime()
{
  if (millis() - m_nStartTime >= m_nAskTimeRate) {
    strftime(acTxt, MAX_STR, "%d.%m.%y  %S", localtime(&now));
    gslc_ElemSetTxtStr(&m_gui, m_pElemDatum, acTxt);
    
    strftime(acTxt, MAX_STR, "%H:%M:%S", localtime(&now));
    gslc_ElemSetTxtStr(&m_gui, m_pElemUhr, acTxt);
    debugacTxt();
    m_nStartTime = millis();
  }
}
```
I take the time now on the left side and date on right. The seconds are now out of the screen.

Uli 

Before I add the `"%S"` for update it was 8. Now 14.
I don't use the "%Y", but also tested.

umsmin, 2022-12-02:
@Pconti31 Now it looks nice but the touch now nearly unusable. Mostly I have to touch a button extremely long that he react. After this he rewrite the screen 3 times. It looks like this blocks the screen completely for update time and date. 
I normally use the `millis()` function without problems. But here not. 
Maybe you have an idea for a better update every second.

Thanks

Uli

umsmin, 2022-12-02:
The next funny thing is that the time will not be updated when a popup is running. Is this normally?

Uli

Pconti31, 2022-12-02:
@umsmin There are **hardware timer libraries** you can google, Some made just for esp32's and some for Arduinos and ESPxxxx. 

If your touch button is really slow to react I assume you must be doing something in your loop() beyond just keeping time that is drawing too CPU intensive.

If so, and you are using a esp32 you might want to look into the fact that esp32 MCU has more then one cpu.  GUIslice and any other Arduino code always runs in core 1 leaving core 0 idle.  You can check out
[How to use ESP32 Dual Core with Arduino IDE](https://randomnerdtutorials.com/esp32-dual-core-arduino-ide/)
you can also google for pthreads.  

This really has gone beyond what this issues forum was designed to handle.  You might want to open a discussion about how to deal with touch response time.  

As for the poup I added one to my ntp-test and it does not update time while the popup.  I have no idea if this is a bug or a design point.  The API is beyond my control.  Here I'm just another user like you. I suggest you post a new issue to catch @ImpulseAdventure Calvin's attention.   Although I wouldn't hold your breath waiting for a response.
Paul--

umsmin, 2022-12-11:
@Pconti31 Thanks for your help at all. The problems about hanging screens and buttons is fixed now. My code blocked a few times the system to run well. I restructured the most and now it's good. 
The problem about refreshing the clock when popup is shown exist. I've no solution. I putted it into "Discussions" without answer. 

Uli

closed 2022-12-11 by umsmin