ImpulseAdventure/GUIslice #499: Fatal Error has occurred.

MericR972, 2023-01-13:
everytime i save my project there is a fatal error and a crash log is created

MericR972, 2023-01-13:
class java.lang.IllegalArgumentException: Comparison method violates its general contract!
java.base/java.util.ComparableTimSort.mergeHi(ComparableTimSort.java:870)
java.base/java.util.ComparableTimSort.mergeAt(ComparableTimSort.java:487)
java.base/java.util.ComparableTimSort.mergeForceCollapse(ComparableTimSort.java:426)
java.base/java.util.ComparableTimSort.sort(ComparableTimSort.java:222)
java.base/java.util.Arrays.sort(Arrays.java:1041)
java.desktop/sun.awt.shell.Win32ShellFolderManager2.get(Win32ShellFolderManager2.java:315)
java.desktop/sun.awt.shell.ShellFolder.get(ShellFolder.java:277)
java.desktop/javax.swing.plaf.metal.MetalFileChooserUI$DirectoryComboBoxModel.addItem(MetalFileChooserUI.java:1033)
java.desktop/javax.swing.plaf.metal.MetalFileChooserUI.doDirectoryChanged(MetalFileChooserUI.java:721)
java.desktop/javax.swing.plaf.metal.MetalFileChooserUI$5.propertyChange(MetalFileChooserUI.java:810)
java.desktop/java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:343)
java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:335)
java.desktop/java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:268)
java.desktop/java.awt.Component.firePropertyChange(Component.java:8722)
java.desktop/javax.swing.JFileChooser.setCurrentDirectory(JFileChooser.java:609)
java.desktop/javax.swing.JFileChooser.<init>(JFileChooser.java:379)
java.desktop/javax.swing.JFileChooser.<init>(JFileChooser.java:324)
builder@0.17.b17/builder.views.RibbonListener.createFolderDialog(RibbonListener.java:102)
builder@0.17.b17/builder.views.RibbonListener.actionPerformed(RibbonListener.java:631)
java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:354)
java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Actions.actionPerformed(BasicMenuItemUI.java:983)
java.desktop/javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1810)
java.desktop/javax.swing.JComponent.processKeyBinding(JComponent.java:2947)
java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(JMenuBar.java:710)
java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(JMenuBar.java:717)
java.desktop/javax.swing.JMenuBar.processBindingForKeyStrokeRecursive(JMenuBar.java:717)
java.desktop/javax.swing.JMenuBar.processKeyBinding(JMenuBar.java:688)
java.desktop/javax.swing.KeyboardManager.fireBinding(KeyboardManager.java:311)
java.desktop/javax.swing.KeyboardManager.fireKeyboardAction(KeyboardManager.java:297)
java.desktop/javax.swing.JComponent.processKeyBindingsForAllComponents(JComponent.java:3040)
java.desktop/javax.swing.JComponent.processKeyBindings(JComponent.java:3032)
java.desktop/javax.swing.JComponent.processKeyEvent(JComponent.java:2909)
java.desktop/java.awt.Component.processEvent(Component.java:6403)
java.desktop/java.awt.Container.processEvent(Container.java:2266)
java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1952)
java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:883)
java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1150)
java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1020)
java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:848)
java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4882)
java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

Pconti31, 2023-01-13:
@MericR972 Not being a mind reader you need to give me some help here.
And next time for Builder issues post in my repository since I'm just another user here.
Now, what version of Builder did you use?
What op sys and version?
Did you do a Save or SaveAS?  (a few bugs exist with SaveAs)
Zip the GUIsliceBuilder/logs which detail everything I need to debug and post the zip file.
Paul--

Pconti31, 2023-01-13:
@MericR972 The crash indicates you are running the java Metal theme and we crash within the MetalFileChooserUI.java not inside the Builder itself...  Doing a time sort of the folder for some reason.
This is quite odd.  Can you try another theme, like Flat Light?

MericR972, 2023-01-16:
> @MericR972 Not being a mind reader you need to give me some help here. And next time for Builder issues post in my repository since I'm just another user here. Now, what version of Builder did you [...]

I am on GUIslice Builder ver: 0.17.b18
and my computer is on the last version of win11 

[logs.zip](https://github.com/ImpulseAdventure/GUIslice/files/10427081/logs.zip)

Pconti31, 2023-01-16:
@MericR972 Unfortunately  the builder.log only shows you open the builder and did the about command.
Could you delete everything in your logs folder and retry creating a new project with just a text button and doing a save (not saveas) and zip up the logs again assuming a crash.  
Also, what Theme are you using.  The crashes suggest java metal which is an ancient theme.

Pconti31, 2023-01-21:
@MericR972 I have published Release 0.17.b19 using Java 18 and FlatLaf 3.0.  Post here if it works for you.

If not, I'll put together a special version for you to test that uses a much simplified swing.JFileChooser when it detects running on Windows 11.
Paul--