ImpulseAdventure/GUIslice #502: Is this project still maintained?

ilka-schulz, 2023-02-28:
Is this project still maintained? I am asking because there are quite many open PRs...

ImpulseAdventure, 2023-03-01:
Hey @ilka-schulz — yes, I am still keen to continue developing the project but have entered a very busy period at work that has prevented me from closely monitoring the repo or performing my normal testing of PRs.

In the meantime, I am very grateful to @Pconti31 for his incredible help in answering questions!

A possible option to consider for now: if people are able to confirm some degree of testing of their / other PRs and the changes don’t look too invasive (from my review), perhaps I could apply a tentatively merge, and then follow up with my own full regression testing ahead of any release as time permits. Doing so would naturally add some risk of instability in the latest trunk but would alleviate the delays.

Thanks to everyone for being so understanding!

ilka-schulz, 2023-03-01:
@ImpulseAdventure Thank you very much for the quick and elaborate answer! I am happy to hear that the project is still that active and your strategies sound perfect!

I am also busy at work but I guess I will start by reviewing some PRs and opening my own later. However, I am quite a bit dependent on #501 (I would be fine with implementing it myself) and I would love to hear your opinion on that. I really want my final product to look sleek...

btw: I think you can close this issue, right?

DonpK, 2023-03-01:
I just want to thank Cal and Paul for the enormous amount time and effort they have put in on GUISlice. I've used it on a number of my projects.

Don

ImpulseAdventure, 2025-02-17:
Closing with comment added to #501 .

Although my available time continues to be quite limited, I'm happy to accept / review contributions from the community, and make enhancements when the scope is relatively well controlled. I am also extremely grateful to @Pconti31 for his continued efforts in offering expert guidance to users as well, along with the excellent [GUIslice Builder](https://github.com/ImpulseAdventure/GUIslice-Builder/) integration.

closed 2025-02-17 by ImpulseAdventure