ImpulseAdventure/GUIslice #504: Image redraw and text flicker - TFT_eSPI/ESP32/ILI9488

Osocrates, 2023-03-21:
I have a couple of text labels that update through some variables, currently the same counter. A couple of labels are in a box along with a static string and an image, so when the counter changes, only two labels should change. 

However, the static string and the image are also flashing whenever the display updates, and not only that, when checking with DBG_REDRAW you can see that the string and image flash when 3 labels update.

[image]

[image]

Here's the code I use to update my elements:
`  if(((millis()-last_t)>= 10)){
      gslc_ElemXRingGaugeSetVal(&m_gui, m_pElemXRingGauge1, valor);
      gslc_Update(&m_gui);
      gslc_ElemXProgressSetVal(&m_gui, m_pElemProgress1, ((valor - 0) * (42 - 0) / (39.5 - 0) + 0));
      gslc_Update(&m_gui);
      gslc_ElemXRadialSetVal(&m_gui, m_pElemRadial3, valor);
      gslc_Update(&m_gui);
      valor+= 1;
      //memset(txt_speed, 0, sizeof(txt_speed));
      //gslc_ElemSetTxtStr(&m_gui, text_speed , txt_speed);
      //gslc_Update(&m_gui);
      //gslc_ElemSetTxtStr(&m_gui, text_power , txt_speed);
      //gslc_Update(&m_gui);
      //gslc_ElemSetTxtStr(&m_gui, text_travdist , txt_speed);
      //gslc_Update(&m_gui);
      //gslc_ElemSetTxtStr(&m_gui, text_volt , txt_speed);
//gslc_Update(&m_gui);
      //memset(txt_speed, 0, sizeof(txt_speed));
      snprintf(txt_speed, MAX_STR,"%.1f", valor);
      strcat(txt_speed," Km/h"); 
      gslc_ElemSetTxtStr(&m_gui, text_speed , txt_speed);
      gslc_Update(&m_gui);
      
      //memset(txt_speed, 0, sizeof(txt_speed));
      snprintf(txt_speed, MAX_STR,"%.1f", ((valor - 0) * (999.0 - 0) / (40 - 0) + 0));
      strcat(txt_speed," W"); 
      gslc_ElemSetTxtStr(&m_gui, text_power , txt_speed);
      gslc_Update(&m_gui);
      
      //memset(txt_speed, 0, sizeof(txt_speed));
      snprintf(txt_speed, MAX_STR,"%.1f", ((valor - 0) * (99.9 - 0) / (40 - 0) + 0));
      strcat(txt_speed," Km"); 
      gslc_ElemSetTxtStr(&m_gui, text_travdist , txt_speed);
      gslc_Update(&m_gui);
      
      //memset(txt_speed, 0, sizeof(txt_speed));
      snprintf(txt_speed, MAX_STR,"%.1f", ((valor - 0) * (42.0 - 0) / (40 - 0) + 0));
      strcat(txt_speed," V"); 
      gslc_ElemSetTxtStr(&m_gui, text_volt , txt_speed);
      gslc_Update(&m_gui);
      
      if(valor>=40 ){
        valor = 0;
      }
      last_t = millis();
      
  }

  if (valor<20){
    gslc_ElemSetVisible(&m_gui,icon_battery,false);
    gslc_Update(&m_gui);
    gslc_ElemSetVisible(&m_gui,icon_brakes,false);
    gslc_Update(&m_gui);
  }else{
    gslc_ElemSetVisible(&m_gui,icon_battery,true);
    gslc_Update(&m_gui);
    gslc_ElemSetVisible(&m_gui,icon_brakes,true);
    gslc_Update(&m_gui);
  }`

I use a solid color as a background:

`gslc_SetBkgndColor(&m_gui,GSLC_COL_BLACK);`

Pconti31, 2023-03-21:
@Osocrates it's possible some of your refresh issues is being caused by the drawing inside the box element. Try removing them and see if anything improved if so replace them using the line elements to draw your own boxes.

If it's still a problem copy your project to another and remove any sensors and external libs replacing them with hardcoded data so I can run just the GUI. Then post the zip file of your project and I'll see what I can find...

Paul--

Osocrates, 2023-03-22:
Thanks, that seems to have solved the issue. I guess that the error was due to a bigger redraw sector due to the boxes. I still see a little bit of flicker on the labels, but that might be an effect of the update frequency.

closed 2025-02-17 by ImpulseAdventure