ImpulseAdventure/GUIslice #508: Touch IRQ (STMPE610)

Bwanna, 2023-05-01:
**### Feature Request or Enhancement**
**Issue:** Sketch coding is delaying or pausing the response of the touchscreen
**Request:** Configure STMPE610 touch actions to result in an IRQ ISR (or similar) rather than Polling the GPIO.  

Noticed that **esp-tftespi-default-stmpe610.h** includes: "Adafruit_STMPE610 **m_touch** = Adafruit_STMPE610();"
Question:  Can "m_touch" be used in the main sketch to send commands for the STMPE610?

Thanks,
Steve

Pconti31, 2023-05-12:
@Bwanna I would suggest the easiest solution would be to get guidlice out of the way. See ex16_ard_touch_hnd. Feel free to post any questions. I should point out that the stmpe610 is noted to drop data when using interrupt which is why polling is used.
Paul--

Bwanna, 2023-05-14:
Paul,
I appreciate the suggestion. Reviewed the ex and guislice-th files. Seems straight forward.  I am concerned about the dropped data.
Thanks again,
Steve

closed 2023-05-14 by Bwanna