ImpulseAdventure/GUIslice #515: Custom BMP wont load on ILI9341 ft6206

dsull79, 2023-07-09:
I have setup a screen with BMP image buttons.  The images will show if i use BMP from the example sketchs, however if i use images i have created myself in Windows Paint and saved as 24-bit BMP they will not show.
Any thoughts on what i can do to fix this issue?
I have attached the sketch files and the images.
[HHM2.0.zip](https://github.com/ImpulseAdventure/GUIslice/files/11995005/HHM2.0.zip)

Pconti31, 2023-07-09:
@dsull79 
adding some trace statements to guislice give this output
```
GUIslice version [0.17.1.3]:
DBG: InvRgnReset
- Init display handler [ADA_ILI9341(SPI-HW)] OK
- Init touch handler [STMPE610(SPI-HW)] OK
DBG: InvRgnScreen
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: InvRgnPage: Page=0 (0,0)-(-1,-1)
DBG: InvRgnScreen
DBG: PageRedrawGo() InvRgn: En=1 (0,65535)-(320,239) PageRedraw=1
DBG: DrvDrawImage() with ImgBuf address=00000000
DrvDrawBmp24FromSD() Loading image [/home_icon_blk_50.bmp]
ERROR: DrvDrawBmp24FromSD() file not found [/home_icon_blk_50.bmp]
DBG: InvRgnReset
```
The reason is that the sd file system is more like old msdos and likes 8.3 file names
renaming them to home.bmp and homeclk.bmp and dumping them onto the sd card 
allowed them to be found but then another problem appeared:

Coordinates are 0 based your screen goes from 0-319 so placing your icon on 270+50 gives 320 
exactly overflowing thus by one pixel.  Moving your images to x=269 fixes that.
However, there is also a bug inside GUIslice_drv_adagfx.cpp 
```
void gslc_DrvDrawBmp24FromSD(gslc_tsGui* pGui,const char *filename, uint16_t x, uint16_t y)

BUG ===>  if((x >= pGui->nDispW) || (y >= pGui->nDispH)) return;

comment this out to fix

//  if((x >= pGui->nDispW) || (y >= pGui->nDispH)) return;

```
After all of this your images load fine.

dsull79, 2023-07-10:
Very grateful Paul, thankyou.  I have been trying to get that working for sometime.  All working now with my images.

closed 2023-07-10 by dsull79