ImpulseAdventure/GUIslice #516: gslc_SetPageCur does not work reliably

ilka-schulz, 2023-07-18:
### Describe the bug
`gslc_SetPageCur` works just fine for the first couple times switching from one page to another. However, after the first couple switches, it shows an array of spurious phenomena:
- the new page is shown and immediately the old page is shown again before the new page is finally shown again and stays – basically GUIslice switches back to the old page for a short time before finally switching to the new page
- when switching between pages for many times (e.g. 20 times), it happens at some point that the page switching does not succeed: GUIslice shows the new page, then the old page and then sticks with the old page.
- When the above happened, page switching is completely broken (page switches are skipped at random) until the µController is rebooted.

### Device hardware
- *MCU*: ESP32-WROVER
- *Display with link*: generic 2.4" touchscreen with ILI9341 and XPT2046  (https://de.aliexpress.com/item/33015586094.html)
- *GUIslice config selected*:  `ard-adagfx-ili9341-xpt2046.h`

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* yes
- *Touch works correctly with standalone touch library examples without GUIslice:* yes

### Expected behavior
When switching to another page by calling `gslc_SetPageCur` the new page is shown and they previous page is not shown again. All page switching succeeds.

### Initialization messages
TODO: add this – I currently do not have the device connected via USB.

### Additional info
All page switching in my application is done in button callbacks which looks sth. like this:

```c++
enum class Page : int16_t {
    E_PG_STARTUP,
    E_PG_UPDATE,
    E_PG_INFOSCREEN,
    E_PG_MAIN,
    E_PG_SETTINGS_ROOT,
    E_PG_SETTINGS_NETWORK,
    E_PG_SETTINGS_NETWORK_ETHERNET,
    E_PG_SETTINGS_NETWORK_WIFI,
    E_PG_SETTINGS_SYSTEM,
    E_POP_KEYPAD,
    MAX_PAGE
};

std::map<gslc_tsElemRef *, Page>
    settings_node_button_pages;  // associates the IDs of buttons for settings
                                 // nodes with their respective settings page
bool on_setting_node_button_click(void *pvGui, void *pvElemRef,
                                  gslc_teTouch eTouch, int16_t nX, int16_t nY) {
    gslc_tsElemRef *pElemRef = (gslc_tsElemRef *)(pvElemRef);

    Page page;
    try {
        page = settings_node_button_pages.at(pElemRef);
    } catch (std::out_of_range &) {
        ESP_LOGE("GUI", "settings button click: button not found");
        page = Page::E_PG_STARTUP;
    }
    ESP_LOGD("GUI", "show page %i", page);
    gslc_SetPageCur(&m_gui, (int16_t)page);

    return true;
}
```

Adding/removing these compiler flags has no effect on the issue: `-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue`

ilka-schulz, 2023-09-23:
@Pconti31 I just saw that this issue is still open. The fix above works and I think this issue can be closed. Thank you again for your help!

Pconti31, 2023-09-23:
@ilka-schulz I'm happy it's working for you now.  Here I'm just another user like you.  I don't have write access so I can't close issues or merge pull requests myself except inside the GUIslice Builder repository.  However, You should be able to close it since you opened it.
Paul--

closed 2023-09-24 by ilka-schulz