ImpulseAdventure/GUIslice #522: Config file for ESP32 / ILI9488 / GT911 capacitive touchscreen using TFT_eSPI library

BMWK1, 2023-11-09:
### Feature Request or Enhancement
I would like to make this [breakout board](https://fr.aliexpress.com/item/1005004083478309.html?spm=a2g0o.order_list.order_list_main.172.d6fe5e5bqs89RU&gatewayAdapt=glo2fra) works with GUIslice, which is a capacitive 3.5" touchscreen based on ILI9488 and GT911 interface ICs.
But i can't configure GUIslice with the existing config files, and cannot understand how to build a config file by myself.

### Additional context
*Add any other context or screenshots about the feature request here.*

florinbaciuu, 2023-11-14:
If you look at this [PR](https://github.com/ImpulseAdventure/GUIslice/pull/521)I modified the files [GUIslice_drv_tft_espi.h](https://github.com/ImpulseAdventure/GUIslice/pull/521/commits/2b79428ee0652678e3400fa4226586fd29e8a11d) and [GUIslice_drv_tft_espi.cpp](https://github.com/ImpulseAdventure/GUIslice/pull/521/commits/513801e5944cf0b1adeb5107412a87732cd1316b) to work with a touchscreen module similar to yours. GT911 is similar to CST816 or CST820. The [touch library](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/lib/TouchLib) I use is compatible with your touch. Unfortunately, I don't have the device you have and I can't test the code, but please take a look at what I changed. maybe you can manage to write your own driver. the [configuration ](https://github.com/ImpulseAdventure/GUIslice/pull/521/commits/df1cae65cfc05ba53f695a36c994af8bb03d8bdb#)is easy.

ImpulseAdventure, 2025-02-17:
Closing based on suggestion from florinbaciuu

closed 2025-02-17 by ImpulseAdventure