ImpulseAdventure/GUIslice #527: GUIslice_drv_adagfx.cpp is missing button support for Adafruit 1.8" TFT Shield with Joystick (code in description)

Dowster64, 2024-02-01:
### Describe the bug
*Please describe what you observe on the display or any error messages*
When running ex40_ard_ctrls_small.ino and changing input mapping to use the buttons instead of the joystick there is no response.

### Device hardware
- *MCU*: Arduino UNo R4 Minima
- *Display with link*: Adafruit 1.8" Color TFT Shield w/microSD and Joystick - v 2 (https://www.adafruit.com/product/802)
- *GUIslice config selected*: `ard-shld-adafruit_18_joy.h`

### Checklist to try first
*Most display/touch issues can be identified by working through the steps in the [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) guide.*

*Please confirm whether:*
- *Display works correctly with standalone display driver library examples (eg. TFT_eSPI graphicstest, Adafruit graphicstest, etc.) without GUIslice:* Yes
- *Touch works correctly with standalone touch library examples without GUIslice:* - Well buttons and joystick work with examples

### Expected behavior
*A description of what you expected to happen*
Button A,B,C and longpress ABC should work.

### Initialization messages
*Please copy any startup messages / errors reported to the serial monitor window here. You can enable all error messages by setting `DEBUG_ERR 2` in the GUIslice config file.*

### Additional info
*Add any other context about the problem here.*
I apologise but I don't know how to do a pull request, but modifiying GUIslice_drv_adagfx.cpp section starting lin 2924:
`#ifdef DRV_DISP_ADAGFX_SEESAW`
and extending the switch debounce code with:
` else if ((nButtonsLast & TFTSHIELD_BUTTON_1) && !(nButtonsCur & TFTSHIELD_BUTTON_1)) {
      *peInputEvent = GSLC_INPUT_PIN_ASSERT;
      *pnInputVal = GSLC_PIN_BTN_A;
    } else if (!(nButtonsLast & TFTSHIELD_BUTTON_1) && (nButtonsCur & TFTSHIELD_BUTTON_1)) {
      *peInputEvent = GSLC_INPUT_PIN_DEASSERT;
      *pnInputVal = GSLC_PIN_BTN_A;
    } else if ((nButtonsLast & TFTSHIELD_BUTTON_2) && !(nButtonsCur & TFTSHIELD_BUTTON_2)) {
      *peInputEvent = GSLC_INPUT_PIN_ASSERT;
      *pnInputVal = GSLC_PIN_BTN_B;
    } else if (!(nButtonsLast & TFTSHIELD_BUTTON_2) && (nButtonsCur & TFTSHIELD_BUTTON_2)) {
      *peInputEvent = GSLC_INPUT_PIN_DEASSERT;
      *pnInputVal = GSLC_PIN_BTN_B;
    } else if ((nButtonsLast & TFTSHIELD_BUTTON_3) && !(nButtonsCur & TFTSHIELD_BUTTON_3)) {
      *peInputEvent = GSLC_INPUT_PIN_ASSERT;
      *pnInputVal = GSLC_PIN_BTN_C;
    } else if (!(nButtonsLast & TFTSHIELD_BUTTON_3) && (nButtonsCur & TFTSHIELD_BUTTON_3)) {
      *peInputEvent = GSLC_INPUT_PIN_DEASSERT;
      *pnInputVal = GSLC_PIN_BTN_C;
    }`

works for the standard press.  I don't know how to code for the longpress enumerations but I hope this helps someone.

Dowster64, 2024-02-11:
Just a comment about longpress of buttons. Paul has solutions for this (or ones to modify) and other examples here:
https://github.com/Pconti31/GUIslice-Solutions/tree/main/apps/Longpress-Btn

ImpulseAdventure, 2024-02-20:
Hi @Dowster64 -- thank you for pointing this out!

I had been testing the Adafruit seesaw with the joystick control, but you are right in that the standard button mappings (A/B/C) should be offered too.

I am in the process of getting my regression environment up & running again (I had to freeze my work on the repo for an extended period), and then should be able to merge the updates that you suggested here.

thanks!

ImpulseAdventure, 2024-02-26:
> _Add any other context about the problem here._ I apologise but I don't know how to do a pull request, but modifiying GUIslice_drv_adagfx.cpp section starting lin 2924:

@Dowster64 If you'd like to try a simple approach to submitting a request (since these changes are quite limited), you may be able to navigate to the file in the [GUIslice_drv_adagfx.cpp](https://github.com/ImpulseAdventure/GUIslice/blob/master/src/GUIslice_drv_adagfx.cpp) file in the GitHub repo and click on the *Edit* icon. Make your changes to the code and then click *Commit Changes*. From there, you should be given the option to *Create a new branch for this commit and start a pull request*.

Dowster64, 2024-02-28:
@ImpulseAdventure - Thanks for taking the time to explain how the process of creating a pull request works.  I can see the reason for the fork and the branch, it makes sense once you explained it.  I've submitted the request, I just hope I haven't missed any curly brackets! :-)

ImpulseAdventure, 2024-03-01:
Great, thanks for giving it a try! It has been merged now.

closed 2024-03-01 by ImpulseAdventure