ImpulseAdventure/GUIslice #537: feature request: page flag: show/hide base page

ilka-schulz, 2024-05-05:
The `gslc_tsPage` type should get a flag which indicates whether the "base page" is shown on this page or not. A typical base page would be a status bar and such should be shown on every page by default but pages should be exempted by setting a flag. 

**Example:** Some boot screen or some "installing updates" screen should not show the status bar base page but should fill the entire screen.

@ImpulseAdventure I would just implement this feature myself and open a PR at some point. Is that OK for you?

Pconti31, 2024-05-05:
@ilka-schulz I'm not sure this is the right way to do this.  First it could be way more complicated then you might think.  It also would be very difficult to support within the GUIsliceBuilder.

Plus supporting boot, splash, and update screens can already be done easily.
See my GUIslice-Solutions Repro.
 - [ex_boot_screen](https://github.com/Pconti31/GUIslice-Solutions/tree/main/apps/ex_boot_screen)
 - [ex_splash_screen](https://github.com/Pconti31/GUIslice-Solutions/tree/main/apps/ex_splash_screen)
 - [busywait](https://github.com/Pconti31/GUIslice-Solutions/tree/main/apps/busywait)

Of couse, feel free to do as you want.  This is just my opinion...
Paul--

ilka-schulz, 2024-05-06:
Hmm, I see your points. Using the popup stack layer is an elegant solution for my problem. Also, I completely forgot about the builder – I am on Linux and never used it...

To achieve the request functionality, there seems to be an easier solution: I *already* have a wrapper function in my project around `gslc_SetPageCur` and I could have used that wrapper function to hide the base page on certain pages and show it on certain other pages, e.g.:

```cpp
enum class Page : int16_t {
    E_PG_BASE,
    E_PG_STARTUP,
    E_PG_INFOSCREEN,
    // ...
    MAX_PAGE
};
void show_page(const Page page) {
    ESP_LOGD(LOG_TAG, "showing page %i", (int)page);
    gslc_SetPageCur(&m_gui, (int16_t)page);

    if (page == Page::E_PG_STARTUP) {
        ESP_LOGD(LOG_TAG, "hiding status bar");
        gslc_SetPageBase(&m_gui, (int16_t) nullptr);
    } else {
        ESP_LOGD(LOG_TAG, "showing status bar");
        gslc_SetPageBase(&m_gui, (int16_t)Page::E_PG_BASE);
    }
}
```

That would have been so much easier than what I implemented in PR #538 :see_no_evil: That solution requires that the calling code calls the wrapper, not `gslc_SetPageCur` directly but that is not too difficult to ensure.

I think I am going to close this issue and the corresponding PR #538 as the goal can already be achieved with GUIslice although with a slightly different route than proposed in the feature request.

closed 2024-05-06 by ilka-schulz