ImpulseAdventure/GUIslice #543: Image not showing and getting constant "#if !(GSLC_SD_EN)" error when I'm enabling it

eagl1, 2024-05-20:
Hi,

Images don't show on the display.

I'm using both rp2040 and STM32 black pill, all works perfeectly with buttons, text and touch no problems. But the problem is with the images that show up as white blank.

I used your GUIslice_Image2C.

[image]

Pconti31, 2024-05-20:
@eagl1 Post a zip file of your project folder and I'll take a look.
Paul--

Pconti31, 2024-05-20:
@eagl1 In the Builder your image Property View: 
has Image Memory property.

My guess is that originally you had set Image Memory to gslc_GetImageFromSD((const char*)
Thus the reason it kept asking for SD card support.

For C files it should be set to gslc_GetImageFromRam((unsigned char*)
For spiffs it should be gslc_GetImageFromFile(

Make sure when you upload your images use SPIFFS not LittleFS or FatFS.
Paul--

eagl1, 2024-05-21:
OK, first in the GUIslice image2c, I checked both options:
1. C array as PROGMEM storage
2. Output little endian

Here are the files including the libraries and the project folder. I'm using now esp8266 lolin v3.

[libraries.zip](https://github.com/ImpulseAdventure/GUIslice/files/15391533/libraries.zip)
[project3.zip](https://github.com/ImpulseAdventure/GUIslice/files/15391534/project3.zip)

Thanks for all the support,

Pconti31, 2024-05-21:
@eagl1 My 320x480 screen got fried when I was helping another user out.  So I had to rescale yours to my 240x320.
Checkout the values I set on the image in property view for image memory.
Both C and Spiffs worked.
[project3-ram.zip](https://github.com/ImpulseAdventure/GUIslice/files/15394925/project3-ram.zip)
[project3-spiffs.zip](https://github.com/ImpulseAdventure/GUIslice/files/15394926/project3-spiffs.zip)
Paul--

Pconti31, 2024-05-21:
@eagl1 One more thing.  I believe Spiffs is only for ESP32 not rp2040  or STM32 .
Paul--

eagl1, 2024-05-21:
The upload from ram worked with me. Got back to me old sketch, and tried to know what's the issue and I think the solution is to start a new project.

Another question, do you mean that SPIFFS is for ESP32 only ? does that mean ESP8266 is also not supporting SPIFFS ?

eagl1, 2024-05-21:
Worked even on the black pill board:

[image]

eagl1, 2024-05-23:
I'm sorry but the only sketch worked is the one came from you, all other ones I do don't work ! I don't know why.

I'm following all the required steps, choosing `For C files it should be set to gslc_GetImageFromRam((unsigned char*)`.

eagl1, 2024-05-24:
I discovered one thing that could be a reason for deflected pictures. Which is by including the image by the C file and not by the bmp file. If the C formatting is correct then the image would be inserted as the original, if else, then it would be distorted. Like the following:

[image]

eagl1, 2024-05-24:
[image]

Yep, it worked, I have to include the C file and not the image itself and if it includes with no distortion then it will work, if not then the formatting isn't proper.

eagl1, 2024-05-24:
The only pictures worked with me are the raspberry and the last color ones. I don't know why other pictures aren't converted to proper C file ?

Pconti31, 2024-05-24:
@eagl1 Post a zip file of the failing *.bmp and its *.c image file.  I think its possible  bug with image_2c utility.
Paul--

eagl1, 2024-05-24:
[images.zip](https://github.com/ImpulseAdventure/GUIslice/files/15435193/images.zip)

I checked the source library, I really want to work and do any contribution to this wonderful project, but I have no idea where to start and where to start fixing the issues.

I thought of embedding `JPEGDecoder-master` and `PNGdec-master` libraries but that requires knowledge of the connected code in the `image_2c` and the guislice builder source code.

Pconti31, 2024-05-24:
@eagl1 Help is always welcome.  However, No need to modify image_2c for jpg or png there is already support.  Limits are more on GUIslice API.  

I'll look over your zip file and let you know if I find anything.
Paul--

eagl1, 2024-05-24:
Worked perfectly :)

Thanks a lot ..

closed 2024-05-24 by eagl1