ImpulseAdventure/GUIslice #548: Overlap and draw functions.

bear-gha, 2024-07-14:
Is it possible to add a parameter to draw functions that will bind them to specific pages?

Example, before:
gslc_DrawFillCircle(&m_gui, 310, 10, 5, GSLC_COL_GREEN);

after:
gslc_DrawFillCircle(&m_gui, E_PG_POPUP1, 310, 10, 5, GSLC_COL_GREEN);

Since at the moment, if a page appears, the drawing goes all the way through, and overlap is required.

Pconti31, 2024-07-14:
@bear-gha Yes, Of Course.   Basically, you simply need to add a box element to page and add the draw callback.
In the draw callback if u=you only have one page you are drawing on you just issue you draw commands.  More than one page simple test for the matching element referencees and make your draw commands inside the testing if statement.
Anyway, I'm sure this seems more complex then it actually is.  I have put together a respository of examples that are more complex techniques [GUIslice-Solutions](https://github.com/Pconti31/GUIslice-Solutions/tree/main)
and one example in particular matches your request:
[led-panel](https://github.com/Pconti31/GUIslice-Solutions/tree/main/apps/led-panel)

Paul--

bear-gha, 2024-07-14:
Thanks, I will look into this example.

closed 2024-07-14 by bear-gha