ImpulseAdventure/GUIslice #550: gslc_ElemXKeyPadCfgSetButtonSz NOT working

JoeHo8051, 2024-08-21:
### Describe the bug
gslc_ElemXKeyPadCfgSetButtonSz doesn't change the alpha keypad size or display at all 
here is my code :
  static gslc_tsXKeyPadCfg_Alpha sCfgTx;
    sCfgTx = gslc_ElemXKeyPadCfgInit_Alpha();
    gslc_ElemXKeyPadCfgSetButtonSz((gslc_tsXKeyPadCfg*)&sCfg, 10,5);
   m_pElemKeyPadAlpha = gslc_ElemXKeyPadCreate_Alpha(&m_gui, E_ELEM_KEYPAD_ALPHA, E_POP_KEYPAD_ALPHA,
    &m_sKeyPadAlpha, 15, 150, E_FONT_TXT1, &sCfgTx);

    
  gslc_ElemXKeyPadValSetCb(&m_gui, m_pElemKeyPadAlpha, &CbKeypad);

### Device hardware
- *MCU*: ESP32, etc.
- *Display with ILI9341 2.8" resistive

Pconti31, 2024-08-21:
@JoeHo8051 I don't know the size of the font you are trying to use but I think maybe you have made the buttons too small.  The default size for adafruit's builtin font E_BUILTIN5X8 is 6x9 and the button sizes would be 12x25 using the GUIsliceBuilder while you set 10x5???
Say if you wanted say double the size it would be
```
  gslc_ElemXKeyPadCfgSetButtonSz((gslc_tsXKeyPadCfg*)&sCfgTx, 24, 50);
```
If you use the builder to double the size you need to edit->Options select KeyPadAlpha Tab and set a new font,
do not use default sizes, Button Height and Width:
```
Font	BuiltIn(2x)->10x16pt7b
Corners Rounded?	false
Use Default Button Sizes?	false
Button Width	24
Button Height	50
```
Also, make sure your GUIslice_version.h has
```
#define GUISLICE_VER "0.17.1.6"
```
Of course, I don't recommend double the size for 2.8 display...

Paul--

JoeHo8051, 2024-08-22:
Thank you for the reply Paul.

So how to update the GUIslice version to 0.17.1.6, I tried E_FREESERIF9 and E_FREESANSBOLD12 on the function, but none of them work.

I set the size bigger by gslc_ElemXKeyPadCfgSetButtonSz((gslc_tsXKeyPadCfg*)&sCfg, 40, 40); ,still not working.

JoeHo8051, 2024-08-22:
Sorry, my mistake ,  it is supposed to be gslc_ElemXKeyPadCfgSetButtonSz((gslc_tsXKeyPadCfg*)&sCfgTx, 24, 50);
not  gslc_ElemXKeyPadCfgSetButtonSz((gslc_tsXKeyPadCfg*)&sCfg, 24, 50);
I use wrong varibale name ..

Thank you.

ImpulseAdventure, 2024-08-25:
Thanks Paul for the suggestions, and glad it now works.

closed 2024-08-25 by ImpulseAdventure