ImpulseAdventure/GUIslice #556: gslc_FontSet fail to add vlw font file from SPIFFTS

JoeHo8051, 2024-11-28:
### Describe the bug
I put my vlw file into spiffs, and I check the file do exist in the SPIFFS area.
It seems it failed to load the font , I saw the serial return "Font file cairomedium16.vlw not found! " when I enter the page using that font.

Here is my code:

  if (!gslc_Init(&m_gui,&m_drv,m_asPage,MAX_PAGE,m_asFont,MAX_FONT)) { return; }

  if (!SPIFFS.begin(true)) {
    return;
  }

  // ------------------------------------------------
  // Load Fonts
  // ------------------------------------------------
//<Load_Fonts !Start!>
    if (!gslc_FontSet(&m_gui,E_BUILTIN10X16,GSLC_FONTREF_PTR,NULL,2)) { return; }
    if (!gslc_FontSet(&m_gui,E_BUILTIN15X24,GSLC_FONTREF_PTR,NULL,3)) { return; }
    if (!gslc_FontSet(&m_gui,E_BUILTIN5X8,GSLC_FONTREF_PTR,NULL,1)) { return; }
    if (!gslc_FontSet(&m_gui,E_CAIROBOLD16,GSLC_FONTREF_FNAME,CAIROBOLD16_VLW,16)) { return; }
    gslc_FontSetMode(&m_gui, E_CAIROBOLD16, GSLC_FONTREF_MODE_1);

   if (!gslc_FontSet(&m_gui,E_CAIROMEDIUM16,GSLC_FONTREF_FNAME,"cairomedium16.vlw",16)) { return; }

### Device hardware
- *MCU*:  ESP32 
- *Display with link*: ili9488
- *GUIslice config selected*:  GUIslice_config.h:  uncomment  #include "../configs/esp-tftespi-default-xpt2046.h"

Pconti31, 2024-11-29:
@JoeHo8051 Post a zip file of a test project with just one text field using your font and include the full project folder with data folder and your cairomedium16.vlw
and I will take a look.
Also, If you used the builder then include your modified GUIsliceBuilder/templates/builder_fonts.json
Paul--

Pconti31, 2024-11-30:
@JoeHo8051 By the way here is a simple stand alone program without GUIslice or TFT_eSPI to list spiffs file names to serial output.
[list_files.zip](https://github.com/user-attachments/files/17966217/list_files.zip)
Paul--

Pconti31, 2025-03-04:
@JoeHo8051 Seems since I did the original work with SPIFFS the library for JPEG support that TFT_eSPI and TFT_eFEX uses has been converted to LittleFS.  
The easiest solution at the moment is to edit JPEGDecoder.h
```
    #include <SPIFFS.h>
    #define LittleFS SPIFFS
//    #include <LittleFS.h>
//    #define SPIFFS LittleFS
```
This should allows spiffs to load your files.