ImpulseAdventure/GUIslice #561: Callback when changing the page?

u1550l, 2025-02-05:
Hi Paul,
is there a mechanism available to get a callack when changing the active page?
I want to save the pageId in nvram and restore it after power failure.

If not: My first attempt would be to extend the gslc_SetPageCur() function.
What do you recommend?

Regards - Richard

Pconti31, 2025-02-09:
@u1550l No mechanism is available. It might be simplest to modify gslc_SetPageCur().  This will however avoid tracking popups and when a base page is the active page.   Trying to restore their state would be much harder, but if needed you should instead modify
gslc_SetStackPage() which all of the api and internal routines end up calling on page changes...

Paul--

u1550l, 2025-02-09:
@Pconti31 ,
thanks for you help. May be I will come back later with more detailed questions ...
-- Richard

ImpulseAdventure, 2025-02-17:
If I understand correctly, you are using the Builder to link a button action to change the current page, and you want a user callback to be triggered along with this page change.

If using the GUIslice API natively, you would normally be defining a handler for the button event, where you could both trigger a callback and also call SetPageCur().

As an example, see Example 05 (ex05_ard_pages.ino), with my comment added:

[image]

Perhaps this could be a future feature request for the Builder? (Paul, should I transfer the issue?)

In the meantime, trying to hack the SetPageCur() or the SetStackPage() could be a bit tricky you'd probably need to create a new set of functions to register a callback function into the APIs and then have SetPageCur() invoke it.

Another idea that you might be able to use without library changes: perhaps your main loop could simply detect a change in PageCur. In each pass, save GetPageCur() into a variable, and then look for any change. If it changes, then invoke your callback. Might that work?

u1550l, 2025-02-17:
Hi @ImpulseAdventure,
thanks for your suggestions, I never had a closer look to the CbBtnCommon. This is probably the easiest approach.
Regular control of the GUI is done via Touch. Only at boot-up the active page shall be set from other code areas.
 
I don't like the idea of permanent polling so much, although I know that this the core concept of Arduino. The CbBtnCommon() more like event driven.
-Richard

u1550l, 2025-02-19:
Hi @ImpulseAdventure, hi @Pconti31,
just an update:
I implemented my idea this way:

```
// Common Button callback
bool CbBtnCommon(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  // Typecast the parameters to match the GUI and element types
  gslc_tsGui*     pGui     = (gslc_tsGui*)(pvGui);
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem*    pElem    = gslc_GetElemFromRef(pGui,pElemRef);

  int currentPgId, newPgId;

  if ( eTouch == GSLC_TOUCH_UP_IN ) {

    // save current page, before button processing:
    currentPgId = gslc_GetPageCur(pGui);

    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_ELEM_BTN_PAGE_SWITCH_CONTROLS:
        gslc_SetPageCur(&m_gui, E_PG1_2_buttons);
        break;
      case E_ELEM_BTN_PAGE_SERVICE:
        gslc_SetPageCur(&m_gui, E_PG1_3_service);
        break;
      case E_ELEM_BTN16:
        gslc_SetPageCur(&m_gui, E_PG1_4_123bms);
        break;
        ...
        ...
//<Button Enums !End!>
      default:
        break;
    }

    // check if active page has changed:
    newPgId = gslc_GetPageCur(pGui);
    if (newPgId != currentPgId)
    {
      savePageId_toNvram(newPgId);
    }

  }
  return true;
}

```

i.e.: 
when a button is handled, I check if the gslc_GetPageCur() returns a different ID. If yes, it will be saved to Nvram.
At boot time, after initializing the gui, I simply call a gslc_SetPageCur(&m_gui, readPageId_fromNvram());
This solves my problem.

I will close the issue.
Thank you.
-Richard

closed 2025-02-19 by u1550l

ImpulseAdventure, 2025-02-19:
Thanks for sharing your followup solution, Richard!