ImpulseAdventure/GUIslice #565: error: call of overload 'error: call of overloaded 'Adafruit_ILI9341(int, int, int)' is ambiguous

renegadeandy, 2025-02-16:
Hi all,

I am following[ this page](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino) to get my first GUISlice running on my ESP32 with an ili9341 no touch display.

I have uncommented line :   #include "../configs/ard-adagfx-ili9341-notouch.h" from GUIslice_config.h as it looks like what I am using. (I use this display in other projects no problems by the way -- so we know it 'works'.

I go into the ard-adagfx-ili9341-notouch.h to correctly set the pin definitions as follows:

>   // For shields, the following pinouts are typically hardcoded
>   #define ADAGFX_PIN_CS       0    // Display chip select
>   #define ADAGFX_PIN_DC       27     // Display SPI data/command
>   #define ADAGFX_PIN_RST      33     // Display Reset
> 
>   // Display interface type
>   #define ADAGFX_SPI_HW       1	    // Display uses SPI interface: 1=hardware 0=software
> 
>   // Display interface software SPI
>   // - Hardware SPI: the following definitions are unused
>   // - Software SPI: the following pins need to be defined
>   #define ADAGFX_PIN_MOSI     23
>   #define ADAGFX_PIN_MISO     19
>   #define ADAGFX_PIN_CLK      18

When compiling I then get a compilation error which states:

> .pio/libdeps/esp32dev/GUIslice/src/GUIslice_drv_adagfx.cpp:269:92: error: call of overloaded 'Adafruit_ILI9341(int, int, int)' is ambiguous
>      Adafruit_ILI9341 m_disp = Adafruit_ILI9341(ADAGFX_PIN_CS, ADAGFX_PIN_DC, ADAGFX_PIN_RST);
>                                                                                             ^
> In file included from .pio/libdeps/esp32dev/GUIslice/src/GUIslice_drv_adagfx.cpp:68:

Sounds like there is more than one constructor with the same 3 int params, but I don't know why...and instead of diving down the rabbit hole - hope somebody can stop me wasting time and quickly identify where I have gone off the rails! 

Thanks all in advance!

Pconti31, 2025-02-16:
@renegadeandy I'm not sure what is going on but lets start with the fact  you are running an old version of GUIslice since
```
Line 269 Adafruit_ILI9341 m_disp = Adafruit_ILI9341(ADAGFX_PIN_CS, ADAGFX_PIN_DC, ADAGFX_PIN_RST);
```
 is in 0.17.0 while the latest version is 0.17.1.7 has 
```
    ILI9225_due m_disp(ADAGFX_PIN_CS, ADAGFX_PIN_DC, ADAGFX_PIN_RST);
```
 I susgest you  download from the repository using the green "<> CODE " pulldown button and Download zip.

You say you included  Adafruit_ILI9341.h?  GUIslice includes it.  If you use the GUIsliceBuilder it will add
```
#include <Adafruit_GFX.h>
```
to your project.  No need for anything else since GUIslice_drv_adagfx.cpp will pull in Adafruit_BusIO.h and Adafruit_ILI9341.h

Paul--

renegadeandy, 2025-02-16:
I spotted what was wrong.

The solution was changing it to using Software for the SPI interface by switching:

```
// Display interface type
  #define ADAGFX_SPI_HW       1	    // Display uses SPI interface: 1=hardware 0=software
```

to 

```
// Display interface type
  #define ADAGFX_SPI_HW       0	    // Display uses SPI interface: 1=hardware 0=software
```

Code compiles - and works as designed. w00t!

closed 2025-02-16 by renegadeandy