ImpulseAdventure/GUIslice #571: gslc_ElemXListboxAddItem only update 19 chars on each item of Listbox

JoeHo8051, 2025-02-25:
### Describe the bug

The Listbox could only display maximum 19 chars for each item?
The image below is the result of my code to displaying files from the SD card on ListBox 

[image]
The full name of the file is "Genmitsu_Controller.bin"

And here is my code to setup Listbox:

char    m_acListboxBuf6[512 + XLISTBOX_BUF_OH_R];

  pElemRef = gslc_ElemXListboxCreate(&m_gui,E_ELEM_LISTBOX_FIRMWARE,E_OTA_PAGE,&m_sListbox6,
    (gslc_tsRect){43,162,300,120},E_INTER_FONT_9,
    (uint8_t*)&m_acListboxBuf6,sizeof(m_acListboxBuf6),0);
  gslc_ElemXListboxSetSize(&m_gui, pElemRef, 3, 1); // 3 rows, 1 columns
  gslc_ElemXListboxItemsSetSize(&m_gui, pElemRef, 250, 33);
  gslc_ElemSetTxtMarginXY(&m_gui, pElemRef, 0, 5);
  gslc_ElemSetTxtCol(&m_gui,pElemRef,GSLC_COL_WHITE);
  gslc_ElemSetCol(&m_gui,pElemRef,((gslc_tsColor){50,54,78}),((gslc_tsColor){50,54,78}),((gslc_tsColor){95,92,187}));
  gslc_ElemXListboxSetSelFunc(&m_gui, pElemRef, &CbListbox);
  gslc_ElemXListboxItemsSetGap(&m_gui, pElemRef, 1,((gslc_tsColor){50,54,78}));
  gslc_ElemSetTxtAlign(&m_gui,pElemRef,GSLC_ALIGN_MID_LEFT);
  E_ELEM_LISTBOX_FIRMWARE_REF = pElemRef; 

### Device hardware
- *MCU*: ESP32 WROOM 32
- *GUIslice config selected*: ili9488, FT6206

*Please confirm whether:*
- *Display works correctly 
- *Touch works correctly 

### Expected behavior
Display each item of Listbox as long as the Listbox width.

Pconti31, 2025-02-26:
@JoeHo8051 Yes.  This is a feature.  Not a smart feature. It's because of this code in XListbox.c
```
// TODO: Combine with GUIslice MAX_STR
// Defines the maximum length of a listbox item
#define XLISTBOX_MAX_STR        20
```
It also turns out that someone else spotted this defect and made a pull request to use the input element field size.  I'll test out their code and if looks good I'll do the merge and create 0.17.1.9

Paul--

Pconti31, 2025-02-26:
@JoeHo8051 I have tested the pull request and it does work without problems.
It seems @ImpulseAdventure Calvin is going to spin a new major release 0.17.2 soon and likely will do the merge himself.

So in the meantime you can just modify the #define to whatever you need as a work-around.

Paul--

JoeHo8051, 2025-02-27:
Thank you very much !

JoeHo8051, 2025-02-27:
> [@JoeHo8051](https://github.com/JoeHo8051) Yes. This is a feature. Not a smart feature. It's because of this code in XListbox.c
>
> ```
> // TODO: Combine with GUIslice MAX_STR
> // Defines the maximum [...]

Thank you very much !

closed 2025-02-27 by JoeHo8051