ImpulseAdventure/GUIslice #574: Is it possible to get the SPI instance that is being used by the display for other devices

Jonathan-0101, 2025-04-19:
I'm working on a project with a custom ESP32-S3 board and multiple SPI devices sharing the bus and I was wondering how I can the SPI instance that is used by the library and pass it to other devices.

(Have a ST7796 display using TFT_eSPI with XPT2046 internal touch controller).

Thanks in advance.

Pconti31, 2025-04-22:
@Jonathan-0101 This is all handled within the TFT_eSPI library.  
You might want to ask on TFT-eSPI's issue forum.
As far as I can see spi for esp32's is handled inside Processors\TFT_eSPI_ESP32.c

The only thing I can think of is to edit TFT_eSPI.h like so:
```
//#if !defined (TFT_PARALLEL_8_BIT) && !defined (RP2040_PIO_INTERFACE)
  static   SPIClass& getSPIinstance(void); // Get SPI class handle
//#endif
```
Then call GUIslice api:
```
TFT_eSPI display = (TFT_eSPI)gslc_DrvGetDriverDisp(&m_gui);
 SPIClass my_spi = display.getSPIinstance());
```
Although I have no idea if thats useful or not.  I also don't know if there is an easier way.
Paul--

Pconti31, 2025-04-22:
@Jonathan-0101 Oh!  You would also need to edit TFT_eSPI.cpp to match the header...
Paul--

Jonathan-0101, 2025-04-22:
@Pconti31 On an unrelated note, the touch on the program I'm working on only works intermittently (tested with multiple displays) and have tried calibrating and adjusting sensitivity with no luck. Half the time when the program loads no touch will register and the rest of the time, touch will eventually stop working. Any suggestions on how to fix this? Thanks.

closed 2025-08-19 by Jonathan-0101